evcc-io/evcc #12097: Tesla m3 kann immer noch nicht gesteuert werden

pit65535, 2024-02-06:
### Describe the bug

Sorry, wenn ich nerve, aber ich habe nach dem Update auf 0.124.1 als Homeassistant-Plugin wieder dasselbe Problem, dass evcc mein Auto nicht steuern kann.
Ich habe bereits über evcc.io eienn neuen API-Token generiert und die evcc.yaml auf dene neuesten Stand gebracht.
Im Web interface zeigt evcc an Model 3 (offline)

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
chargers:
  - name: WC3Garage
    type: template
    template: twc3
    host: 192.168.245.214 # IP-Adresse oder Hostname 
  - name: WC3Eingang
    type: template
    template: twc3
    host: 192.168.245.247 # IP-Adresse oder Hostname
#    uri: 192.168.0.8:502 # ModBus address

# vehicle definitions
# name can be freely chosen and is used as reference when assigning vehicle to loadpoint
# for documentation see https://docs.evcc.io/docs/devices/vehicles
vehicles:
  - name: Model_3
    type: template
    template: tesla-command
    title: Model 3 # Wird in der Benutzeroberfläche angezeigt # Optional
    accessToken: ...... # Siehe https://docs.evcc.io/docs/devices/vehicles#tesla
    refreshToken: EU_....... # Siehe https://docs.evcc.io/docs/devices/vehicles#tesla
#    vin: 5YJ3E7EB3MF944349 # Erforderlich, wenn mehrere Fahrzeuge des Herstellers vorhanden sind # Optional##Car name: id:1493122444017040
    capacity: 82 # Akkukapazität in kWh # Optional 
#    onIdentify: # set defaults when vehicle is identified
    mode: pv # enable PV-charging when vehicle is identified
##    minSoc: 15 # immediately charge to 0% regardless of mode unless "off" (disabled)
##    targetSoc: 90 # limit charge to 90%
```

### Log details

```text
[site  ] DEBUG 2024/02/06 14:35:09 ----
[lp-1  ] DEBUG 2024/02/06 14:35:09 charge power: 2151W
[lp-2  ] DEBUG 2024/02/06 14:35:09 charge power: 0W
[site  ] DEBUG 2024/02/06 14:35:09 pv power: 3588W
[site  ] DEBUG 2024/02/06 14:35:09 battery soc: 100%
[site  ] DEBUG 2024/02/06 14:35:09 battery power: -792W
[site  ] DEBUG 2024/02/06 14:35:09 grid power: 78W
[site  ] DEBUG 2024/02/06 14:35:09 grid currents: [1.17 0.89 0.95]A
[site  ] DEBUG 2024/02/06 14:35:09 site power: -714W
[lp-1  ] DEBUG 2024/02/06 14:35:09 charge voltages: [236 232 233]V
[lp-1  ] DEBUG 2024/02/06 14:35:09 detected connected phases: 3p
[lp-1  ] DEBUG 2024/02/06 14:35:09 charge currents: [3.1 3.1 3]A
[lp-1  ] DEBUG 2024/02/06 14:35:09 detected active phases: 3p
[lp-1  ] DEBUG 2024/02/06 14:35:09 charger status: C
[lp-1  ] DEBUG 2024/02/06 14:35:09 pv charge current: 1.03A = 0A + 1.03A (-714W @ 3p)
[lp-1  ] ERROR 2024/02/06 14:35:09 max charge current 1A: vehicle not capable of current control
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.124.1

litithio, 2024-02-06:
Das liegt daran, dass das Steuern des Autos noch nicht möglich ist. Siehe #11846 und #11893.

andig, 2024-02-06:
Soweit es das Token betrifft: ist sichergestellt, dass evcc bei Dir auf seine Datenbank zugreifen kann? Dort speichert es nach demnotwendigen Refresh dann die neuen Tokens. Das Token aus der Config expired sehr schnell. Ob gespeichert kannst du mit

    evcc settings get tesla-command

überprüfen.

closed 2024-02-06 by andig

pit65535, 2024-02-06:
Hallo Andig,
danke für Deine Hilfe, aber wo kann ich den Befehl eingeben? Ich habe ein Homeassistant addon und an der shell geht das irgendwie nicht....
Aber ich bin mir ziemlich sicher, dass evcc auf die DB schreiben kann, denn die DB wurde heute um 15:01 geändert und es lief ja auch die ganze Zeit.....

andig, 2024-02-06:
> Ich habe ein Homeassistant addon und an der shell geht das irgendwie nicht....

Ist leider so. Deshalb steht auch im Issuetemplate:

> NOTE if you're using HomeAssistant or Docker we ask you to reproduce the problem on plain Linux or Windows first.

pit65535, 2024-02-06:
Vielleicht bring das log was, hab grad mal evcc neu gestartet:

Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/config/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2024/02/06 16:12:03 evcc 0.124.1
[main  ] INFO 2024/02/06 16:12:03 using config file: /config/evcc.yaml
[main  ] INFO 2024/02/06 16:12:03 starting ui and api at :7070
[mqtt  ] INFO 2024/02/06 16:12:03 connecting evcc-333824260 at tcp://192.168.245.134:1883
[main  ] ERROR 2024/02/06 16:12:03 creating vehicle Model_3 failed: cannot create vehicle type 'template': cannot create vehicle type 'tesla-command': missing token setting for 131a71ca-f310-486c-aa97-424090d2fbbd: not found
[lp-1  ] WARN 2024/02/06 16:12:03 deprecated: minCurrent setting is ignored, please remove
[lp-1  ] WARN 2024/02/06 16:12:03 deprecated: maxcurrent setting is ignored, please remove
[lp-1  ] WARN 2024/02/06 16:12:03 deprecated: phases setting is ignored, please remove
[lp-2  ] WARN 2024/02/06 16:12:03 deprecated: minCurrent setting is ignored, please remove
[lp-2  ] WARN 2024/02/06 16:12:03 deprecated: maxcurrent setting is ignored, please remove
[lp-2  ] WARN 2024/02/06 16:12:03 deprecated: phases setting is ignored, please remove
[site  ] WARN 2024/02/06 16:12:03 battery configured but residualPower is missing or <= 0 (add residualPower: 100 to site), see https://docs.evcc.io/en/docs/reference/configuration/site#residualpower
[site  ] INFO 2024/02/06 16:12:03 site config:
[site  ] INFO 2024/02/06 16:12:03   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2024/02/06 16:12:03     grid:      power ✓ energy ✗ currents ✓
[site  ] INFO 2024/02/06 16:12:03     pv 1:      power ✓ energy ✗ currents ✗
[site  ] INFO 2024/02/06 16:12:03     pv 2:      power ✓ energy ✗ currents ✗