evcc-io/evcc #12105: Wallbox bricht ständig das Laden ab. Wallbox Go-e: charger out of sync: expected enabled, got disabled

Juppme, 2024-02-06:
### Describe the bug

Ich weiß leider nicht mehr weiter. 
Der Go-E bricht ständig das laden ab.
Fehlermeldung: Wallbox Go-e: charger out of sync: expected enabled, got disabled
Wlanverbindung ist stark und stabil. Bin leider total am verzweifeln. 
Der Ladevorgang dauert jetzt statt ca 3 h bis zu 6 h und ist bestimmt auch nicht gut für den Akku. 

Problem ist erst seit 2-3 Wochen. 

### Steps to reproduce

1. Laden mit der Funktion "Schnell"
2.
3.
...

### Configuration details

```yaml
network:
  # schema is the HTTP schema
  # setting to `https` does not enable https, it only changes the way URLs are generated
  schema: https
  # host is the hostname or IP address
  # if the host name contains a `.local` suffix, the name will be announced on MDNS
  # docker: MDNS announcements don't work. host must be set to the docker host's name.
  host: 192.168.100.138
  # port is the listening port for UI and api
  # evcc will listen on all available interfaces
  port: 7070

interval: 55s # control cycle interval

# database configuration for persisting charge sessions and settings
# database:
#   type: sqlite
#   dsn: <path-to-db-file>

# sponsor token enables optional features (request at https://sponsor.evcc.io)
sponsortoken: xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx # your token from sponsor.evcc.io

# telemetry enables aggregated statistics
#
# Telemetry allows collecting usage data (grid and green energy, charge power).
# Data is aggregated, no individual charging sessions are tracked. The collected,
# anonymous data can be retrieved using https://api.evcc.io.
#
# See https://github.com/evcc-io/evcc/pull/4343 or details.
#
# For time being, this is only available to sponsors, hence data is associated with
# the sponsor token's identity.
#
# telemetry: true

# log settings
log: info
levels:
  site: debug
  lp-1: debug
  lp-2: debug
  cache: error
  db: error

# modbus proxy for allowing external programs to reuse the evcc modbus connection
# each entry will start a proxy instance at the given port speaking Modbus TCP and
# relaying to the given modbus downstream device (either TCP or RTU, RS485 or TCP)
modbusproxy:
  #  - port: 5200
  #    uri: solar-edge:502
[... 283 lines cut ...]
[site  ] DEBUG 2024/02/06 17:48:44 ----
[lp-1  ] DEBUG 2024/02/06 17:48:44 charge power: 0W
[site  ] DEBUG 2024/02/06 17:48:44 pv power: 0W
[site  ] DEBUG 2024/02/06 17:48:44 grid power: 4733W
[site  ] DEBUG 2024/02/06 17:48:44 grid powers: [414 4042 277]W
[site  ] DEBUG 2024/02/06 17:48:44 grid currents: [3.21 18.4 2.3]A
[site  ] DEBUG 2024/02/06 17:48:44 site power: 4733W
[lp-1  ] DEBUG 2024/02/06 17:48:44 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2024/02/06 17:48:44 charge total import: 1331.022kWh
[lp-1  ] DEBUG 2024/02/06 17:48:44 charger status: B
[lp-1  ] INFO 2024/02/06 17:48:44 stop charging <-
[lp-1  ] DEBUG 2024/02/06 17:48:44 wake-up timer: start
[lp-1  ] DEBUG 2024/02/06 17:48:44 charge total import: 1331.022kWh
[lp-1  ] DEBUG 2024/02/06 17:48:52 soc estimated: 95.45% (vehicle: 95.00%)
[lp-1  ] DEBUG 2024/02/06 17:48:52 vehicle soc: 95%
[lp-1  ] DEBUG 2024/02/06 17:48:52 vehicle soc limit: 0%
[lp-1  ] DEBUG 2024/02/06 17:48:52 vehicle range: 48km
[lp-1  ] WARN 2024/02/06 17:48:52 charger out of sync: expected enabled, got disabled
[lp-1  ] DEBUG 2024/02/06 17:48:52 charger enable
[lp-1  ] DEBUG 2024/02/06 17:48:52 wake-up timer: start
```

### What type of operating system are you running?

HomeAssistant Add-on

Juppme, 2024-02-06:
Hier noch ein Screenshot der Wallbox. die das ladeverhalten zeigt. 
Ich hoffe ich habe alles richtig beschrieben das mit jemand helfen kann bzw. das Supoort Team damit zurecht kommt. 
Im Voraus herzlichen Dank
 

[image]

VolkerK62, 2024-02-06:
vor 3 Wochen hat alles funktioniert?
Gab es ein Update beim go-e?
Ist da irgendeine go-e interne regelung aktiv? (z.B. Eco Modus)

Juppme, 2024-02-06:
Eco Modus ist nicht aktiv. 
Es gibt ein Update. Habe ich installiert und bin wieder zur Vorgänger Version gewechselt. 
Fehler ist bei beiden Softwareversionen.

Juppme, 2024-02-06:
Also ganz ehrlich. 
Wie bescheuert muß man eigentlich sein. 
### Ich bin so blöd. 
Habe in einer geistigen Umnachtung Celver-PV getestet und die Wallbox nicht wieder aus der Konfiguration genommen. 
Sorry an alle die sich bereits damit beschäftigt haben

closed 2024-02-06 by Juppme