evcc-io/evcc #12111: Tesla Toke still expiring

buliwyf42, 2024-02-06:
### Describe the bug

I'm using the latest nightly after the issues with 0.124 but it seems my token expires after some time anyway.
I configured it via the UI. 
I had this several time since yesterday.
Writing to the db is possible. 

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
Configured via UI
```

### Log details

```text
Feb  6 13:35:27 evcc evcc[6262]: [db    ] INFO 2024/02/06 13:35:27 using sqlite database: /var/lib/evcc/evcc.db
Feb  6 13:35:27 evcc systemd[1]: ssh@26-192.168.203.61:22-192.168.203.125:65387.service: Deactivated successfully.
Feb  6 13:35:27 evcc systemd[1]: session-4066.scope: Deactivated successfully.
Feb  6 13:35:27 evcc evcc[6262]: [mqtt  ] INFO 2024/02/06 13:35:27 connecting evcc-1141834791 at tcp://mqtt.home.lan:1883
Feb  6 13:35:28 evcc evcc[6262]: [main  ] ERROR 2024/02/06 13:35:28 creating vehicle db:4 failed: cannot create vehicle type 'template': cannot create vehicle type 'tesla-command': token expired
Feb  6 13:35:28 evcc evcc[6262]: [lp-2  ] WARN 2024/02/06 13:35:28 poll mode '{always 30s}' may deplete your battery or lead to API misuse. USE AT YOUR OWN RISK.
Feb  6 13:35:28 evcc evcc[6262]: [site  ] WARN 2024/02/06 13:35:28 vehicle 'Warmwasser' does not support automatic detection
Feb  6 13:35:28 evcc evcc[6262]: [site  ] WARN 2024/02/06 13:35:28 vehicle 'NEO (offline)' does not support automatic detection
Feb  6 13:35:28 evcc evcc[6262]: [site  ] INFO 2024/02/06 13:35:28 site config:

evcc settings get tesla-command
[main  ] INFO 2024/02/06 20:33:31 evcc 0.124.1
[main  ] INFO 2024/02/06 20:33:31 using config file: /etc/evcc.yaml
[db    ] INFO 2024/02/06 20:33:31 using sqlite database: /root/.evcc/evcc.db
[mqtt  ] INFO 2024/02/06 20:33:31 connecting evcc-1003487051 at tcp://mqtt.home.lan:1883
root@evcc:~# evcc configure
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.124.1

andig, 2024-02-06:
Could you share a fresh token with info@evcc.io?

buliwyf42, 2024-02-06:
Both Token?

andig, 2024-02-06:
Identical to what youve used in the ui

buliwyf42, 2024-02-06:
You have mail.

andig, 2024-02-07:
Looks like Token works on first use. If it is not refreshed _and_ evcc restarted while token expires, then code would not use the refresh token from config and instead fail. This is fixed by https://github.com/evcc-io/evcc/pull/12121.

closed 2024-02-07 by andig

naltatis, 2024-02-07:
@buliwyf42 Yes, this looks like a UI or communication issue. Can you open a new bug with some more details? Browser version and screenshot from the network tab of the browser development tools would help to narrow in on this issue.

andi0b, 2024-02-12:
@andig I'm also having issues with Tesla token refresh, next expiry is in 7 hours and I set up collection logs (running on nightly 0.124.1 (39bb31a1))

Just one question about the token refresh process: Shouldn't the refreshed token be stored somewhere even after the first startup of evcc?

I looked into the sqlite db and couldn't find it there:

```
sqlite3 "file:./evcc.db?mode=ro"
SQLite version 3.37.2 2022-01-06 13:25:41
Enter ".help" for usage hints.
sqlite> select * from settings;
lp1.planEnergy|0
lp1.planTime|0001-01-01T00:00:00Z
bufferSoc|100
smartCostLimit|0.01
lp1.mode|pv
vehicle.a.planTime|0001-01-01T00:00:00Z
vehicle.a.planSoc|0
lp1.limitSoc|0
bufferStartSoc|0
prioritySoc|60
lp1.minCurrent|6
lp1.maxCurrent|32
lp1.phasesConfigured|3
batteryDischargeControl|false
vehicle.b.planTime|2024-02-12T06:00:00Z
vehicle.b.planSoc|70
vehicle.b.limitSoc|70
vehicle.b.minSoc|25
```

andig, 2024-02-12:
It will only be refreshed if expired. If your yaml access token is still valid, we‘ll use that. You could try placing any expired token there for testing though.

andig, 2024-02-12:
Please open new issue, this seems more about ui refresh than token expiry now.