evcc-io/evcc #12112: Loadpoint not visible in load history

leetillu, 2024-02-06:
### Describe the bug

Since February my the loadpoints are not displayed in load history anymore.
Latest update installed.
Going back to January, they are displayed.

[image]
[image]

### Steps to reproduce

1. Update to latest version
2. Go to load history
3. Find no load points
...

### Configuration details

```yaml
meters:

- type: template

  template: solax 

  id: 1  

  host: 192.168.178.73  

  port: 502  

  usage: grid  

  modbus: tcpip  

  name: grid1

- type: template

  template: solax 

  id: 1  

  host: 192.168.178.73  

  port: 502  

  usage: pv  

  modbus: tcpip  

  name: pv2

- type: template

  template: solax 

  id: 1  

  host: 192.168.178.73  

  port: 502  

  usage: battery  

  modbus: tcpip  

  name: battery3

chargers:

- type: template

  template: go-e-v3 

  host: 192.168.178.70  

  name: wallbox5

- type: template

  template: go-e-v3 

  host: 192.168.178.68  

  name: wallbox6

vehicles:

- type: template

  template: kia 

  title: Kia Niro  

  user: xx

  password: xx

  vin: xx

  capacity: 64  

  phases: 3  

  icon: car  

  cache: 15m  

  language: de  

  mode: now  

  minCurrent: 6  

  maxCurrent: 16   

  name: ev4

- type: template

  template: kia 

  title: Kia ev6  

  user: xx

  password:xx

  vin: xx

  capacity: 77  

  phases: 3  

  icon: car  

  cache: 15m  

  language: de  

  mode: pv  

  minCurrent: 6  

  maxCurrent: 16  

  name: ev5

loadpoints:

- title: Wallbox 2

  charger: wallbox5

  mode: now

  phases: 0

  mincurrent: 6

  maxcurrent: 16

- title: Wallbox 1

  charger: wallbox6

  mode: pv

  phases: 0

  mincurrent: 6

  maxcurrent: 16

site:

  title: Home

  meters:

    grid: grid1

    pv:

    - pv2

    battery:

    - battery3

  residualPower: 100
```

### Log details

```text
site  ] DEBUG 2024/02/06 21:07:32 grid power: -45W
[site  ] DEBUG 2024/02/06 21:07:32 site power: 523W
[lp-2  ] DEBUG 2024/02/06 21:07:32 charge currents: [0 0 0]A
[lp-2  ] DEBUG 2024/02/06 21:07:32 charge total import: 779.415kWh
[lp-2  ] DEBUG 2024/02/06 21:07:32 charger status: A
[site  ] DEBUG 2024/02/06 21:07:41 ----
[lp-1  ] DEBUG 2024/02/06 21:07:41 charge power: 0W
[lp-2  ] DEBUG 2024/02/06 21:07:41 charge power: 0W
[site  ] DEBUG 2024/02/06 21:07:41 pv power: 0W
[site  ] DEBUG 2024/02/06 21:07:42 battery soc: 58%
[site  ] DEBUG 2024/02/06 21:07:42 battery power: 500W
[site  ] DEBUG 2024/02/06 21:07:42 grid power: -65W
[site  ] DEBUG 2024/02/06 21:07:42 site power: 535W
[lp-1  ] DEBUG 2024/02/06 21:07:42 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2024/02/06 21:07:42 charge total import: 1776.155kWh
[lp-1  ] DEBUG 2024/02/06 21:07:42 charger status: B
[site  ] DEBUG 2024/02/06 21:07:51 ----
[lp-1  ] DEBUG 2024/02/06 21:07:51 charge power: 0W
[lp-2  ] DEBUG 2024/02/06 21:07:51 charge power: 0W
[site  ] DEBUG 2024/02/06 21:07:51 pv power: 0W
[site  ] DEBUG 2024/02/06 21:07:52 battery soc: 58%
[site  ] DEBUG 2024/02/06 21:07:52 battery power: 406W
[site  ] DEBUG 2024/02/06 21:07:52 grid power: -0W
[site  ] DEBUG 2024/02/06 21:07:52 site power: 506W
[lp-2  ] DEBUG 2024/02/06 21:07:52 charge currents: [0 0 0]A
```

### What type of operating system are you running?

Linux

### Version

Latest

andig, 2024-02-06:
All same Loadpoint? Then its probably hidden.

leetillu, 2024-02-06:
Two different loadpoints, or will it just be shown when a second one was used?

andig, 2024-02-06:
Exactly- maybe thats a bit too clever.

@naltatis only hide vehicles/loadpoints on existence, not in actual usage? This issue has come up before.

leetillu, 2024-02-06:
Sorry, but yes, I was not expecting that. Can be closed if no further discussion needed

naltatis, 2024-02-07:
I've removed the hiding logic. Predictability and simplicity for the win 😄

PR #12124 (2024-02-07): Sessions: always show loadpoint and vehicle
fixes #12112

- removed logic to only show loadpoint/vehicle name if different ones exist in sessions list

This attempt to optimize session log for simple installations (1 vehicle and/or 1 loadpoint) created more confusion than we gained through saving screen estate by hiding the info.
closed 2024-02-07 by andig