evcc-io/evcc #12115: nach dem Update startet das Programm nicht mehr

JensPrange, 2024-02-06:
Ich habe die 124.1 eingespielt (auf einem Windows Rechner)
wenn ich dann das Programm in der Konsole starte kommt folgende Fehlermeldung und es geht nicht weiter.

C:\evcc>evcc
[main  ] INFO 2024/02/06 23:56:14 evcc 0.124.1
[main  ] INFO 2024/02/06 23:56:14 using config file: C:\evcc\evcc.yaml
[main  ] INFO 2024/02/06 23:56:14 starting ui and api at :6060
[db    ] INFO 2024/02/06 23:56:14 using sqlite database: C:\Users\kreat\.evcc\evcc.db
[main  ] ERROR 2024/02/06 23:56:16 creating vehicle i4 failed: cannot create vehicle type 'template': cannot create vehicle type 'bmw': cannot get vehicles: unexpected status: 403 (Forbidden)
[main  ] FATAL 2024/02/06 23:56:16 cannot create vehicle 'Megane': cannot create vehicle type 'template': invalid key: targetsoc
[main  ] FATAL 2024/02/06 23:56:16 will attempt restart in: 5m0s

Hier meine evcc config

network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 6060

log: info
levels:
  cache: error

# unique installation id
plant: 6a8a3e4a7c62ffacc2a88461e851bb5f9204c0701e629b498b780c515bcc58fa

interval: 45s # control cycle interval

sponsortoken: xxx

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

meters:
- type: template
  template: kostal-plenticore 
  id: 71  
  host: 172.26.60.142  
  port: 1502  
  usage: pv 
  modbus: tcpip  
  name: pv1
- type: template
  template: kostal-plenticore 
  id: 71  
  host: 172.26.60.142  
  port: 1502  
  usage: battery  
  modbus: tcpip  
  name: battery2
- type: template
  template: kostal-ksem 
  id: 71  
  host: 172.26.60.143  
  port: 502  
  usage: grid  
  modbus: tcpip  
  name: grid3
  

chargers:
- type: template
  template: keba 
  host: 172.26.60.133  
  serial: 22182806  
  name: wallbox6

vehicles:
- type: template
  template: renault 
  title: Megane  
  user: xxx 
  password: xxx
  vin: xxx
  capacity: 40  
  phases: 3  
  cache: 15  
  mode: pv  
  minSoC: 10  
  targetSoC: 100  
  minCurrent: 6  
  maxCurrent: 16  
  name: Megane

- name: i4
  type: template
  template: bmw
  title: BMW i4
  user: xxx
  password: xxx # password
  vin: xxx
  capacity: 67 # kWh
  cache: 5m

loadpoints:
- title: Garage
  charger: wallbox6
  mode: now
  phases: 3
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: true

site:
  title: Zuhause
  meters:
    grid: grid3
    pv:
    - pv1
    battery:
    - battery2
  residualPower: 100
  bufferSoC: 90 # hier 90%
  prioritySoC: 85 # hier 85% (Hausbatterie bekommt bis zum SoC 90% Priorität beim laden)
    

Vllt kann mir ja jemand helfen.

Danke

LG
Jens

VolkerK62, 2024-02-07:
Da sind einige alte Parameter drin, die raus müssen.

vehicles: `minsoc`, `targetsoc`
loadpoints: `phases`, `mincurrent`, `maxcurrent`, `resetOnDisconnect`
site: `buffersoc`, `prioritysoc`

Das kann jetzt alles in der Web Ansicht eingestellt werden und wird in der Datenbank gespeichert.

closed 2024-02-07 by andig

JensPrange, 2024-02-07:
Danke.... hat geklappt. Ich wusste zwar das man das über die Seite jetzt einstellen kann, aber habe gedacht es ist nicht schlim wenn es beim Programstart schon vorgegeben wird.
Aber OK... jetzt läuft es wieder...super