evcc-io/evcc #12117: Werte von Wallbox werden falsch angezeigt und evcc steuer Wallbox nicht

lakatusch, 2024-02-07:
### Describe the bug

Verbrauch in evcc weicht von tatsächlichem Verbrauch ab und auch von der Vorgabe, welche ich in evcc angegeben habe:

[image]

in wirklichkeit wird aber mit 16a geladen:

[image]

die Shelly s bestätigen den 3,6kw Verbrauch
[image]

Wallbox ist die Cfox Power Brain 2.0 mit SW-Version ,1.24.4 in der 11kw Variante

### Steps to reproduce

Maximaler Ladestrom wurde eingestellt:

[image]

anschließend wurde geplantes laden ausgewählt

### Configuration details

```yaml
chargers:
  - name: my_charger
    type: template
    template: cfos
    host: 192.168.0.227 # IP-Adresse oder Hostname
```

### Log details

```text
kein Log
```

### What type of operating system are you running?

HomeAssistant Add-on

julred, 2024-02-07:
Welche evcc Version setzt du ein?
Funktioniert bei dir die Phasenumschaltung der cfos Box?

Hast du den Energiemeter im evcc.yaml eingetragen?

```
meters:
- type: template
  template: cfos
  id: 1
  host: IP-DER-WALLBOX:4701
  usage: charge
  modbus: tcpip
  name: charge_brain

loadpoints:
- title: Garage
  charger: cfos_brain
  meter: charge_brain
  mode: pv
  phases: 3
  mincurrent: 6
  maxcurrent: 16
```

Dann nimmt evcc die tatsächlichen Verbrauchswerte und nicht die prognostizierten.

lakatusch, 2024-02-07:
evcc Version ist 0.124.1

Meter habe ich jetzt ergänzt, war bisher nicht in der Config.

Die automatische Phasenumschaltung habe ich nicht aktiviert. Manuell kann ich aber zwischen 1 und 3 Phasen wechseln.

aber... selbst wenn die Verbrauchsanzeige jetzt richtig ist (noch nicht getestet), dann ist immer noch die Frage, warum mit 16a geladne wird, wenn ich in evcc auf max 8a begrenzt habe...

julred, 2024-02-07:
Welche Einstellungen hast du in der cfos?
Poste mal deine komplette evcc.yaml