evcc-io/evcc #12120: Using retained mqtt information, even if they're old

binderth, 2024-02-07:
**Is your feature request related to a problem? Please describe.**
I'm using an external vehicle integration, which sends MQTT-messages for all relevant vehicle information. It sends messages with the "retain"-flag and don't send same values until they've changed.
Which leads to evcc to not know SoC and range of the vehicle

**Describe the solution you'd like**
Let evcc use retained messages if the vehicle changes for a loadpoint - even if the value is older than configured timeout.

**Describe alternatives you've considered**
tried to change the source to send MQTT-messages regularly - but it's not possible

**Additional context**
[image]

(1) as the vehicleStatus changed, it is "connected"
(2) as the SoC/range information didn't change, it is "disconnected"

andig, 2024-02-07:
You'll have to set timeout to a long value like 365d.

closed 2024-02-07 by andig

binderth, 2024-02-07:
yes, but that is my point: "normally" 20mins are long enough.
The request is to use retained messages "onChange" of a vehicle. Setting a 365d timeout is not the solution here.

andig, 2024-02-07:
How would you to that? I repeat:

> There is no way for the client to distinguish between retained or not.

binderth, 2024-02-07:
let the "onChange"-event ignore the timeout.

andig, 2024-02-07:
Thats not possible. You‘ll need to configure a higher timeout for that purpose.

andig, 2024-02-07:
I really don't understand what you're asking here. The plugins obtain data and relay the information if that data is current. They have no notion of the context they operate in (and should not). Either data is received in time per plugin or it is not. How you configure that plugin is users choice and responsibility. If that's not sufficient you'll need to workaround whatever deficit is observed outside of evcc.

andig, 2024-02-07:
> evcc is not capable of processing non-live data out of context

That's just plain wrong. It can process anything as long as user declares it valid, i.e. not deprecated.

> That's really not a solution, sorry.

If you know the retained messages are valid then just configure a long timeout. With plugins it is your responsibility to deliver data and manage its validity.

binderth, 2024-02-07:
> If you know the retained messages are valid then just configure a long timeout. 

That's not the purpose of the timeout-parameter.

> That's just plain wrong. It can process anything as long as user declares it valid, i.e. not deprecated.

no. evcc won't process information that were sent to it while the vehicle is not active. That's the whole point of my feature request!

You're still not answering my question: Will evcc process information in future, if all vehicle information is not loadpoint-based anymore? Then everything will be fine.

andig, 2024-02-07:
> That's not the purpose of the timeout-parameter.

It is in evcc and as far as MQTT is concerned. 

> evcc won't process information that were sent to it while the vehicle is not active.

It doesn't- simply because there is no notion of an "inactive vehicle" entity. That is however totally independent of MQTT or whatever other protocol or retained messages.

binderth, 2024-02-07:
> Hope this clarifies things.

still not. sorry.

nevertheless last try: will evcc process information of not connected, but configured vehicles, once all vehicle information is moved from loadpoint to its own vehicle environment?