evcc-io/evcc #12123: Tesla Information is not pulled from the DB

buliwyf42, 2024-02-07:
### Describe the bug

I switched the config of my Tesla from Yaml to UI.
While the past to the db seems to work. (Other Issues aside) if i try edit the vehicle entry the popup comes up empty.
[image]
If i enter data it seems the entry is indeed updatet in the DB. At least there is no new entry. created. 

### Steps to reproduce

1. Add tesla
2. Validate and save config.
3.Restart
4. try to edit again.
...

### Configuration details

```yaml
I tried it with Chrome. Firefox and Safari on MacOs.
The same behavior occurs on an iPhone in Chrome
```

### Log details

```text
Nothing specific in the log.
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.124.1 (de1fb86d)

andig, 2024-02-07:
As requested before: please add screenshots from network tab

buliwyf42, 2024-02-07:
Here it is. 
[image]

naltatis, 2024-02-07:
I'm a little surprised about the `content.js` script that throws log errors. This is not from evcc. Can you please disable all browser plugins and try again. And try if this behavior also appears in another browser.

andig, 2024-02-07:
Regarding network tab: we need a look at the XHRs, not the rest.

buliwyf42, 2024-02-07:
Safari:
[image]
Chrome:
[image]
Firefox:
[image]

andig, 2024-02-08:
> I'm a little surprised about the content.js script that throws log errors. This is not from evcc. Can you please disable all browser plugins and try again.

@buliwyf42 hattest Du das repariert? Sonst würde ich hier zu machen.

buliwyf42, 2024-02-08:
> > I'm a little surprised about the content.js script that throws log errors. This is not from evcc. Can you please disable all browser plugins and try again.
> 
> @buliwyf42 hattest Du das repariert? Sonst würde ich hier zu machen.
[image]

Sorry hatte ich vergessen.  In Chrome habe ich alle Extension deaktiviert. In Safari und Firefox habe ich keine extensions. 
Hat es eventuell was mit meinem Hostnamen zu tun? evcc.home.lan ?

buliwyf42, 2024-02-08:
Hier in Bewegung

https://github.com/evcc-io/evcc/assets/4658822/2d6c2a0f-92aa-4c16-986b-bd3675b75c51

andig, 2024-02-08:
@naltatis sieht ähnlich aus wie https://github.com/evcc-io/evcc/issues/12015?

andig, 2024-02-08:
Zeig uns bitte mal den Inhalt query/response des letzten Requests wenn der Tesla zum editieren geladen wird.

buliwyf42, 2024-02-08:
[image]
Meinst du diesne Response so wie bei #12015 ?

andig, 2024-02-08:
Sieht gut aus. Keine Idee, warum das nicht im UI ankommen sollte :( @naltatis

scharmerd, 2024-02-10:
Hallo,
scheine wohl das gleiche Problem zu haben.

Bis jetzt fiel mir das Problem nur beim EVCC update auf, das ich per Docker update anstoße.

Nach dem Update wurde das Fahrzeug immer mit dem Zusatz offline angezeigt. SOC konnte natürlich nicht ermittelt werden.

Nach dem erneuten Eingeben von die Tokens funktioniert es dann wieder.

Falls ihr also weitere Mitschnitte benötigt kann ich ebenfalls unterstützen.

Ps: eingabe von die Tokens aktuell immer per Brave auf Android Handy.

andig, 2024-02-10:
Dann machen wir hier als doppelt zu. Weiter im anderen Issue.

closed 2024-02-10 by andig

RTTTC, 2024-02-15:
Still the same problem for me (Tesla and Citroen) using Windows/Edge and Mac/Safari
https://github.com/evcc-io/evcc/discussions/12265

andig, 2024-02-15:
Thats why https://github.com/evcc-io/evcc/issues/12015 is still open!