evcc-io/evcc #12127: Double Value since 0.124.0

michaestraub, 2024-02-07:
### Describe the bug

Hi,
I found out that Solar input is shown double as it is real.
Grid seams to be half od real value.
I get back on 0.123.9 and everything is shown fine.
Version 0.124.0 and 0.124.1, same
I use Fenecon 10 Home to fetch the data of Grid, Solar, Battery.

THX Michael

### Steps to reproduce

Its visibal, depending if you have a 123 or 124
[log.txt](https://github.com/evcc-io/evcc/files/14193935/log.txt)
evcc meter --log trace [log_0_123_9.txt](https://github.com/evcc-io/evcc/files/14193936/log_0_123_9.txt)
evcc meter --log trace [log_0_124_1.txt](https://github.com/evcc-io/evcc/files/14193937/log_0_124_1.txt)
[log_123_9.txt](https://github.com/evcc-io/evcc/files/14193938/log_123_9.txt)

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: 10.0.0.39 # .local suffix announces the hostname on MDNS
  port: 7070

#uri: 0.0.0.0:7070

tariffs:
  currency: EUR # (default EUR)
  grid:
    type: fixed
    price:  # [currency]/kWh

  feedin:
    type: fixed
    price:  # [currency]/kWh

log: trace
levels:
  cache: error

# unique installation id
plant: 

interval: 10s # control cycle interval

sponsortoken: 

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: true

meters:
- name: grid1
  type: template
  template: openems 
  usage: grid  
  host: 10.0.0.30  
  password: user  
  
- name: pv2
  type: template
  template: openems 
  usage: pv  
  host: 10.0.0.30  
  password: user  
  
- name: battery3
  type: template
  template: openems 
  usage: battery  
  host: 10.0.0.30  
  password: user  
  

chargers:
- type: template
  template: easee 
  user: 
  password: 
  charger: 
  timeout: 10s
  name: wallbox4
  authorize: true
- type: template
  template: easee 
  user: 
  password: 
  charger: 
  timeout: 10s
  name: wallbox5
  authorize: true

loadpoints:
- title: Aussen
  charger: wallbox4
  vehicle: smart
  mode: pv #off now minpv pv
  phases: 1
  mincurrent: 6
  maxcurrent: 16

- title: Garage
  charger: wallbox5
# vehicle:
  mode: pv
  phases: 0
# enable:
# threshold: 0
# delay: 1m
  mincurrent: 6
  maxcurrent: 16

site:
  title: HGW4a
  meters:
    grid: grid1
    pv:
    - pv2
    battery:
    - battery3
  #bufferSoc: 80
  #prioritySoc: 95

vehicles:
  - name: smart
    type: template
    template: smart
    title: SmartEQ
    user: 
    password: 
#  - name: mercedes
#    type: mercedes
#    title: C300e 
#    capacity: 
#    clientId: 
#    clientSecret: a
#    vin: 
#    cache: 2m
```

### Log details

```text
plz have a look at attachment
```

### What type of operating system are you running?

Linux

### Version

0.123.9 and 0.124.1

andig, 2024-02-07:
@premultiply related to https://github.com/evcc-io/evcc/pull/11479?

andig, 2024-02-07:
Please show

    evcc settings get MeterRef

I assume this might be another occurrence of https://github.com/evcc-io/evcc/issues/12130

premultiply, 2024-02-07:
> related to #11479?

Nein, das ist ja nur für Fehler mit 0 zu überschreiben.

Denke auch dass es an einer Dopplung (evcc.yaml+DB) liegt.

Ich denke wir brauchen irgendwie einen erweiterten Konfig-Dump der auch die DB-seitige Konfigurationen mit einschließt.

michaestraub, 2024-02-07:
Wenn ich unterstützen kann, gerne

andig, 2024-02-07:
Steht oben…

andig, 2024-02-08:
Gerne wieder auf machen wenn die Informationen vorliegen.

closed 2024-02-08 by andig

michaestraub, 2024-02-08:
Um welche Infos geht es denn? Kann ich leider nicht herauslesen.

andig, 2024-02-08:
Siehe https://github.com/evcc-io/evcc/issues/12127#issuecomment-1932322143

michaestraub, 2024-02-09:
Sorry, bin kein Developer, Das Problem wir nun wieder wichtig, wenn die Sonne wieder scheint und dann das Überschuss laden nicht passt.
Ich verstehe leider nicht ob das Problem behoben ist oder ob hier noch ein todo ist.

Danke für die Unterstützung, Gruß Michael

andig, 2024-02-09:
Du musst bitte die jetzt 2x angefragten Informationen liefern. Ohne gibts hier keine weitere Bearbeitung.

michaestraub, 2024-02-09:
das hier?
root@evcc:/tmp# evcc settings get Meter
Error: unknown command "settings" for "evcc"
Run 'evcc --help' for usage.
unknown command "settings" for "evcc"

michaestraub, 2024-02-09:
Ist es das was du benötigst?

andig, 2024-02-09:
Das sieht korrekt aus. Bitte nochmal

    evcc --log trace,db:error meter

Welcher Meßwert davon ist falsch (und mit Vorversion korrekt)?

andig, 2024-02-09:
> falsch

Blöd.

michaestraub, 2024-02-09:
Bitte was ist blöd?

michaestraub, 2024-02-09:
Sorry, bin kein Entwickler oder so, ich sehe nur das Ergebnis, das ich hier doppelte Werte habe, deshalb habe ich diesen Incident aufgemacht.

michaestraub, 2024-02-09:
https://github.com/evcc-io/evcc/discussions/12089

Hier findet man die entsprechenden Screenshots, hatte ich initial angehängt..

andig, 2024-02-09:
Ich frag nochmal: welcher Wert im "falsch" Log ist falsch? Ich kann nicht hellsehen!

michaestraub, 2024-02-09:
[image]
Hilft das?

andig, 2024-02-09:
Nein

michaestraub, 2024-02-09:
Ok, ich sehe das schein nichts zu werden, dann ist das für einen Leihen scheinbar nicht dir richtige Anwendung!
Ich habe das Problem seit der 124, dann bleibe ich eben auf 123 und hoffe daß irgendwann jemand das Problem hat, der es besser beschreiben kann.
Ich verstehe nicht warum ein Entwickler nicht den eingekreisten Wert den Variablen im Log zuordnen kann, ich kann es leider nicht.

michaestraub, 2024-02-09:
Müsste vielleicht der sein?
pv2
---
[http  ] TRACE 2024/02/09 13:05:51 GET http://10.0.0.30/rest/channel/_sum/ProductionActivePower
[http  ] TRACE 2024/02/09 13:05:51 {"address":"_sum/ProductionActivePower","type":"INTEGER","accessMode":"RO","text":"Total production; always positive","unit":"W","value":430}
Power: 430W

Wenn das die aktuelle Erzeugung ist?

michaestraub, 2024-02-14:
Guten Morgen, sind nun alle Infos da die benötigt werden um mein Problem zu verstehen?

Danke Michael

michaestraub, 2024-02-15:
Funktioniert wieder richtig in 0.124.2, danke