evcc-io/evcc #12128: Leistungsbegrenzung auf Grund des Autos

Wisley7, 2024-02-07:
Moin,
Da ich im Sommer bei PV Überschuss 1Phasig lade, und im Winter normal 3Phasig ergibt sich folgendes Problem:
Die Wallbox kann 32A aber der Tesla kann nur 11KW, egal ob 1/3 Phasig. Evcc denkt jetzt bei Zielladung über Tibber, das mit 32A 3Phasig geladen werden kann, und errechnet so die erforderten Zeitslots. 
Der Tesla kann aber halt nur 16A 3Phasig. 
Ich weiß, in der config kann der max. Strom vom Auto eingestellt werden, aber Gebrauch wird hier die max. Leistung. 
vielen Dank
Markus T.

naltatis, 2024-02-07:
Was spricht dagegen MaxCurrent beim Tesla auf 16A zu konfigurieren?

Wisley7, 2024-02-07:
Weil dann 1Phasig nicht mit 32a geladen werden kann

Wisley7, 2024-02-07:
Weil dann nicht mehr mit 32a 1Phasig geladen werden kann

naltatis, 2024-02-07:
32A@1p darfst du in Deutschland doch ohnehin nicht -> Schieflast (max 20A).
Unterschiedliche Current Limits pro Phasenanzahl hatten wir immer mal wieder in der Diskussion. Ist aber schwierig umzusetzen. @andig

Wisley7, 2024-02-07:
Schieflast ist nicht das Problem, da sämtliche anderen Verbraucher wie Wärmepumpe etc. auf den anderen beiden verteilt sind. Und die Wallbox bei mir auf 30A begrenzt ist. Und die 30A auch nur bei voller PV-Leistung geregelt werden.