evcc-io/evcc #12136: Wattpilot: evcc 124.1UI nicht erreichbar bzw. laden nicht möglich

Demophobie, 2024-02-07:
### Discussed in https://github.com/evcc-io/evcc/discussions/12061

<div type='discussions-op-text'>

<sup>Originally posted by **wiggal** February  5, 2024</sup>
Wenn ich auf eine aktuelle nightly (z.B. 0.124.0+1707099390) gehe ist die UI nicht mehr erreichbar (Fehler: Verbindung fehlgeschlagen), bei der 124.0 geht es noch.
hier meine Konfiguration:
[evcc.txt](https://github.com/evcc-io/evcc/files/14161785/evcc.txt)

Bin wieder auf die  124.0 zurück gegangen, aber vielleicht hilfts bei einer Fehlersuche.

</div>

wiggal, 2024-02-08:
Hier noch Infos aus der Diskussion:

Der Fehler tritt auch mit der neuen Version 124.1 auf.

Ich konnte das Problem noch eingrenzen.
Wenn beim Neustart von EVCC der Wattpilot lädt, dann startet die Oberfläche nicht mehr. 
Lädt der Wattpilot nicht, kann man die Oberfläche zwar aufrufen, aber EVCC lädt dann überhaupt nicht mehr, egal welchen Modus ich wähle.
Mit der Version 124.0 funktioniert es noch, hier startet die Oberfläche egal ob der Wattpilot lädt oder nicht.

Hier noch Infos (log: trace)
Version 124.1: Feb 06 09:48:23 bis Feb 06 09:53:59 (hier lädt EVCC nicht)
Version 124.0: Feb 06 09:54:04 bis ENDE (hier lädt EVCC ohne Probleme)
[Trace_evcc_124.txt](https://github.com/evcc-io/evcc/files/14204837/Trace_evcc_124.txt)

andig, 2024-02-08:
@mabunixda weiterer Bug in 1.6.4?

andig, 2024-02-08:
Das Log sieht ein wenig aus wie https://github.com/evcc-io/evcc/issues/12093.

@Demophobie kannst Du mal bitte analog https://github.com/evcc-io/evcc/issues/12093#issuecomment-1929337104 verfahren?

wiggal, 2024-02-08:
@andig,
da die Meldung und die Daten von mir stammen, hab ich mich mal angesprochen gefühlt.

Hab die 0124.1 nochmal installiert und mit `evcc --profile` gestartet.
Die Oberfläche startet (vermutlich weil der Wattpilot nicht am laden war), aber egal auf welchen modus ich schalte, es wird nicht geladen.
Hier das Logfile im Debugmodus und die goroutine
[debug.txt](https://github.com/evcc-io/evcc/files/14209385/debug.txt)
[goroutine.txt](https://github.com/evcc-io/evcc/files/14209386/goroutine.txt)

Ich hoffe es hilft

andig, 2024-02-08:
> aber egal auf welchen modus ich schalte, es wird nicht geladen.

Läuft denn die Anwendung weiter, also gibt es im Logfile kondinutierlich neue Meldungen? Oder stoppt das Logfile an der Stelle? Was sagt /api/health?

mabunixda, 2024-02-08:
Und was sagt parallel dazu die Wattpilot App ?

andig, 2024-02-08:
@mabunixda das sieht nach Deadlock aus:

```
1 @ 0x441d0e 0x453278 0x45324f 0x471d65 0x481ebd 0x16d5579 0x16d5548 0x16dad85 0x16da7fb 0x17c7dee 0x21135e8 0x2113525 0x21137fd 0x21190d6 0x212b5f5 0x212c590 0x218cbe5 0x4763e1
#	0x471d64	sync.runtime_SemacquireMutex+0x24					runtime/sema.go:77
#	0x481ebc	sync.(*Mutex).lockSlow+0x15c						sync/mutex.go:171
#	0x16d5578	sync.(*Mutex).Lock+0x58							sync/mutex.go:90
#	0x16d5547	github.com/mabunixda/wattpilot.(*Wattpilot).getRequestId+0x27		github.com/mabunixda/wattpilot@v1.6.4/wattpilot.go:237
#	0x16dad84	github.com/mabunixda/wattpilot.(*Wattpilot).sendUpdate+0x84		github.com/mabunixda/wattpilot@v1.6.4/wattpilot.go:629
#	0x16da7fa	github.com/mabunixda/wattpilot.(*Wattpilot).SetProperty+0x2da		github.com/mabunixda/wattpilot@v1.6.4/wattpilot.go:593
#	0x17c7ded	github.com/evcc-io/evcc/charger.(*Wattpilot).Phases1p3p+0x4d		github.com/evcc-io/evcc/charger/fronius-wattpilot.go:144
#	0x21135e7	github.com/evcc-io/evcc/core.(*Loadpoint).scalePhases+0x87		github.com/evcc-io/evcc/core/loadpoint.go:1023
#	0x2113524	github.com/evcc-io/evcc/core.(*Loadpoint).scalePhasesIfAvailable+0x64	github.com/evcc-io/evcc/core/loadpoint.go:1007
#	0x21137fc	github.com/evcc-io/evcc/core.(*Loadpoint).fastCharging+0x1c		github.com/evcc-io/evcc/core/loadpoint.go:1041
#	0x21190d5	github.com/evcc-io/evcc/core.(*Loadpoint).Update+0xab5			github.com/evcc-io/evcc/core/loadpoint.go:1614
#	0x212b5f4	github.com/evcc-io/evcc/core.(*Site).update+0x634			github.com/evcc-io/evcc/core/site.go:822
#	0x212c58f	github.com/evcc-io/evcc/core.(*Site).Run+0x2af				github.com/evcc-io/evcc/core/site.go:950
#	0x218cbe4	github.com/evcc-io/evcc/cmd.runRoot.func4+0x24				github.com/evcc-io/evcc/cmd/root.go:281

```

wiggal, 2024-02-08:
> > aber egal auf welchen modus ich schalte, es wird nicht geladen.
> 
> Läuft denn die Anwendung weiter, also gibt es im Logfile kondinutierlich neue Meldungen? Oder stoppt das Logfile an der Stelle? Was sagt /api/health?

Das Logfile bleibt solange stehen, bis ich auf eien anderen Modus umschalte:
```
...
[lp-1  ] DEBUG 2024/02/08 14:16:08 vehicle range: 90km
[lp-1  ] DEBUG 2024/02/08 14:16:08 available power 173W < 4140W min 3p threshold
[lp-1  ] DEBUG 2024/02/08 14:16:08 phase scale1p in 0s
[lp-1  ] DEBUG 2024/02/08 14:16:42 set charge mode: minpv
[lp-1  ] DEBUG 2024/02/08 14:16:42 pv timer reset
[lp-1  ] DEBUG 2024/02/08 14:16:42 pv timer inactive
-->> hier habe ich auf Schnell geklickt
[lp-1  ] DEBUG 2024/02/08 14:20:46 set charge mode: now
[lp-1  ] DEBUG 2024/02/08 14:20:46 phase timer inactive
```
wie komme ich zu "/api/health"?

wiggal, 2024-02-08:
> Und was sagt parallel dazu die Wattpilot App ?

[image]

mabunixda, 2024-02-08:
Kannst du die Umgebungsvariable `WATTPILOT_LOG` mal auf Trace setzen? Da sollte dann auf stderr Daten herauskommen.

mabunixda, 2024-02-08:
> @mabunixda das sieht nach Deadlock aus:

ja - da hab ich im [wattpilot/PR31](https://github.com/mabunixda/wattpilot/pull/31) schon etwas umgebaut, schau mir das aber in dem Zusammenhang nochmal genauer an. 
Ich werde auch die Log Infos nochmal umstellen/verbessern und mach morgen einen PR mit der neuen Version auf.

wiggal, 2024-02-08:
> Kannst du die Umgebungsvariable `WATTPILOT_LOG` mal auf Trace setzen? Da sollte dann auf stderr Daten herauskommen.

wie mach ich das?

brenndorfler, 2024-02-08:
Hallo,

ich bin auch betroffen (Wattpilot SW Version 40.7).
Wenn kein Auto am Wattpilot hängt, läuft die 0.124.1 übrigens.
Ich hatte #12093 hier schon einige Logs geposted.

Per Solar.wattpilot-App lässt sie die WB noch bedienen, auch wenn die GUI hängt.

mabunixda, 2024-02-08:
Hat wer evcc von euch als Container laufen? Es stünde [mabunixda/evcc:testing](https://hub.docker.com/layers/mabunixda/evcc/testing/images/sha256-b6c513057902801cc4825c36dfb3cc7c3dc7ef24e552b2bcd4f7b762c3998597?context=explore) bereit zu testen mit der 1.6.5 Preview Version

closed 2024-02-08 by andig

eilmi, 2024-02-09:
@mabunixda kann ich machen, wenn die neue tesla-command api implementiert wird

andrebeekhuis, 2024-02-10:
> Hat wer evcc von euch als Container laufen? Es stünde [mabunixda/evcc:testing](https://hub.docker.com/layers/mabunixda/evcc/testing/images/sha256-b6c513057902801cc4825c36dfb3cc7c3dc7ef24e552b2bcd4f7b762c3998597?context=explore) bereit zu testen mit der 1.6.5 Preview Version. Bei mir lief eine Vorversion schon diese Woche, aktuell ist diese auch schon drauf und lädt, evcc ist bedienbar

Bin/war auch betroffen. Die mabunixda/evcc:testing startet mit angeschlossenem Auto einwandfrei und lädt auch.