evcc-io/evcc #12159: Batteriesteuerung: Modbus Nachrichten werden auch nach dem deaktivieren weiter gesendet

dbhorst, 2024-02-09:
### Describe the bug

Für die Batteriesteuerung wird dem WR kontinuierlich (im Abstand watchdog/2) modbus Nachrichten mit dem minSoC Wert geschickt. Nachdem die Batteriesteuerung deaktiviert wird, sendet EVCC weiterhin minSoC modbus Nachrichten.

Ich nutze EVCC mit einem Kostal Plenticore, KSEM, BYD Batterie und go-e Charger. Es kann sein, dass dieses Problem aber auch bei anderen WR auftritt. 

Vollständige Log: 
[evcc.log](https://github.com/evcc-io/evcc/files/14219699/evcc.log)

### Steps to reproduce

1. Konfiguration für den plenticore vornehmen und logging für modbus auf trace stellen
2. Batteriesteuerung einschalten
3. Eine Minute warten
4. Batteriesteuerung ausstellen
5. Eine Minute warten
6. modbus Nachrichten erscheinen immer noch im Log (suchen nach modus Adresse 04 12 für den minSoC)

### Configuration details

```yaml
meters:
- type: template
  template: kostal-plenticore 
  id: 71  
  host: 10.11.11.21  
  port: 1502  
  usage: pv  
  modbus: tcpip 
  endianness: little
  capacity: 7.68
  minsoc: 11 
  maxsoc: 100 
  watchdog: 60s 
  name: pv1
- type: template
  template: kostal-plenticore 
  id: 71  
  host: 10.11.11.21  
  port: 1502  
  usage: battery  
  modbus: tcpip
  endianness: little
  capacity: 7.68 
  minsoc: 11
  maxsoc: 100
  watchdog: 60s
  name: battery2
- type: template
  template: kostal-ksem-inverter 
  id: 71  
  host: 10.11.11.21  
  port: 1502  
  usage: grid  
  modbus: tcpip  
  name: grid3

chargers:
- type: template
  template: go-e 
  host: 10.11.11.23  
  name: wallbox6

vehicles:
- type: template
  template: renault 
  title: ZOE  
  user:   
  password: 
  vin:   
  capacity: 40  
  phases: 3  
  icon: car  
  cache: 30m  
  mode: pv 
  minCurrent: 6  
  maxCurrent: 16
  alternativewakeup: false  
  name: ev4
- type: template
  template: kia
  title: Kia e-Soul
  user: 
  password: 
  vin: 
  capacity: 64
  language: de
  phases: 1
  icon: car
  cache: 15m
  mode: pv
  minCurrent: 6
  maxCurrent: 16
  name: ev5

loadpoints:
- title: Garage
  charger: wallbox6
  vehicle: ev4
  mode: pv
  phases: 1
  mincurrent: 6
  maxcurrent: 16

site:
  title: Mein Zuhause
  meters:
    grid: grid3
    pv:
    - pv1
    battery:
    - battery2
  residualPower: 200
```

### Log details

```text
[site  ] DEBUG 2024/02/08 08:52:00 set battery discharge control: true
...
[modbus] TRACE 2024/02/08 08:52:07 modbus: send 00 d4 00 00 00 0b 47 10 04 12 00 02 04 00 00 41 30
[modbus] TRACE 2024/02/08 08:52:08 modbus: recv 00 d4 00 00 00 06 47 10 04 12 00 02
[site  ] DEBUG 2024/02/08 08:52:08 set battery mode: normal
...
[site  ] DEBUG 2024/02/08 08:58:16 set battery discharge control: false
...
[modbus] TRACE 2024/02/08 09:00:31 modbus: send 02 c4 00 00 00 0b 47 10 04 12 00 02 04 00 00 41 30
[modbus] TRACE 2024/02/08 09:00:31 modbus: recv 02 c4 00 00 00 06 47 10 04 12 00 02
```

### What type of operating system are you running?

Linux

### Version

0.124.1

andig, 2024-02-09:
> Es kann sein, dass dieses Problem aber auch bei anderen WR auftritt.

Was ist denn das konkrete Problem? Modbuskommunikation ist erstmal keines!

dbhorst, 2024-02-09:
Das Problem ist: Wenn ich im EVCC die Batteriesteuerung über die UI deaktiviere, werden trotzdem weiterhin modbus Nachrichten mit dem minSoC an den WR geschickt.

andig, 2024-02-09:
...und was ist das Problem damit? Modbuskommunikation ist erstmal keines!

dbhorst, 2024-02-09:
Wenn ich die Batteriesteuerung deaktiviere würde ich erwarten, dass EVCC die Batterie auch nicht mehr steuert. Aktuell werden aber noch modbus Nachrichten gesendet in denen der minSoC aktiv gesetzt wird.
Natürlich sind da noch mehr Modus Nachrichten, mit denen sich EVCC Infos aus dem Plenticore holt, aber er sollte dann keine Werte (für die Batteriesteuerung) mehr schreiben.
Wenn ich EVCC mit deaktivierter Batteriesteuerung starte werden auch keine Nachrichten geschrieben, bis ich die Batteriesteuerung einschalte.

andig, 2024-02-09:
Verstanden. Kommt ins Backlog, Prio gering.

closed 2024-02-09 by andig

menkej, 2024-02-11:
> Verstanden. Kommt ins Backlog, Prio gering.

Nachdem die verknüpften Tickets jetzt alle geschlossen sind: wo findet sich denn das Backlog, wenn nicht in den Issues? Bevor es weiter neue Tickets im Kontext Kostal/Batteriesteuerung gibt. Wir haben ja aktuell alleine 4 Discussions dazu offen. Da wird es langsam unübersichtlich. ;-)

andig, 2024-02-11:
Es gilt das in https://github.com/evcc-io/evcc/issues/12159#issuecomment-1935583008 gesagte.