evcc-io/evcc #12164: Creating vehicle ev failed: cannot create vehicle type 'template': cannot create vehicle type 'tesla': cannot get vehicles: 412 Precondition Failed

KevinGruber, 2024-02-09:
### Describe the bug

I have the same problem as in https://github.com/evcc-io/evcc/issues/11857

But I am using valid tokens and the normal Tesla template.

I found out that the vehicle list endpoint gives now the error.
{
    "response": null,
    "error": "Endpoint is only available on fleetapi. Visit https://developer.tesla.com/docs for more info",
    "error_description": ""
}

the vehicle detail endpoints still work.

that's why it was working before as it still had the list of vehicles in cache and the detail endpoint worked anyways.

So I see two possibilities either mix the endpoints or make the new Fleet API also capable of setting params.

What is the plan for the future?

BR,
Kev

### Steps to reproduce

1. start evcc
2. get error

### Configuration details

```yaml
"old" Tesla template
```

### Log details

```text
creating vehicle ev failed: cannot create vehicle type 'template': cannot create vehicle type 'tesla': cannot get vehicles: 412 Precondition Failed
```

### What type of operating system are you running?

Linux

### Version

124.1

andig, 2024-02-09:
Tesla is stopping support of owners api.

closed 2024-02-09 by andig

KevinGruber, 2024-02-10:
Hi Andig,

You still have template: tesla  in the application and in the documentation that obviously doesn't work anymore.

At the same time you say use the Tesla-command template which can't be used as it doesn't allow to set the limit yet

So what is your suggestion?

Is there the plan to make it work through the proxy soon? I saw the draft PR.

One temporary option in my opinion would be to either put in the vehicle id's manually for now and still use the owners api in the first case or combine it with the fleet api and get the ids from there.

BR,
Kev

andig, 2024-02-11:
We'll need to wait for https://github.com/evcc-io/evcc/issues/11846 being implemented.