evcc-io/evcc #12165: Batteriesteuerung Plenticore: Modbus Nachrichten blockieren interne Plenticore Batteriesteuerung

dbhorst, 2024-02-09:
### Describe the bug

Wenn in EVCC die Batteriesteuerung eingeschaltet ist, wird im im "battery mode normal" kontinuierlich der minSoC auf 20% gesetzt. Dies führt dazu, dass der Plenticore immer im Modus "externe Steuerung" ist und die Hausbatterie nach einem schnell-Ladevorgang nicht mehr automatisch mit Überschuss geladen oder entladen wird. D.h. der Plenticore verlässt sich in diesem Modus darauf, dass EVCC die lade/entlade Steuerung übernimmt (siehe Doku unten), was EVCC sicherlich nicht soll.

Lösung:
EVCC sollte den minSoC nur im "battery mode hold" schreiben. Ein schreiben des minSoC im "battery mode normal" ist nicht notwendig, da der Plenticore nach 60 Sekunden (default, kann auch kürzer eingestellt werden) automatischen auf den internen Modus zurückfällt und den minSoC selbst zurücksetzt.

Hier sieht man wie der Strom nach beenden des schnell laden (battery mode normal) eingespeist wird, anstatt in der Batterie gespeichert zu werden:

[image]

Ausschnitt aus der Plenticore Doku:

[image]

Abgrenzung zu https://github.com/evcc-io/evcc/issues/12159 ist, dass dieser Fehler bei aktiver EVCC Batteriesteuerung und nur beim Plenticore auftritt.

### Steps to reproduce

1. Konfiguration für den plenticore vornehmen
2. Batteriesteuerung einschalten
4. Auto schnell laden
5. Schnell laden beenden
6. Die Hausbatterie wird nicht mehr geladen/entladen

### Configuration details

```yaml
meters:
- type: template
  template: kostal-plenticore
  id: 71
  host: 10.11.11.21
  port: 1502
  usage: pv
  modbus: tcpip
  endianness: little
  capacity: 7.68
  watchdog: 60s
  name: pv1
- type: template
  template: kostal-plenticore
  id: 71
  host: 10.11.11.21
  port: 1502
  usage: battery
  modbus: tcpip
  endianness: little
  capacity: 7.68
  watchdog: 60s
  name: battery2
- type: template
  template: kostal-ksem-inverter
  id: 71
  host: 10.11.11.21
  port: 1502
  usage: grid
  modbus: tcpip
  name: grid3
```

### Log details

```text
Batterie wird nicht geladen obwohl Überschuss zur verfügung steht:
[site  ] DEBUG 2024/02/09 12:10:18 battery soc: 15%
[site  ] DEBUG 2024/02/09 12:10:18 battery power: 8W
[site  ] DEBUG 2024/02/09 12:10:18 grid power: -210W
[site  ] DEBUG 2024/02/09 12:10:18 site power: -2W

Der minSoC wird auf 20% gesetzt. Woher kommt dieser Wert? Im kostal-plenticore template ist kein default für minSoC konfiguriert.
[modbus] TRACE 2024/02/09 12:10:21 modbus: send 0e 9a 00 00 00 0b 47 10 04 12 00 02 04 00 00 41 a0
[modbus] TRACE 2024/02/09 12:10:21 modbus: recv 0e 9a 00 00 00 06 47 10 04 12 00 02
```

### What type of operating system are you running?

Docker container

### Version

0.124.1

menkej, 2024-02-09:
Woher der default kommt, hab ich auch noch nicht gefunden. Du kannst in der evcc.yaml aber einen MinSoc für die Batterie setzen, dann wird dieser genommen. Fehlt noch in der Doku.

dbhorst, 2024-02-09:
Danke, es ist mir bekannt, dass ich einen konfigurieren kann. 
Ich habe hier bewusst keinen Wert gesetzt, weil ich gar nicht möchte, das EVCC einen minSoC setzt (außer im "battery mode hold", aber da wird ja dann der aktuelle SoC gesetzt und nicht der minSoC).

andig, 2024-02-09:
Duplikat zu https://github.com/evcc-io/evcc/issues/12159

closed 2024-02-09 by andig