evcc-io/evcc #12166: [UX] Device Configuration: Generisch Wechselrichter und Hybridwechselrichter als erstes nennen

StefanSchoof, 2024-02-09:
**Is your feature request related to a problem? Please describe.**

Es kommt öfter vor, dass bei der Einrichtung von Geräten statt, den generische Wechselrichter oder Hybridwechselrichter ein anderes Gerät wählen, dass in der Liste ist. Wenn ein Nutzer zum Beispiel ein Fronius Symo hat, er aber das Fronius GEN24 Template wählt, da ihm nicht bewusst ist das die generische Unterstützung funktioniert.

**Describe the solution you'd like**
Ähnlich wie bei den Fahrzeugen, könnte man Generischen Wechselrichter und Generischen Hybridwechselrichter als erstes nennen. Ich denke es wäre auch gut statt nur "Wechselrichter" das in der UI "Generischer Wechselrichter" zu nennen. Damit ist es zumindest offensichtlicher das man das Template probieren könnte und außerdem wirkt "Wechselrichter" zwischen den ganzen Produktnamen deplatziert.
[image]
[image]

**Describe alternatives you've considered**
Weiter den Leuten in den Diskussionen helfen, das richtige Template zu nehmen ;)

**Additional context**

naltatis, 2024-02-10:
Ja definitiv. Hier sollten wir auch den Namen ändern. Inhaltlich ist das ja Sunspec. Also sowas wie: "Generischer Wechselrichter (SunSpec)". Ich würd vorschlagen den abgesetzt am Ende der Liste (also wie vehicles aber anders herum) zu platzieren. Annahme ist: der ist eher die "Last Resort Option" weil wir sehr viele WRs direkt unterstützen. Bei den Fahrzeugen gibts viele Hersteller / Fahrzeuge die keine API haben. Daher wird das "Generische Fahrzeug" öfters verwendet.

StefanSchoof, 2024-02-10:
Da die Liste so lang ist, sehe beim Ende das Problem, dass wenn ein Nutzer ein Wechselrichter von einem Hersteller am Anfang des Alphabets besitzt, nicht bis zum Ende scrollt und die Option findet. Dagegen ist allgemein bekannt, dass man lieber eine spezifischere Option nimmt, wenn vorhanden und man sieht ja auf dem ersten Blick, dass es Templates für verschiedene Hersteller gibt.

premultiply, 2024-02-11:
So ist es.
Steht der Eintrag am Ende der Liste werden wir stärker mit Anfragen überhäuft werden warum Modell XY nicht in der Liste ist oder irgendwas mit dem für den Fall falschen Hersteller-Template XYZ angeblich nicht funktionieren würde.

Erst als der Passus über SunSpec ganz oben auf der Geräteseite in die Doku kam wurde es spürbar weniger.

Letztendlich bin ich aber eigentlich auch ein großer Fan der Funktionalität und deren weiterem Ausbau von `evcc detect`.
Idealerweise würden die meisten Geräte und die dazu passenden Templates über eine bestimmte Signatur weitgehend automatisch erkannt.

naltatis, 2024-02-11:
Gut. Dann packen wir den Eintrag an den Anfang. Der Name "Generischer Wechselrichter (SunSpec)" passt für euch?

StefanSchoof, 2024-02-11:
Ja, denke da können Laien was drunter verstehen und wer weiß was SunSpec ist kann damit auch arbeiten

closed 2024-02-15 by andig