evcc-io/evcc #12168: Battery percentage calculation not correct when using multiple batteries

umrath, 2024-02-09:
### Describe the bug

When a setup with 2 independent storage batteries are present, evcc calculates the average soc of the batteries and displays those in the interface.
But this gives a wrong impression about the real state of the batteries, as the capacities are not accounted for.

Example:
Two batteries, one with 10 kWh capacity, one with 20 kWh capacity.
If the first is charged to full and the second is empty, evcc would display 50% SoC.
The real capacity should be 33%, though, as the batteries (in sum) are charged to 1/3 of their combined capacity.

### Steps to reproduce

1. add to batteries with differenct capacities to your setup
2. charge both batteries to an individually different soc
3. observer the soc displayed by evcc

### Configuration details

```yaml
most likely not relevant
can provide later if really required
```

### Log details

```text
most likely not relevant
can provide later if really required
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

latest

andig, 2024-02-09:
Mhhm. Can you spot the bug? 

```go
			// battery soc and capacity
			var batSoc, capacity float64
			if meter, ok := meter.(api.Battery); ok {
				batSoc, err = soc.Guard(meter.Soc())

				if err == nil {
					// weigh soc by capacity and accumulate total capacity
					weighedSoc := batSoc
					if m, ok := meter.(api.BatteryCapacity); ok {
						capacity = m.Capacity()
						totalCapacity += capacity
						weighedSoc *= capacity
					}

					site.batterySoc += weighedSoc
					if len(site.batteryMeters) > 1 {
						site.log.DEBUG.Printf("battery %d soc: %.0f%%", i+1, batSoc)
					}
				} else {
					site.log.ERROR.Printf("battery %d soc: %v", i+1, err)
				}
			}
		}

		site.publish(keys.BatteryCapacity, totalCapacity)

		// convert weighed socs to total soc
		if totalCapacity == 0 {
			totalCapacity = float64(len(site.batteryMeters))
		}
		site.batterySoc /= totalCapacity

		site.log.DEBUG.Printf("battery soc: %.0f%%", math.Round(site.batterySoc))
		site.publish(keys.BatterySoc, site.batterySoc)
```

premultiply, 2024-02-09:
Missing log & config.

Did you set `capacity` on the individual batteries?

closed 2024-02-09 by premultiply

andig, 2024-02-09:
> Two batteries, one with 10 kWh capacity, one with 20 kWh capacity.

According to this the capacity is set.

umrath, 2024-02-09:
Actually, no. The capacity is not set.
How do I set it? I cannot find a reference in the documentation for the batteries. The same way as for vehicles?
(I was under the impression that this is read from the device.)

umrath, 2024-02-09:
Ok, looks like it's set the same way as for cars.
At least I did not see any error message when simply adding "capacity: 9.6" for one of the batteries.

I still have a deviation from what I calculate manually (51 vs. 54 %), though. But it looks like it's closer.