evcc-io/evcc #12186: Add Philips Hue Bridge for use of smart plug

p7lejo, 2024-02-10:
**Is your feature request related to a problem? Please describe.**
I'm frustrated when I can not use my smart plugs with evcc

**Describe the solution you'd like**
Electrical heater is powered by smart plug from Philips

**Describe alternatives you've considered**
I have to use older plugs, which are wifi based. There are locations without wifi but ZigBee connected (Hue Bridge and mesh)
I have to use work arrounds, like create dummy loadpoints and charger and translate commands via smart home integrations (fhem, openhab, homeassist)

**Additional context**
https://www.philips-hue.com/de-de/p/hue-smart-plug-weiss/8719514342309

andig, 2024-02-10:
Is there a Hue API? I guess @thierolm could be interested in helping out if someone would donate testing hardware.

p7lejo, 2024-02-10:
natürlich gibt es das, auch gut beschrieben: https://developers.meethue.com/develop/get-started-2/
Allerdings sind die Steckdosen (meines Wissens) leider ohne Strommessung. Was in Summe dann vielleicht doch keine so gute Idee ist. Für eine Heizpatrone kann man den Stromverbrauch natürlich konstant setzen, die Optionen bietet evcc ja heute schon.

thierolm, 2024-02-11:
@andig ich kann gerne für die Hue API ein Charger Modul bauen.
Im Kontext der Tapo-Steckdosen PR hattest allerdings einmal erwähnt, das wir Steckdosen ohne Strommessung nicht supporten.
@p7lejo Gibt es denn ein Hue Steckdosenmodell MIT Strommessung?

p7lejo, 2024-02-11:
Nein, gibt es meines Wissens nicht. Ich hatte gehofft, es gibt schon (versteckt) eine Funktion fertig. Mit Strommessung ist tatsächlich sehr viel besser, ich denke, ich besorge mir eine Alternative (DECT von AVM wird ja z.B. unterstützt und hat auch andere Reichweite als WiFi). Ist in jedem Fall auch einfache Lösung.

closed 2024-02-11 by andig