evcc-io/evcc #12191: Mercedes car Integration via unofficial mobile api

dm82m, 2024-02-11:
**Is your feature request related to a problem? Please describe.**
A lot of Mercedes drivers are unhappy about the fact that evcc offers no direct integration of their car.

**Describe the solution you'd like**
There are several projects like home assistant, iobroker, openhab, … who are using the unofficial mobile app api to integrate the Mercedes cars into their home automation system. One of the developers wrote a how to but I am no go dev so it is not possible to implement that on my own. But maybe someone here wants to do that: https://community.home-assistant.io/t/mercedes-me-component/41911/1714

**Describe alternatives you've considered**
My alternative at the moment is using the data out of my home assistant instance. This works for me but other people not having a home automation system are in the need of installing a whole home automation system only for the Mercedes soc und distance.

andig, 2024-02-11:
Happy to take a PR. From the linked description it is not entirely clear to me what the api really is.

ReneNulschDE, 2024-02-11:
Hi @andig ,

before I start the coding, let me explain what I plan todo. I'm new to your platform and want to make sure that I'm not running in the wrong direction or the PR gets to big.

High level idea for V1:

Configuration: 
RefreshToken, Vin, Account 
- The procedure to get RT is Pin based. I can add a command line tool to get the RT
- One account can have multiple cars, Account is used to keep the RT and AT synchron

Identity: 
Use the RefreshToken (RT) to get the AccessToken (AT), store the new RT via settings.SetJson, Lifetime of the AT is 2h.

Api: 
Check if AT is still valid, call the Rest API with AccessToken+Vin --> Result is a Protobuf Message --> Unmarshal the message and store the Soc, Max-Soc, Odo, Location, ...

Open questions so far:
- Can I create multiple cars with one config entry? If not, how can I make sure that only one car updates the AT.
- How are units handled? Is everything stored in Km and the system handles the conversation?

V2:
- Create a background websocket to send the command to change the Max-Soc or activate defined charging programs.

andig, 2024-02-12:
Sounds great!

> store the new RT via settings.SetJson

Put RT into config- vehicles shouldn't even know there's permananent storage (yeah, broke that rule for tesla command, but had to since even the refresh token is short-lived)

> If not, how can I make sure that only one car updates the AT.

Create singleton instance, see tesla command. You'll need a key for doing so. TC does by looking at the token's subject.

> Is everything stored in Km

yes

andig, 2024-02-14:
@dm82m does that make sense and it it enough to get you started?

dm82m, 2024-02-14:
As far as I understood @ReneNulschDE will step into it, I will maybe join if that is useful.

ReneNulschDE, 2024-02-14:
I'm on it - had to change my car simulator a little bit to support this scenario and have to learn evcc and golang :-)

[image]
[image]

Now I have to solve the token refresh before I switch to the mb environment.

@dm82m : Happy to handover it to you afterwards.

markus76, 2024-02-19:
Hi @ReneNulschDE 

if you have anything to test, I will do...
Same like in Home Assistant ;-)
MB Car integration would be nice... properly with an automatic identification if its possible

dm82m, 2024-02-29:
Works partially with 0.126.6. Should work fully with 0.126.7 and works with current nightly! Thank you very much @ReneNulschDE for adding this to evcc! Much appreciated!

markus76, 2024-02-29:
Great Thank you very much @ReneNulschDE

sebischnee, 2024-04-16:
es fuktioniert ja schon perfekt seit 0.124.x und auhc mit der aktuellen 0.125.0! Danke dafür!

leider zeigt es den Soc immer nu an, wenn ich das auto grad angesteckt habe. geht das auch, dass der soc immer angezeigt wird?