evcc-io/evcc #12192: Planner: UI update - Nachträgliches anpassen

vkr2791, 2024-02-11:
### Describe the bug

ggf. ist das auch eher ein Feature-Request ...

- Planner einstellen und aktivieren
- Planner erneut öffnen und z.B. die Abfahrtszeit anpassen 
=> dann erscheint "Anwenden"
[image]

Den Schalter "Atkiv" finde ich noch transparent und nachvollziehbar.
Aber wenn ich bei einem aktivierten Plan etwas ändere sollte die Änderung beim Schließen ohne ein "Anwenden" übernommen werden, das wäre doch intuitiv.
Alternativ könnte man auch eine Hinweis-Meldung/Frage beim Schließen der Eintstellungen anzeigen:
z.B. "Sie haben Änderungen am Plan vorgenommen und nicht angewendet. 
Änderungen übernehmen? [Ja|Nein]" 
Intuitiver finde ich aber das direkte übernehmen von Änderungen bei einem aktiven Plan ohne "Anwenden".

Weiterhin sollten diese fahrzeugspezifischen Einstellungen (Planner) zusätzlich und intuitiver neben dem Fahrzeug erreichbar sein (siehe Screenshot)
und ein aktivierte Plan sollte dann auch in schwarzer Schrift erscheinen. 
Wenn kein Plan aktiviert ist, dann gerne in Grau:
[image]

### Steps to reproduce

siehe Description

### Configuration details

```yaml
für diesen Issue nicht notwendig
```

### Log details

```text
für diesen Issue nicht notwendig
```

### What type of operating system are you running?

Linux

### Version

0.124.1

andig, 2024-02-11:
@naltatis können wir noch klarer machen, dass der angepasste Plan nicht aktiv ist?

naltatis, 2024-02-11:
Sind ja zwei unterschiedliche Themen:

1. Fahrzeugspezifisches (Min Soc, ...) am Fahrzeug speichern. Ja, das ist sinnvoll und haben wir auf dem Plan. Wird früher oder später kommen. Daher würd ich das Thema jetzt hier in der Diskussion mal ausklammern.

2. Ändern eines aktiven Plans

> Intuitiver finde ich aber das direkte übernehmen von Änderungen bei einem aktiven Plan ohne "Anwenden".

In der Tat war das in der ersten Version dieser Maske genau so. Das Problem/Feedback war da, dass es nicht immer wünschenswert ist die Änderung sofort aktiv zu haben. Das führt im Hintergrund ja dazu, dass sich der Planungsalgorithmus schon auf die neuen Werte anpasst und ggf. schon eine Ladung startet oder abbricht. Daher haben wir bewusst die "Plan Vorschau" wieder eingeführt.

Einen Hinweis der beim Schließen eines veränderten, aber ungespeicherten Plan kommt, kann ich mir gut vorstellen. Das wäre zumindest erstmal einfach umsetzbar und verhindert, dass man es vergisst.

naltatis, 2024-02-11:
> Sollte der Schalter "Aktiv" im Planner ggf. den Modus "PV" direkt mit aktivieren??? Würde man das erwarten???

Würde ich erstmal nicht machen. Das wäre mMn. ein Seiteneffekt den man nicht unbedingt erwarten würde. Den Hinweis, dass der Plan nur im PV-Modus aktiv ist geben wir im Dialog (siehe dein erster Screenshot) ja. Das muss man natürlich einmal gelernt/verstanden haben. Ich glaub aber, dass es danach dann unproblematisch ist und dass das subtile Ausgrauen genug Hinweis gibt, dass hier etwas gerade nicht aktiv ist. Beim Klick sieht man dann ja den erklärenden Hinweistext im Layer wieder.

vkr2791, 2024-02-11:
> Den Hinweis, dass der Plan nur im PV-Modus aktiv ist geben wir im Dialog (siehe dein erster Screenshot) ja. Das muss man natürlich einmal gelernt/verstanden haben. Ich glaub aber, dass es danach [...]

@naltatis Danke für die Erklärungen, und ich bin mal gespannt ob es meine Frau (absoluter non-techi) so intuitiv hinbekommt wie du sagst :-)
Starten wir mal so und warten ab ...

naltatis, 2024-02-11:
> ich bin mal gespannt ob es meine Frau (absoluter non-techi) so intuitiv hinbekommt wie du sagst :-)

Ich bin gespannt, was deine Tests ergeben :D Ist, halt immer der Spagat aus UI simpel halten. Aber trotzdem genug Infos geben, dass es im Alltag gut funktioniert.

closed 2024-03-09 by andig