evcc-io/evcc #12194: Tesla token keeps expiring

umrath, 2024-02-11:
### Describe the bug

Whenever I restart evcc after it has run for some time, I am greeted with a token expired error message (Tesla).
This did not happen when using the old syntax - which is sadly disabled for my account/car already as it seems.

### Steps to reproduce

1. Add a Tesla to evcc via the new tesla-command syntax
2. generate a token via the dedicated evcc page
3. restart evcc
4. watch the car being recongnized properly
5. leave evcc running for some time (days?)
6. restart evcc
7. watch token error in log/car unrecognized

### Configuration details

```yaml
# vehicle definitions
vehicles:
  - name: Tesmeralda
    type: template
    template: tesla-command
    title: Tesmeralda
    accessToken: <redacted>
    refreshToken: <redacted>
    vin: <redacted>
    capacity: 77 # Akkukapazität in kWh
    mode: pv # enable PV-charging when vehicle is identified
```

### Log details

```text
cannot run this command in Home Assistant contained (at least I don't know how to)
(sadly I forgot to capture the log output in Home Assistant before fixing it)
```

### What type of operating system are you running?

Linux

### Version

0.124.1

andig, 2024-02-11:
Nothing to do without full trace log for tesla-command.

closed 2024-02-11 by andig

umrath, 2024-02-11:
How am I supposed to provide this, if the environment is not capable to do that?

andig, 2024-02-11:
As usual: choose a different environment. Nothing forces you to run evcc on HA. That's why the issue template mentions that we do not support issues on HA platform. It severely lacks the means to get meaningful data for analysis.

I'm locking this threads before this derails into another OT discussion.