evcc-io/evcc #12213: Seat Mii Electric Car Integration Request

FlorianPanten, 2024-02-12:
I can see that VW We Connect is already integrated. Seat Connect has a very similar (almost identical) UI as VW We Connect. I would bat that the netcode is probably the same as well. It would mean alot to me if this car would be added.

StefanSchoof, 2024-02-12:
There is a seat support https://docs.evcc.io/en/docs/devices/vehicles/#seat. Does none of those work for the Mii?

FlorianPanten, 2024-02-12:
Let me try them but I am pretty sure they are somethign different

FlorianPanten, 2024-02-12:
Well I cannot login into the Cupra Connect App. I think Cupra connect is an own software and Seat Mii Connect is a copy or a fork of the We Connect software

mucki12, 2024-02-12:
Why don't you just test the SEAT template in evcc? According to the discussions, several people use an MII...

FlorianPanten, 2024-02-12:
Oh okay, I am still in setup and there are other parts im having trouble with so I can only test it once I have set it up correctly. Ill give you a heads up as soon as I get it working

gr5604, 2024-02-12:
I am actually using a Seat Mii in EVCC with the "seat" template.

extract of EVCC.yaml ( vehicles part only )

...
vehicles:
- type: template
  template: seat
  title: Seat Mii electric
  user: your user
  password: your password
  vin: your vin
  capacity: 36.8
  phases: 2 # ev phases (default 3)
  mode: pv
  name: ev4
...

Hope this helps.