evcc-io/evcc #12217: Zusammenfassen der Ladungen pro Fahrzeug pro Tag

kafetzke, 2024-02-12:
Aktuell wird nach jedem Neuverbinden eines Fahrzeuges ein neuer Ladevorgang protokolliert. 

Klassischer Usecase: das Auto ist den kompletten Tag angeschlossen, Mittags ist man mal eben kurz zur Kita und steckt dann erneut an. Es werden zwei unterschiedliche Ladevorgänge erstellt.

Es wäre sehr cool, wenn es die Möglichkeit gäbe, alle Ladevorgänge eines Fahrzeuges am gleichen Tag zusammenfassen zu lassen. Dies würde Auswertungen "was habe ich tageweise in meine unterschiedlichen Fahrzeuge geladen" erheblich vereinfachen (und viel Excel-Pfuscherei vermeiden) :-)

andig, 2024-02-12:
Ich würde nichts zusammen fassen, wir sollten aber sehrwohl schauen, ob sich ein angefangener Ladevorgang nicht fortsetzen liesse. Wie würdest Du mehrtägige Ladungen behandeln (…ohne sie nachts zwangszutrennen)?

kafetzke, 2024-02-12:
Moin,
mathematisch korrekt wäre wohl eine Betrachtung "wie bei einer Trennung", also alles bis Mitternacht / ab Mitternacht. Insb. bei einem Wechsel auf den 1. des Folgemonats würde dies das Ergebnis auf diese Weise nicht beeinflussen.

andig, 2024-02-12:
> Wie würdest Du mehrtägige Ladungen behandeln (…ohne sie nachts zwangszutrennen)?

naltatis, 2024-02-13:
Grafische Auswertung der Ladevorgangsdaten werden wir auf jeden Fall machen. Also ne Tages/Monats/Jahresansicht auf unterschiedliche Werte und Entitäten (Fahrzeug / Ladepunkt). Die Datengrundlage dafür haben wir inzwischen ja. Muss "nur noch" schön visualisiert werden. Ich glaub, wir haben es noch nicht als Issue hier in der Liste. Daher würde ich jetzt diese Anfrage mal dazu umformulieren.

**Thema Tagesgrenze:** Aktuell ist bei uns immer der Ladebeginn (Anstecken) ausschlaggebend. Wir brechen Ladevorgänge bewusst nicht künstlich auf, um andere Probleme (Anzahl Ladevorgänge, Zeitzonen, Ladevorgänge ohne Energie, ...) zu vermeiden. Das würde ich, gerade im Kontext Ladevorgänge auch so weiter machen. IdR. kann man mit dieser Unschärfe sicher leben. Wer es heute schon genauer braucht, kann/muss sich etwas mit Influx/Grafana bauen. Die Daten veröffentlichen wir ja.

**Theme rein zeitbasierte Auswertungen:** Im Bereich Schaltsteckdosen und Wärmeerzeugung passt der Begriff Ladevorgang ja heute schon nicht. Wir haben da Workarounds eingebaut, um bspw. durch aktives An/Ausschalten einzelne Ladevorgänge zu erzeugen. Da brauchen wir sicher noch ne bessere Lösung. Also bspw. ein 15min Speichern der jeweiligen Energiemengen pro Ladepunkt. Kann dann auch Grundlage für Zeitgraphen werden, wie man sie aus vielen anderen Energiemanagementsystemen kennt. Das ist eine größere Baustelle, die wir evtl. irgendwann angehen. Unser Fokus ist ja Fahrzeuge Laden. Da haben wir noch ein paar andere Themen, die aktuell wichtiger sind. Daher würde ich das rein zeitbasierte hier jetzt auch erst mal ausklammern, aber nicht ausschließen.

kafetzke, 2024-02-13:
Moin @naltatis ,

gute Idee! Vielen Dank...

andig, 2024-02-14:
Issue highjacking. Hier gings um Zusammenfassung. Ich würde mal zu machen da OT.

closed 2024-02-14 by andig