evcc-io/evcc #12220: Sofar HYD: Battery Control

premultiply, 2024-02-12:
https://www.photovoltaikforum.com/thread/221806-sofar-hyd-mit-evcc-akkuentladung-verhindern/?postID=3598510#post3598510

in Template übersetzen

cschlipf, 2024-03-29:
@premultiply Lass mich wissen, wenn Du das angehst. Habe mittlerweile in der Integration einiges geändert. Siehe auch: https://homeassistant-solax-modbus.readthedocs.io/en/latest/sofar-energy-storage-modes/

andig, 2024-09-14:
@cschlipf wäre es möglich hier kurz die relevanten Register und Werte zu beschreiben? Dann ist das schnell gemacht. Danke!

cschlipf, 2024-09-14:
Gerne.

Um den Passive Mode zu setzen ist dieses Register zu schreiben:
[image]

Mit dem Wert 3 geht es in den Passive Mode. Wenn man die Batterie wieder aktiviert, sollte das Register wieder zurück auf 0 (Self-Use) gesetzt werden.

Nun könnte es noch sein, dass im Passive Mode die Werte nicht stimmen. Dazu müssen diese Register zusammen in einer Schreiboperation gesetzt werden:

[image]

- Passive_Manual_Gdes 1187/1188: auf 0
- Passive_Manual_Blo 1189/118A: auf 0 (um die Entladung zu verhindern)
- Passive_Manual_Bup 118B/118C: auf 2147483647 (um trotzdem eine Ladung zu ermöglichen, falls vom Dach doch mehr kommt als die Autos schlucken können)

cschlipf, 2024-09-14:
[SOFAR Modbus Protocol G3_2023-12-22_1.22_en-INT.xlsx](https://github.com/user-attachments/files/17003351/SOFAR.Modbus.Protocol.G3_2023-12-22_1.22_en-INT.xlsx)

Bei der Gelegenheit hier die aktuellsten ModBus Register.

andig, 2024-09-14:
Ich brauche bitte einfach die Liste der Register mit Wert je Modus. Keine Screenshots, keine Textaufgabe. Danke!

cschlipf, 2024-09-14:
OK, ich dachte das hätte ich schon genau genug beschrieben.

- Batterieentladung deaktivieren:
  - Energystorage mode auf Passive setzen: U16 Register 1110 auf 3
  - Leistungen setzen zusammen in einer einzelnen Schreiboperation:
    - I32 Register 1187/1188 auf 0
    - I32 Register 1189/118A auf 0
    - I32 Register 118b/118C auf 2147483647
- Batterieentladung wieder deaktiveren/Normalbetrieb
  - Energystorage mode auf Self-Use setzen: U16 Register 1110 auf 0

Hoffe das war kurz und prägnant genug.

andig, 2024-09-14:
Danke, das hilft dem Entwickler mit wenig Zeit- Netzladung gibts nicht?

andig, 2024-09-14:
Gerne testen

cschlipf, 2024-09-14:
Netzladung geht genauso, nur mit anderen Werten:

- Netzladung deaktivieren:
  - Energystorage mode auf Passive setzen: U16 Register 1110 auf 3
  - Leistungen setzen zusammen in einer einzelnen Schreiboperation:
    - I32 Register 1187/1188 auf 0
    - I32 Register 1189/118A auf \<Ladeleistung\>
    - I32 Register 118b/118C auf \<Ladeleistung\>

\<Ladeleistung\> ist hier der Wert in W mit der der Akku geladen werden soll. Beim BTS Speicher ist das die Hälfte der Kapazität (z.B. ein BTS Speicher mit 10 kWh Kapazität braucht hier einen Wert von 5000 für die maximale Ladeleistung)

- Zurück auf Normalbetrieb
  - Energystorage mode auf Self-Use setzen: U16 Register 1110 auf 0

cschlipf, 2024-09-14:
> Gerne testen

Muss leider warten: Morgen früh geht es in den Urlaub.

wardwygaerts, 2024-09-19:
> Gerne testen

How can I test? Can I install this branch in Docker?

closed 2024-09-20 by andig

Frintrop, 2024-09-23:
@cschlipf  @andig 

Gilt dies nicht auch für die anderen Wechselrichter von Sofar funktionieren, die das -g3 Template benutzen. Im Code von Solax_Modbus sehe ich keine anderen Register für die anderen Wechselrichter. ( Insbesonder HYD xKTL-3PH ) ?!?!

Zwer2k, 2024-09-23:
Genau, sollte nicht eigentlich sofarsolar-g3.yaml angepasst werden?
Bei mir tauchen die Einstellungen nicht auf.
Ich glaube auch, dass mit der Anpassung die Register einzeln geschrieben werden, das wird vermutlich nicht funktionieren. Damit es funktioniert, sollten aber 3 Register gleichzeitig geschrieben werden.
Hier noch mein Diskussionsbeitrag als Querverweis https://github.com/evcc-io/evcc/discussions/15722#discussion-7098388

Frintrop, 2024-09-24:
Sollen wir ein neues Ticket aufmachen für das  sofarsolar-g3.yaml Template ? @andig @cschlipf  ?

andig, 2024-09-24:
Jetzt wäre es doch prima, diesen PR _überhaupt_ mal zu testen. Dann können wir ihn ja einfach kopieren falls die Register identisch sind.

cschlipf, 2024-09-24:
Genau. Das wurde im falschen Template eingebaut. Der HYD braucht das G3 Template. Ich wurde empfehlen den PR zu reverten. Die Register der alten Wechselrichter sind komplett anders. So kann ich das auch leider nicht testen.

andig, 2024-09-25:
Neues Nightly in 20min

brandsb, 2024-09-25:
Hello, I'm trying to accomplish the same thing (first aid topic #15798) but for the HYD5000-ES inverter. The addresses don't match with those of the sofarsolar nor with the G3 template. I've posted the modbus comms documentation [Here](https://github.com/user-attachments/files/17094042/Sofar.HYD.and.ME3000SP.Modbus.comms.protocol.pdf)

Anyone able to assist me on this? thx!

sahomm, 2024-09-25:
Hier das versprochene Feedback...
Setup: HYD 15 KTL mit 20kw BTS Batterie
EVCC-Version: 0.130.12+1727247451

Szenario: 
1. Fahrzeug anschließen im PV-Mode
2. Umschalten auf Schnell
3. nach ca 2 Minuten zurück auf PV

Ergebnis:

- Das Fahrzeug wurde erkannt und nicht geladen, da kein PV-Überschuss. >> OK
- Nach Umschaltung auf "Schnell" wurde das Fahrzeug geladen und der Akku abgeschaltet (HYD im Passive-Mode) >> OK
Hier wird aber zyklisch folgende Fehlermeldung geworfen:

`[site ] ERROR 2024/09/25 20:22:31 battery mode: modbus: response data size '18' does not match count '4'` 

- Die Umschaltung zurück auf PV wird von EVCC ordnungsgemäß durchgeführt, die Umschaltung des HYD auf "SelfUse" um die Akkunutzung wieder zu aktivieren jedoch nicht. >>notOK

Das Log hänge ich an, SKI.Keys wurden entfernt.

Vielen Dank für eure tolle Arbeit!

Saho
[evcc-20240925-204241-debug-no-ski.log](https://github.com/user-attachments/files/17137396/evcc-20240925-204241-debug-no-ski.log)

wardwygaerts, 2024-09-25:
I have a HYD 5000-EP,  and getting this error when trying to charge my battery from the grid:

```[site ] ERROR 2024/09/25 21:14:01 battery mode: modbus: response data size '18' does not match count '4'
```

sahomm, 2024-09-25:
@wardwygaerts 
yes, that is also my observation (see the post before), but switching off the battery works!

wardwygaerts, 2024-09-25:
@sahomm sorry, didn't saw your last post.
As soon as I set it to charge from the grid, the Storage Mode indeed changes to Self Use.
Other question: if it would work, with how much watts will it charge the battery?

cschlipf, 2024-09-25:
@sahomm and @wardwygaerts 

You both are using the LSE-3, right? Then this response is expected as the LSE-3 os returning in invalid response length on write. See https://www.photovoltaikforum.com/thread/217362-sofar-hyd-serie-kommunikation-%C3%BCber-modbus-tcp-erfahrungen/?postID=3873925#post3873925

andig, 2024-09-25:
If modbus is a marketed feature then thats a bug and should be fixed. We cannot just ignore arbitrary messages.

wardwygaerts, 2024-09-25:
> You both are using the LSE-3, right?

I am yes.
Currently I don't have a car connected, so I'm only trying to use it to charge my home battery. But when setting a high price limit (higher than my tarriff), nothing happens (except setting storage mode to Self Use, of this was not the case), only this error.

cschlipf, 2024-09-25:
> If modbus is a marketed feature then thats a bug and should be fixed. We cannot just ignore arbitrary messages.

Not sure it's a marketed feature of the LSE-3. I think it's rather unofficial.

Please note that I did not say 'ignore'. My suggestion was just to lower the severity to Warning. In general an invalid response to a write request does not necessarily mean it's a hard error.

andig, 2024-09-26:
It does. It means the write did not succeed in the general case.

Frintrop, 2024-09-27:
> I am yes. Currently I don't have a car connected, so I'm only trying to use it to charge my home battery. But when setting a high price limit (higher than my tarriff), nothing happens (except [...]

@cschlipf Did you try to load the battery ?

PeterPablo, 2024-09-30:
there might be a regression, related to introduction of battery control, ref. #16432

Zwer2k, 2024-09-30:
I use waveshare adapters for communication, but I still get the error messages in the log. The system switches between passive mode and selfuse mode, but the values for passive mode are not adjusted. I think the values are sent separately to the inverter and not as a data packet as the inverter expects

Pitchen84, 2024-10-05:
> Hier das versprochene Feedback... Setup: HYD 15 KTL mit 20kw BTS Batterie EVCC-Version: 0.130.12+1727247451
>
> Szenario:
>
> 1. Fahrzeug anschließen im PV-Mode
> 2. Umschalten auf Schnell
> 3. nach ca 2 [...]

Bei mir verhält es sich exakt genau so.

andig, 2024-10-05:
@Pitchen84 Lse3? Laut https://github.com/evcc-io/evcc/issues/12220#issuecomment-2375131718 geht das nicht, also bitte an den Hersteller wenden.

cschlipf, 2024-10-05:
@andig

Es fliegen nur Fehler wegen der ungültigen Response. Es wird aber durchgeführt.

Prinzipiell funktioniert es ja, nur bei schalten auf Self Use scheint noch ein Problem zu sein.

cschlipf, 2024-10-05:
Also hier der Case 1 (funktioniert nicht) sollte genau dasselbe machen wie Case 2 (funktioniert). Nur eben mit dem Wert 0 (self-use), statt 3 (passive)

```yaml
    switch:
    - case: 1 # normal
      set:
        source: const
        value: 0 # self-use
        set:
          source: modbus
          {{- include "modbus" . | indent 8 }}
          register:
            address: 0x1110
            type: writemultiple
            decode: int16
    - case: 2 # hold
      set:
        source: sequence
        set:
        - source: const
          value: 3 # passive
          set:
            source: modbus
            {{- include "modbus" . | indent 10 }}
            register:
              address: 0x1110
              type: writemultiple
              decode: int16
```

Dinge die mir hier auffallen:
- Case 1 ist keine Liste unter `set:`. Fehlt hier ein `-` davor?
- Laut Protokoll ist das Register 0x1110 ein uint16 wert. Aber warum tut dann Case 2?

andig, 2024-10-05:
Int/uint ist bei positiven Werten egal. case 1 braucht keine Liste weil nur 1 Wert gesetzt wird, in allen anderen Cases aber mehrere.

> Prinzipiell funktioniert es ja, nur bei schalten auf Self Use scheint noch ein Problem zu sein.

Das Problem hier ist dann ein rein optisches? Was spricht dagegen, sich einfach an den Hersteller zu wenden? Hat das überhaupt schonmal jemand probiert?

Zwer2k, 2024-10-05:
Mit dem fehlenden - in Case 1 wirst du recht haben, es wird vermutlich dadurch einfach nicht ausgeführt. Es gibt auch keine Fehlermeldung im Log.
In Case 1 wird zwar "pasive mode" mit dem Register 0x1110 gesetzt, die Leistungswerte werden aber nicht angepasst, d.h. Register  0x1187, 0x1189 und 0x118b bleiben unverändert und es wird eine Fehlermeldung im `battery mode: modbus: exception '3' (illegal data value), function '16'` geworfen. Wie gesagt, ich verwende Waveshare nicht LSE3

Hier sieht man, dass die 3 Werte nicht verändert worden sind, sondern auf meine vor eingestellten Werten geblieben sind
[image]
 

> Also hier der Case 1 (funktioniert nicht) sollte genau dasselbe machen wie Case 2 (funktioniert). Nur eben mit dem Wert 0 (self-use), statt 3 (passive)
> 
> ```yaml
>     switch:
>     - case: 1 # normal
>       set:
>         source: const
>         value: 0 # self-use
>         set:
>           source: modbus
>           {{- include "modbus" . | indent 8 }}
>           register:
>             address: 0x1110
>             type: writemultiple
>             decode: int16
>     - case: 2 # hold
>       set:
>         source: sequence
>         set:
>         - source: const
>           value: 3 # passive
>           set:
>             source: modbus
>             {{- include "modbus" . | indent 10 }}
>             register:
>               address: 0x1110
>               type: writemultiple
>               decode: int16
> ```
> 
> Dinge die mir hier auffallen:
> 
> * Case 1 ist keine Liste unter `set:`. Fehlt hier ein `-` vor `source: const`?
> * Laut Protokoll ist das Register 0x1110 ein uint16 wert. Aber warum tut dann Case 2?

Mit dem fehlenden - in Case 1 wirst du recht haben, es wird vermutlich dadurch einfach nicht ausgeführt. Es gibt auch keine Fehlermeldung im Log.
In Case 1 wird zwar "pasive mode" mit dem Register 0x1110 gesetzt, die Leistungswerte werden aber nicht angepasst, d.h. Register  0x1187, 0x1189 und 0x118b bleiben unverändert und es wird eine Fehlermeldung im `battery mode: modbus: exception '3' (illegal data value), function '16'` geworfen. Wie gesagt, ich verwende Waveshare nicht LSE3
Es ist wichtig, dass die 3 Register mit einem Befehl an den WR geschickt wird und nicht mit 3 einzelnen aufrufen, anderer falls wird vom WR ein Fehler zurück gegeben.

Hier sieht man, dass die 3 Werte nicht verändert worden sind, sondern auf meine vor eingestellten Werten geblieben sind.
Hier hatte ich auch schon das Schnellladen wieder ausgeschaltet, d.h. Modus hätte sich wieder auf "self use" ändern sollen.
[image]

cschlipf, 2024-10-05:
Dann verstehe ich ehrlich gesagt nicht, warum Case 1 nicht funktioniert, aber Case 2.

Ja, ist ein rein optisches Problem im Log. Deswegen habe ich oben ja vorgeschlagen das in eine Warning zu ändern, aber ich verstehe auch Deine Argumente dagegen.

Ich habe das schon an meine Kontakte von Sofar Deutschland weitergeleitet. Aber
- Modbus TCP ist kein offizielles Feature vom LSE-3. 
- Der LSE-3 Stick wird nicht von Sofar hergestellt, sondern von einem dritten Hersteller (IGEN Tech, Solarman), welcher auch noch andere Wechselrichter unterstützt.
- Die von Sofar unterstützte Schnittstelle ist RS 485.

andig, 2024-10-05:
> Der LSE-3 Stick wird nicht von Sofar hergestellt, sondern von einem dritten Hersteller (IGEN Tech, Solarman), welcher auch noch andere Wechselrichter unterstützt.

Das ändert nix dran, dass das Teil Schrott ist. Dann bei diesem Hersteller beschweren. 

> Die von Sofar unterstützte Schnittstelle ist RS 485.

...oder einfach auf eine funktionierende Schnittstelle umsteigen. Wir halten fest: evcc funktioniert wie erwartet.

> In Case 1 wird zwar "pasive mode" mit dem Register 0x1110 gesetzt, die Leistungswerte werden aber nicht angepasst, d.h. Register 0x1187, 0x1189 und 0x118b bleiben unverändert und es wird eine [...]
Es ist wichtig, dass die 3 Register mit einem Befehl an den WR geschickt wird und nicht mit 3 einzelnen aufrufen, anderer falls wird vom WR ein Fehler zurück gegeben.

@Zwer2k das ist spannend. Die Vorgabe war:

> Energystorage mode auf Self-Use setzen: U16 Register 1110 auf 0

Du sagst das funktioniert so nicht? Was muss stattdessen geschrieben werden?

Pitchen84, 2024-10-05:
> @Pitchen84 Lse3? Laut [#12220 (comment)](https://github.com/evcc-io/evcc/issues/12220#issuecomment-2375131718) geht das nicht, also bitte an den Hersteller wenden.
> 
> Und: gibts nur Fehler oder wird nicht umgeschaltet?

LSE3, HYD15 KTL mit 10 kWh BTS und 61er Firmware

andig, 2024-10-05:
> Der LSE-3 Stick wird nicht von Sofar hergestellt, sondern von einem dritten Hersteller (IGEN Tech, Solarman), welcher auch noch andere Wechselrichter unterstützt.

@cschlipf was mich jetzt wirklich irritiert ist der Text im Template:

> Es wird empfohlen die Verbindung über einen LSE-3 Logger Stick mittels ModBus TCP herzustellen (LSW-3 WLAN Stick wird nicht unterstützt). Bei seriellem Anschluss via RS485 mit entsprechendem Adapter am COM Port ist zu beachten, dass wechselrichterseitig für eine Terminierung des RS485 Busses zu sorgen ist.

Wenn der LSE Schrott ist müssten wir das korrigieren?

Zwer2k, 2024-10-05:
Case 1 (also zurückschalten auf "self use") funktioniert bei mir nicht. So wie ich es verstanden habe, auch bei anderen funktioniert es nicht.
Aus meiner Sicht liegt es so wie @cschlipf beschrieben hat an fehlenden `-` in Template
[image]

cschlipf, 2024-10-05:
@andig Wie gesagt - es ist nur eine falsche Fehlermeldung. Es tut ja trotzdem.

> Energystorage mode auf Self-Use setzen: U16 Register 1110 auf 0

Doch, das ist richtig. Und dieses eine Register kann auch alleine für sich geschrieben werden. Ich verstehe daher nicht, warum der Self-Use nicht wieder aktiviert wird. Wie gesagt: Meine Vermutung war die fehlende Liste.

Hier nochmal die Specs:
[image]

Was 0x1187, 0x1189 und 0x118b betrifft: Hier ist offensichtlich das Problem, dass diese nicht zusammen in einer Write Operations geschrieben werden:
[image]

(siehe letzte Spalte unter Remarks)

cschlipf, 2024-10-05:
> @cschlipf was mich jetzt wirklich irritiert ist der Text im Template:
>
>
> Wenn der LSE Schrott ist müssten wir das korrigieren?

Bisher haben wir nur ausgelesen. Da funktioniert der LSE-3 wunderbar und ist die einfachste Methode. Beim Schreibzugriff kommt eben der falsche Response Code zurück - das ist nicht schön. Aber nochmal: Das hat nun mit den Problemen hier nichts zu tun. In Home Asssistant ignoriere ich den fehlerhaften Return Code eben und es tut alles.

RS485 ware die sauberste Methode. Ja. Nur ist das eben für unbedarfte einiges an zusätzlicher Komplexität: Richtige Terminierung, richtige Verkabelung, kompatiblen Adapter finden (hier gibt es zig Forumspost dazu), korrekte Adaptereinstellungen,... wenn das alles passt, die bessere Lösung - nur kannst das halt nicht jedem zumuten.

andig, 2024-10-05:
> Wie gesagt: Meine Vermutung war die fehlende Liste.

Da gibts nix zu vermuten. Ob das passiert ist ja im Logfile zu sehen. Und es _wird_ ja geschrieben denn sonst gäbe es den Fehler nicht.

> Was 0x1187, 0x1189 und 0x118b betrifft: Hier ist offensichtlich das Problem, dass diese nicht zusammen in einer Write Operations geschrieben werden:

Verstehe ich- auch bei gutem Willen- nicht. Das mag so sein. Aber: Wollen wir jetzt diskutieren, dass das Feature _gar nicht_ funktioniert? Das habe ich zumindest oben noch nicht raus hören können.

Long story short: ich hab keinen Überblick mehr, worum es hier eigentlich geht und würde mich daher ausklingen. Fehlerhafte Hardware/Firmware ist bitte beim Hersteller zu reklamieren.

> In Home Asssistant ignoriere ich den fehlerhaften Return Code eben und es tut alles.

Prima. Dann ist das die Lösung. Die Alternative habe ich jetzt schon x-mal beschrieben und die besteht darin ein fehlerhaftes Produkt zu reklamieren.

cschlipf, 2024-10-05:
Also ich sehe aktuell 2 Fehler:

- Self-Use wird nicht gesetzt um die Batterieladung wieder zu aktivieren. Warum das nicht tut, verstehe ich nicht. Register und Werte sind korrekt.
- Werte für 0x1187, 0x1189 und 0x118b werden nicht gesetzt. Dies ist in vielen Fällen kein Problem, da die Default Werte hier 0 sind (Batterie deaktiviert). Es ist aber ein Problem, wenn jemand diese Werte nicht auf Default gesetzt hat, weil er z.B. via HA damit den Speicher lädt oder weil EVCC auch mal den Speicher laden will. Dann werden beim Umschalten auf Passive eben die zuletzt gesetzten Werte genommen und diese können dann statt Speicher deaktivieren eben ganz andere Effekte haben.

andig, 2024-10-05:
> Werte für 0x1187, 0x1189 und 0x118b werden nicht gesetzt. Dies ist in vielen Fällen kein Problem, da die Default Werte hier 0 sind (Batterie deaktiviert). Es ist aber ein Problem, wenn jemand diese [...]

...und das ist hier völlig OT. Ich mache die Diskussion an der Stelle mal zu da sie sich tot gelaufen hat.

Zwer2k, 2024-10-05:
> > Wie gesagt: Meine Vermutung war die fehlende Liste.
> 
> Da gibts nix zu vermuten. Ob das passiert ist ja im Logfile zu sehen. Und es _wird_ ja geschrieben denn sonst gäbe es den Fehler nicht.

Das ist es ja, in case 1 gibt es keine Fehlermeldung. Fehler Wird nur in case 2 geworfen, wegen der falschen Behandlung von den 3 Registern (0x1187, 0x1189 und 0x118b).

andig, 2024-10-05:
@cschlipf @Zwer2k  erlaubt die Doku auch die beiden letzten Register zusammen zu schreiben? Liest sich für mich so. Dann könnte auch https://github.com/evcc-io/evcc/pull/16509 helfen. Auch das müsste aber mal jemand probieren...

Drei Register zu schreiben gibt unsere Modbuslösung aktuell nicht mehr, dann müssten wir das Feature reverten.

andig, 2024-10-05:
@Zwer2k du trägst leider auch nicht zur Klarheit bei. Erst hiess es:

> Case 1 (also zurückschalten auf "self use") funktioniert bei mir nicht. 

und jetzt:

> Fehler Wird nur in case 2 geworfen, wegen der falschen Behandlung von den 3 Registern (0x1187, 0x1189 und 0x118b).

Das ist ein Supportalptraum!

andig, 2024-10-05:
Letzte Variante: wir schmeissen die Register 0x1187, 0x1189, 0x118B einfach raus und verlassen uns auf die Defaults. Zur Not müsste man noch erklären wie die zu setzen sind.

cschlipf, 2024-10-05:
> @cschlipf @Zwer2k erlaubt die Doku auch die beiden letzten Register zusammen zu schreiben? Liest sich für mich so. Dann könnte auch #16509 helfen. Auch das müsste aber mal jemand probieren...
> 
> Drei Register zu schreiben gibt unsere Modbuslösung aktuell nicht her, dann müssten wir das Feature reverten.

Nein. Es müssen immer alle Register geschrieben werden. Ja das ist sehr nervig und glaub mir, das hier ist einer der harmlosesten Fälle

andig, 2024-10-05:
Ok. Ich würde nochmal einen Versuch unternehmen, das Problem grundsätzlich zu lösen. Dafür bräuchte ich aber Remotezugang zu einem WR um das auch testen zu können. Wer helfen will schickt bitte VPN/SSH an info@evcc.io.

cschlipf, 2024-10-05:
> Letzte Variante: wir schmeissen die Register 0x1187, 0x1189, 0x118B einfach raus und verlassen uns auf die Defaults. Zur Not müsste man noch erklären wie die zu setzen sind.

Denke sollte für die meisten eine gute Lösung sein. Wer die Register mit anderen Automationen ändert weiß auch wie er die wieder zurücksetzen kann. Sollte nur entsprechend dokumentiert werden, dass EVCC erwartet, dass diese Register alle auf 0 gesetzt sind.

andig, 2024-10-05:
> Sollte nur entsprechend dokumentiert werden, dass EVCC erwartet, dass diese Register alle auf 0 gesetzt sind.

Das wiederspricht  https://github.com/evcc-io/evcc/issues/12220#issuecomment-2351111170

Zwer2k, 2024-10-05:
> @Zwer2k du trägst leider auch nicht zur Klarheit bei. Erst hiess es:
> 
> > Case 1 (also zurückschalten auf "self use") funktioniert bei mir nicht.
> 
> und jetzt:
> 
> > Fehler Wird nur in case 2 geworfen, wegen der falschen Behandlung von den 3 Registern (0x1187, 0x1189 und 0x118b).
> 
> Das ist ein Supportalptraum!

Ist aus meiner Sicht auch nichts falsch. 
- Case 1 funktioniert nicht, wirft aus meiner Sicht aber auch kein Fehler. Ich glaube der wird aus irgend einem Grund einfach nicht ausgeführt. 
- Case 2 funktioniert nur teilweise, wirft aber ganze Zeit Fehlermeldungen    
Alle Fehlermeldungen scheinen vom Case 2 zu kommen. Kann ich aber nicht zu 100% zuordnen, da case 1 kurz nach dem case 2 ausgeführt wird, nach dem das Schnellladen deaktiviert worden ist.

cschlipf, 2024-10-05:
> > Sollte nur entsprechend dokumentiert werden, dass EVCC erwartet, dass diese Register alle auf 0 gesetzt sind.
> 
> Das wiederspricht [#12220 (comment)](https://github.com/evcc-io/evcc/issues/12220#issuecomment-2351111170)

Nun, 118b (upper limit) ist per default auf 0. Jemand, der nur EVCC einsetzt wird diesen nie ändern. Nutzt man hier den default von 0 hat es nur die Auswirkung, dass der Speicher nicht während des Schnellladens der Fahrzeuge mit Überschuss geladen wird, wenn vom Dach noch mehr Leistung als die maximale Ladeleistung  der Wallboxen + Hausverbrauch kommt. Damit kann man vermutlich leben.

Zwer2k, 2024-10-05:
> @cschlipf @Zwer2k erlaubt die Doku auch die beiden letzten Register zusammen zu schreiben? Liest sich für mich so. Dann könnte auch #16509 helfen. Auch das müsste aber mal jemand probieren...

Die Doku schein zu passen. Es lassen sich auch nur die 4 letztere Register beschreiben (eben getestet). Das mit dem 64bit blob sollte daher gehen. 

> 
> Drei Register zu schreiben gibt unsere Modbuslösung aktuell nicht her, dann müssten wir das Feature reverten.

Wieso wurde die Funktion wieder revertet?

> > > Sollte nur entsprechend dokumentiert werden, dass EVCC erwartet, dass diese Register alle auf 0 gesetzt sind.
> > 
> > 
> > Das wiederspricht [#12220 (comment)](https://github.com/evcc-io/evcc/issues/12220#issuecomment-2351111170)
> 
> Nun, 118b (upper limit) ist per default auf 0. Jemand, der nur EVCC einsetzt wird diesen nie ändern. Nutzt man hier den default von 0 hat es nur die Auswirkung, dass der Speicher nicht während des Schnellladens der Fahrzeuge mit Überschuss geladen wird, wenn vom Dach noch mehr Leistung als die maximale Ladeleistung der Wallboxen + Hausverbrauch kommt. Damit kann man vermutlich leben.
> 
> Alle Werte auf 0 heißt, dass der Speicher gar nichts mehr macht und praktisch deaktiviert ist.

Um hier Probleme zu vermeiden, würde es eher sinn machen die Einstellung für "Self use"/"passive mode" dem User zu überlassen und nur die Lade-/Entladeleistung durch EVCC zu beeinflussen. Dadurch wird zusätzlich vermieden, dass der EEPROM zu oft beschrieben wird und dadurch vorzeitig altert. Die Register 0x1187, 0x1189, 0x118b werde nur in RAM vorgehalten und dürfen beliebig oft überschrieben werden.
Würde heißen nur Register 0x1189 + 0x118B als 64bit Blob beschreiben:
- case 1: Register 0x1189 = 0xFFFFB1E0 (-20000W), Register 0x118B = 0x00004E20 (+20000W)
- case 2: Register 0x1189 = 0x00000000 (0W), Register 0x118B = 0x00004E20 (+20000W)

cschlipf, 2024-10-05:
@Zwer2k Was Du vorschlägst macht keinen Sinn. Wenn man 0x1189 un 0x118b per 64Bit Blob schreiben kann und EVCC das unterstützt, dann kann man auch Case 1 mit 0x1110 auf Self-Use machen und Case 2 mit 0x1110 auf Passive Mode beibehalten.

Case 1 würde dann weiterhin den Wechselrichter wieder auf Ausgangssituation bringen, so wie der Wechselrichter vom Solarteur und man müsste nicht im Default Fall mit einem Passive Mode arbeiten.

Zwer2k, 2024-10-05:
Für mich macht es durchaus Sinn. Ich verwende “Passive: Desired Grid Power” (Register 0x1187), um die Netzentnahme durch das Überschwingen zu verhindern. Je mehr PV-Überschuss zur Verfügung steht, desto höhere negative Zahlen gebe ich hier an und desto weniger Netzentnahme gibt es. Das Setzen auf “Self use” durch EVCC würde mir diese Möglichkeit nehmen.

Ich weiß, dass der Wechselrichter “PCC power bias” hat, aber diese Funktion ist mir zu unflexibel und verfälscht zudem die Statistik. Mir ist klar, dass dies sehr spezifisch ist und außer mir vermutlich niemand so verwendet. Es sollte natürlich für die breite Masse der Nutzer funktionieren. Es wäre fein, wenn es hier eine Anpassungsmöglichkeit per evcc.conf gäbe.

Frintrop, 2024-10-06:
> Ok. Ich würde nochmal einen Versuch unternehmen, das Problem grundsätzlich zu lösen. Dafür bräuchte ich aber Remotezugang zu einem WR um das auch testen zu können. Wer helfen will schickt bitte VPN/SSH an [info@evcc.io](mailto:info@evcc.io).

Hast Du schon einen VPN Zugang ?

[57 more replies]