evcc-io/evcc #12230: SMA Sunny Island Batteriesteuerung

trsen, 2024-02-13:
### Describe the bug

Wenn ich die Batteriesteuerung aktiviere und meinen eGolf im Schnellmodus lade dann entläd er nicht die Hausbatterie sondern er läd sie mit der vollen Leistung.

### Steps to reproduce

1. Auto anschließen und Ladung auf Schnell stellen.
2. Batterieeinstellung "Batteriesteuerung" auf EIN stellen.
3. Der Sunny Island fängt an die Hausbatterie mit voller Leistung zu laden.

[image]

[image]

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: info
levels:
  cache: error

# unique installation id
plant: XXX

interval: 30s # control cycle interval

sponsortoken: XXX
# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

meters:
- type: template
  template: sma-home-manager 
  usage: grid  
  host: XXX  
  name: grid1
- type: template
  template: sma-inverter 
  usage: pv  
  host: XXX
  password: XXX 
  name: pv2
- type: template
  template: sma-inverter 
  usage: pv  
  host: XXX  
  password: XXX  
  name: pv3
- type: template
  template: sma-inverter 
  usage: pv  
  host: XXX
  password: XXX  
  name: pv4
#- type: template
#  template: sma-inverter 
#  usage: battery  
#  host: XXX
#  password: XXX
#  name: battery5
- type: template
  name: battery5
  usage: battery
  template: sma-si-modbus
  #Modbus TCP
  modbus: tcpip
  id: 3
  host: XXX
  port: 502
- type: template
  template: shelly-3em 
  usage: charge  
  host: XXX  
  name: charge8

chargers:
- name: wallbox7
  type: template
  template: abl  
  modbus: rs485tcpip
  host: XXX    
  id: 1  
  port: 502
  timeout: 10s

vehicles:
- type: template
  template: id 
  title: e-Golf  
  user: XXX  
  password: XXX  
  vin:XXX  
  capacity: 35.8  
  name: ev6

loadpoints:
- title: Carport
  charger: wallbox7
  meter: charge8
  mode: pv
  enable:
    threshold: -2600
    delay: 1m

site:
  title: Mein Zuhause
  meters:
    grid: grid1
    pv:
    - pv2
    - pv3
    - pv4
    battery:
    - battery5
  residualPower: 100

messaging:
  events:
    start:
      title: Laden gestartet
      msg: e-Golf wir im ${mode} Mode geladen.
    stop:
      title: Laden gestoppt
      msg: In den e-Golf wurden ${chargedEnergy:%.1fk}kWh in ${chargeDuration} geladen.
    connect:
      title: e-Golf angeschlossen
      msg: Der e-Golf wurde an die Wallbox angeschlossen, es ist ${pvPower:%.1fk}kW PV-Leistung vorhanden.
    disconnect:
      title: e-Golf abgesteckt
      msg: Der e-Golf wurde nach ${connectedDuration} von der Wallbox getrennt.
  services:
    - type: telegram
      token: XXX
      chats:
        - -XXX

#mqtt message broker
mqtt:
  broker: XXX:1883
  topic: evcc # root topic for publishing, set empty to disable publishing
  # clientid: foo
  user: XXX
  password: XXX

#mqtt:
#  broker: XXX:1884
#  topic: evcc
#  user: XXX
#  password: XXX

tariffs:
  currency: EUR # (default EUR)
  grid:
    # static grid price
    type: fixed
    price: 0.3767 # [currency]/kWh

  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.119 # [currency]/kWh
```

### Log details

```text
[main  ] INFO 2024/02/13 12:28:47 evcc 0.124.1
[main  ] INFO 2024/02/13 12:28:47 using config file: /etc/evcc.yaml
[main  ] INFO 2024/02/13 12:28:47 starting ui and api at :7070
[db    ] INFO 2024/02/13 12:28:47 using sqlite database: /home/XXXX/.evcc/evcc.db
[mqtt  ] INFO 2024/02/13 12:28:47 connecting evcc-1175332669 at tcp://XX.XX.XX.XX:1883
[mqtt  ] DEBUG 2024/02/13 12:28:47 tcp://XX.XX.XX.XX:1883 connected
[lp-1  ] WARN 2024/02/13 12:28:53 deprecated: minCurrent setting is ignored, please remove
[lp-1  ] WARN 2024/02/13 12:28:53 deprecated: maxcurrent setting is ignored, please remove
[lp-1  ] WARN 2024/02/13 12:28:53 deprecated: phases setting is ignored, please remove
[lp-1  ] DEBUG 2024/02/13 12:28:53 charge total import: 1050.362kWh
[site  ] INFO 2024/02/13 12:28:54 site config:
[site  ] INFO 2024/02/13 12:28:54   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2024/02/13 12:28:54     grid:      power ✓ energy ✓ currents ✓
[site  ] INFO 2024/02/13 12:28:54     pv 1:      power ✓ energy ✓ currents ✓
[site  ] INFO 2024/02/13 12:28:54     pv 2:      power ✓ energy ✓ currents ✓
[site  ] INFO 2024/02/13 12:28:54     pv 3:      power ✓ energy ✓ currents ✓
[site  ] INFO 2024/02/13 12:28:54     battery 1: power ✓ energy ✓ currents ✗ soc ✓ capacity ✗
[site  ] INFO 2024/02/13 12:28:54   vehicles:
[site  ] INFO 2024/02/13 12:28:54     vehicle 1: range ✓ finish ✓ status ✓ climate ✓ wakeup ✓
[lp-1  ] INFO 2024/02/13 12:28:54 loadpoint 1:
[lp-1  ] INFO 2024/02/13 12:28:54   mode:        pv
[lp-1  ] INFO 2024/02/13 12:28:54   charger:     power ✗ energy ✗ currents ✗ phases ✗ wakeup ✓
[lp-1  ] INFO 2024/02/13 12:28:54   meters:      charge ✓
[lp-1  ] INFO 2024/02/13 12:28:54     charge:    power ✓ energy ✓ currents ✓
[lp-1  ] DEBUG 2024/02/13 12:28:54 phase timer inactive
[lp-1  ] DEBUG 2024/02/13 12:28:54 pv timer inactive
[lp-1  ] DEBUG 2024/02/13 12:28:54 guard timer inactive
[lp-1  ] DEBUG 2024/02/13 12:28:54 max charge current: 6A
[lp-1  ] DEBUG 2024/02/13 12:28:54 guard enable in 5m0s
[main  ] FATAL 2024/02/13 12:28:54 could not open port- check that evcc is not already running (listen tcp :7070: bind: address already in use)
[site  ] WARN 2024/02/13 12:28:54 interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval
[site  ] DEBUG 2024/02/13 12:28:54 ----
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.124.1evcc

premultiply, 2024-02-13:
Oha. Das ist eventuell was vertauscht. 😅
Ich schaue es mir an.

julred, 2024-02-14:
@trsen: Habe auch einen eGolf - wie bekommst du den Ladezustand in deine Übersicht rein? Sehe anhand deiner evcc.yaml config keine Zugriffe auf eine API?!

trsen, 2024-02-14:
Der API Zugang zu meinem eGolf ist in der config unter vehicle abgelegtGruß Torge

premultiply, 2024-03-08:
Ich kann im Screenshot nicht sehen dass die Batterie **geladen** wird.
Das Log enthält leider gar keine sinnvollen Daten. Bitte mal passendes Log dazu.

Bei dem SMA-Systemen dauert es mehrere Minuten bis die Sperre voll greift, da hier vom System ein sehr langsames internes Ramping gefahren wird. Man muss daher erstmal viele Minuten warten.

Ich mache hier erstmal zu.
Falls sich das nicht damit erklären lässt bitte nochmal weitere Infos beisteuern die das Fehlverhalten belegen.

closed 2024-03-08 by premultiply