evcc-io/evcc #12251: Cursor Änderung in Batterieeinstellungen

VolkerK62, 2024-02-14:
Dass man auf die %-Zahlen (im Standard 0 und 100) klicken kann ist auch nicht wirklich ersichtlich, es ändert sich auch nicht der Cursor.

_Originally posted by @KlausHans in https://github.com/evcc-io/evcc/discussions/8177#discussioncomment-8467733_

VolkerK62, 2024-02-14:
Das stimmt, der Mauszeiger bleibt unverändert. Wäre schön, wenn sich das (wie sonst auch) in diesen "Finger" ändern würde.

StevieC121176, 2024-02-15:
Das wäre zu den anderen Sichten wie Loadpoint aber nicht konsistent. Ein Auswahlmenü wird ja dadurch suggeriert das es unterstrichen ist. Dieser Finger kommt überall da, wo man z.Bsp. eine Aktion ausführt (z.Bsp. am Loadpoint durchschalten kann zwischen Preis, Sonnenanteil usw.). Der Link öffnet hingegen das Auswahlmenü um diese Sachen direkt auszuwählen..

VolkerK62, 2024-02-15:
Ah, ok. So tief steck ich "in den Regeln" nicht drin.
Aber, wo ist denn der Unterschied, ob ich am Loadpoint durchschalte oder in den batteriesinstellungen einen anderen Prozentwert wähle? Ist doch beides eine "Liste" in der eine Auswahl getroffen wird.

closed 2024-02-15 by andig