evcc-io/evcc #12252: Shelly Pro EM - 50

peterle-k, 2024-02-14:
I would need a template for Shelly Pro EM - 50 as meter. It works well as a charger but not as meter. Both templates shelly-3em and shelly-pro-3em don't work.

`meters:
- type: template
  template: shelly-3em
  usage: charge
  host: 192.168xxxx
  name: shelly

chargers:
- type: template
  name: wallbox6
  template: shelly
  host: 192.168xxxx
  standbypower: 15`

many thanks
Peter

thierolm, 2024-02-14:
Wenn das `shelly` Charger-Template funktioniert, dann müsste das `shelly-1pm` Meter-Template passen.

Falls es nicht funktioniert, ergänze bitte wie dokumentiert ein Trace-Log (siehe Doku), sonst können wir nicht unterstützen.

thierolm, 2024-02-14:
Ich hab die Ausgabe mal formatiert. Folgende Frage: Wie sieht dein konkretes Szenario aus?
Du hast an dem Shelly nur EINEN Schaltkanal, aber ZWEI Messkanäle (`em1:0` und `em1:1`). 
In deinem Setting willst du mit dem Meter
a) die SUMME beider Messkanäle oder
b) nur EINEN der beiden Messkanäle
messen?

```json
{
    "ble": {},
    "cloud": {
        "connected": false
    },
    "em1:0": {
        "id": 0,
        "current": 0.020,
        "voltage": 224.6,
        "act_power": 0.0,
        "aprt_power": 4.6,
        "pf": 0.00,
        "freq": 50.0,
        "calibration": "factory"
    },
    "em1:1": {
        "id": 1,
        "current": 22.780,
        "voltage": 224.6,
        "act_power": 5080.3,
        "aprt_power": 5142.4,
        "pf": 0.99,
        "freq": 50.0,
        "calibration": "factory"
    },
    "em1data:0": {
        "id": 0,
        "total_act_energy": 0.00,
        "total_act_ret_energy": 0.00
    },
    "em1data:1": {
        "id": 1,
        "total_act_energy": 6644.90,
        "total_act_ret_energy": 2595.26
    },
    "eth": {
        "ip": null
    },
    "modbus": {},
    "mqtt": {
        "connected": true
    },
    "switch:0": {
        "id": 0,
        "source": "HTTP_in",
        "output": true,
        "temperature": {
            "tC": 29.4,
            "tF": 85.0
        }
    },
    "sys": {
        "mac": "08F9E0E67204",
        "restart_required": false,
        "time": "18:16",
        "unixtime": 1707930973,
        "uptime": 28980,
        "ram_size": 241940,
        "ram_free": 117332,
        "fs_size": 524288,
        "fs_free": 208896,
        "cfg_rev": 20,
        "kvs_rev": 0,
        "schedule_rev": 0,
        "webhook_rev": 0,
        "available_updates": {
            "stable": {
                "version": "1.2.0"
            }
        },
        "reset_reason": 3
    },
    "wifi": {
        "sta_ip": "192.168.8.9",
        "status": "got ip",
        "ssid": "K1",
        "rssi": -60
    },
    "ws": {
        "connected": false
    }
}
```

thierolm, 2024-02-14:
Was liefert dir der Shelly Charger zurück?
Bitte einmal die Ausgabe der folgenden Befehle teilen:
`evcc -l trace charger`
und
`evcc -l trace meter`

peterle-k, 2024-02-14:
> Ich hab die Ausgabe mal formatiert. Folgende Frage: Wie sieht dein konkretes Szenario aus? Du hast an dem Shelly nur EINEN Schaltkanal, aber ZWEI Messkanäle (`em1:0` und `em1:1`). In deinem Setting [...]

der Shelly Pro EM50 hat einen Schaltkanal, Messkanal habe ich nur den em1:1 belegt und soll auch nur dieser verwendet werden

**pi@IOB-KEP:~ $ evcc -l trace charger**
[main  ] INFO 2024/02/14 18:45:45 evcc 0.124.1
[main  ] INFO 2024/02/14 18:45:45 using config file: /etc/evcc.yaml
[db    ] INFO 2024/02/14 18:45:45 using sqlite database: /home/pi/.evcc/evcc.db
[db    ] TRACE 2024/02/14 18:45:45 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2024/02/14 18:45:45 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2024/02/14 18:45:45 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/02/14 18:45:45 SELECT * FROM `settings` 6 <nil>
[mqtt  ] INFO 2024/02/14 18:45:45 connecting evcc at tcp://192.168.8.164:1884
[mqtt  ] DEBUG 2024/02/14 18:45:45 tcp://192.168.8.164:1884 connected
[mqtt  ] ERROR 2024/02/14 18:45:45 tcp://192.168.8.164:1884 connection lost: EOF
[db    ] TRACE 2024/02/14 18:45:45 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="devices" -1 <nil>
[db    ] TRACE 2024/02/14 18:45:45 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="device_details" -1 <nil>
[mqtt  ] DEBUG 2024/02/14 18:45:45 tcp://192.168.8.164:1884 connected
[db    ] TRACE 2024/02/14 18:45:45 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="configs" -1 <nil>
[db    ] TRACE 2024/02/14 18:45:45 CREATE TABLE `configs` (`id` integer PRIMARY KEY AUTOINCREMENT,`class` integer,`type` text) 0 <nil>
[db    ] TRACE 2024/02/14 18:45:45 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="config_details" -1 <nil>
[db    ] TRACE 2024/02/14 18:45:45 CREATE TABLE `config_details` (`config_id` integer,`key` text,`value` text,CONSTRAINT `fk_configs_details` FOREIGN KEY (`config_id`) REFERENCES `configs`(`id`) ON DELETE CASCADE ON UPDATE CASCADE) 0 <nil>
[db    ] TRACE 2024/02/14 18:45:45 CREATE INDEX `idx_unique` ON `config_details`(`config_id`,`key`) 0 <nil>
[db    ] TRACE 2024/02/14 18:45:45 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT ""fk_devices_details"" %" OR sql LIKE "%CONSTRAINT fk_devices_details %" OR sql LIKE "%CONSTRAINT `fk_devices_details`%" OR sql LIKE "%CONSTRAINT [fk_devices_details]%" OR sql LIKE "%CONSTRAINT 	fk_devices_details	%") -1 <nil>
[db    ] TRACE 2024/02/14 18:45:45 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%""device_id"" %" OR sql LIKE "%device_id %" OR sql LIKE "%`device_id`%" OR sql LIKE "%[device_id]%" OR sql LIKE "%	device_id	%") -1 <nil>
[shelly] TRACE 2024/02/14 18:45:45 GET http://192.168.8.9/shelly
[db    ] TRACE 2024/02/14 18:45:45 SELECT * FROM `configs` WHERE `configs`.`class` = 1 ORDER BY id 0 <nil>
[shelly] TRACE 2024/02/14 18:45:45 {"name":"Wallbox Carport ","id":"shellyproem50-08f9e0e67204","mac":"08F9E0E67204","slot":0,"model":"SPEM-002CEBEU50","gen":2,"fw_id":"20240130-115202/1.2.0-beta1-gd959786","ver":"1.2.0-beta1","app":"ProEM","auth_en":false,"auth_domain":null}
wallbox5
--------
[twc3  ] TRACE 2024/02/14 18:45:45 GET http://192.168.8.124/api/1/vitals
[twc3  ] TRACE 2024/02/14 18:45:45 {"contactor_closed":false,"vehicle_connected":false,"session_s":0,"grid_v":228.8,"grid_hz":50.050,"vehicle_current_a":0.3,"currentA_a":0.0,"currentB_a":0.3,"currentC_a":0.0,"currentN_a":0.0,"voltageA_v":0.0,"voltageB_v":0.0,"voltageC_v":0.0,"relay_coil_v":11.9,"pcba_temp_c":12.0,"handle_temp_c":6.3,"mcu_temp_c":18.7,"uptime_s":890877,"input_thermopile_uv":-184,"prox_v":0.0,"pilot_high_v":11.8,"pilot_low_v":11.8,"session_energy_wh":9865.300,"config_status":5,"evse_state":1,"current_alerts":[]}
Power:          0W
Current L1..L3: 0A 0.3A 0A
Voltage L1..L3: 0V 0V 0V
Charge status:  A
Enabled:        false
Charged:        9.9kWh
Duration:       0s

wallbox6
--------
[shelly] TRACE 2024/02/14 18:45:45 POST http://192.168.8.9/rpc/Shelly.GetStatus
[shelly] TRACE 2024/02/14 18:45:45 {"id":0,"on":false,"src":"evcc","method":"Shelly.GetStatus"}
--
{"ble":{},"cloud":{"connected":false},"em1:0":{"id":0,"current":0.020,"voltage":221.1,"act_power":0.0,"aprt_power":4.5,"pf":0.00, "freq":50.0,"calibration":"factory"},"em1:1":{"id":1,"current":30.647,"voltage":221.1,"act_power":6756.2,"aprt_power":6810.5,"pf":0.99, "freq":50.0,"calibration":"factory"},"em1data:0":{"id":0,"total_act_energy":0.00,"total_act_ret_energy":0.00},"em1data:1":{"id":1,"total_act_energy":9446.24,"total_act_ret_energy":2595.26},"eth":{"ip":null},"modbus":{},"mqtt":{"connected":true},"switch:0":{"id":0, "source":"HTTP_in", "output":true,"temperature":{"tC":31.5, "tF":88.7}},"sys":{"mac":"08F9E0E67204","restart_required":false,"time":"18:45","unixtime":1707932745,"uptime":30753,"ram_size":241888,"ram_free":116808,"fs_size":524288,"fs_free":208896,"cfg_rev":20,"kvs_rev":0,"schedule_rev":0,"webhook_rev":0,"available_updates":{"stable":{"version":"1.2.0"}},"reset_reason":3},"wifi":{"sta_ip":"192.168.8.9","status":"got ip","ssid":"K1","rssi":-60},"ws":{"connected":false}}
[shelly] TRACE 2024/02/14 18:45:45 POST http://192.168.8.9/rpc/Shelly.GetStatus
[... cut ...] , "source":"HTTP_in", "output":true,"temperature":{"tC":31.6, "tF":88.8}},"sys":{"mac":"08F9E0E67204","restart_required":false,"time":"18:47","unixtime":1707932870,"uptime":30877,"ram_size":241888,"ram_free":116768,"fs_size":524288,"fs_free":208896,"cfg_rev":20,"kvs_rev":0,"schedule_rev":0,"webhook_rev":0,"available_updates":{"stable":{"version":"1.2.0"}},"reset_reason":3},"wifi":{"sta_ip":"192.168.8.9","status":"got ip","ssid":"K1","rssi":-61},"ws":{"connected":false}}
Power:  0W
Energy: 0.0kWh

peterle-k, 2024-02-14:
pi@IOB-KEP:~ $ evcc -l trace charger
[main  ] INFO 2024/02/14 18:45:45 evcc 0.124.1
[main  ] INFO 2024/02/14 18:45:45 using config file: /etc/evcc.yaml
[db    ] INFO 2024/02/14 18:45:45 using sqlite database: /home/pi/.evcc/evcc.db
[db    ] TRACE 2024/02/14 18:45:45 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2024/02/14 18:45:45 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2024/02/14 18:45:45 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/02/14 18:45:45 SELECT * FROM `settings` 6 <nil>
[mqtt  ] INFO 2024/02/14 18:45:45 connecting evcc at tcp://192.168.8.164:1884
[mqtt  ] DEBUG 2024/02/14 18:45:45 tcp://192.168.8.164:1884 connected
[mqtt  ] ERROR 2024/02/14 18:45:45 tcp://192.168.8.164:1884 connection lost: EOF
[db    ] TRACE 2024/02/14 18:45:45 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="devices" -1 <nil>
[db    ] TRACE 2024/02/14 18:45:45 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="device_details" -1 <nil>
[mqtt  ] DEBUG 2024/02/14 18:45:45 tcp://192.168.8.164:1884 connected
[db    ] TRACE 2024/02/14 18:45:45 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="configs" -1 <nil>
[db    ] TRACE 2024/02/14 18:45:45 CREATE TABLE `configs` (`id` integer PRIMARY KEY AUTOINCREMENT,`class` integer,`type` text) 0 <nil>
[db    ] TRACE 2024/02/14 18:45:45 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="config_details" -1 <nil>
[db    ] TRACE 2024/02/14 18:45:45 CREATE TABLE `config_details` (`config_id` integer,`key` text,`value` text,CONSTRAINT `fk_configs_details` FOREIGN KEY (`config_id`) REFERENCES `configs`(`id`) ON DELETE CASCADE ON UPDATE CASCADE) 0 <nil>
[db    ] TRACE 2024/02/14 18:45:45 CREATE INDEX `idx_unique` ON `config_details`(`config_id`,`key`) 0 <nil>
[db    ] TRACE 2024/02/14 18:45:45 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT ""fk_devices_details"" %" OR sql LIKE "%CONSTRAINT fk_devices_details %" OR sql LIKE "%CONSTRAINT `fk_devices_details`%" OR sql LIKE "%CONSTRAINT [fk_devices_details]%" OR sql LIKE "%CONSTRAINT 	fk_devices_details	%") -1 <nil>
[db    ] TRACE 2024/02/14 18:45:45 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%""device_id"" %" OR sql LIKE "%device_id %" OR sql LIKE "%`device_id`%" OR sql LIKE "%[device_id]%" OR sql LIKE "%	device_id	%") -1 <nil>
[shelly] TRACE 2024/02/14 18:45:45 GET http://192.168.8.9/shelly
[db    ] TRACE 2024/02/14 18:45:45 SELECT * FROM `configs` WHERE `configs`.`class` = 1 ORDER BY id 0 <nil>
[shelly] TRACE 2024/02/14 18:45:45 {"name":"Wallbox Carport ","id":"shellyproem50-08f9e0e67204","mac":"08F9E0E67204","slot":0,"model":"SPEM-002CEBEU50","gen":2,"fw_id":"20240130-115202/1.2.0-beta1-gd959786","ver":"1.2.0-beta1","app":"ProEM","auth_en":false,"auth_domain":null}
wallbox5
--------
[twc3  ] TRACE 2024/02/14 18:45:45 GET http://192.168.8.124/api/1/vitals
[twc3  ] TRACE 2024/02/14 18:45:45 {"contactor_closed":false,"vehicle_connected":false,"session_s":0,"grid_v":228.8,"grid_hz":50.050,"vehicle_current_a":0.3,"currentA_a":0.0,"currentB_a":0.3,"currentC_a":0.0,"currentN_a":0.0,"voltageA_v":0.0,"voltageB_v":0.0,"voltageC_v":0.0,"relay_coil_v":11.9,"pcba_temp_c":12.0,"handle_temp_c":6.3,"mcu_temp_c":18.7,"uptime_s":890877,"input_thermopile_uv":-184,"prox_v":0.0,"pilot_high_v":11.8,"pilot_low_v":11.8,"session_energy_wh":9865.300,"config_status":5,"evse_state":1,"current_alerts":[]}
Power:          0W
Current L1..L3: 0A 0.3A 0A
Voltage L1..L3: 0V 0V 0V
Charge status:  A
Enabled:        false
Charged:        9.9kWh
Duration:       0s

wallbox6
--------
[shelly] TRACE 2024/02/14 18:45:45 POST http://192.168.8.9/rpc/Shelly.GetStatus
[shelly] TRACE 2024/02/14 18:45:45 {"id":0,"on":false,"src":"evcc","method":"Shelly.GetStatus"}
--
{"ble":{},"cloud":{"connected":false},"em1:0":{"id":0,"current":0.020,"voltage":221.1,"act_power":0.0,"aprt_power":4.5,"pf":0.00, "freq":50.0,"calibration":"factory"},"em1:1":{"id":1,"current":30.647,"voltage":221.1,"act_power":6756.2,"aprt_power":6810.5,"pf":0.99, "freq":50.0,"calibration":"factory"},"em1data:0":{"id":0,"total_act_energy":0.00,"total_act_ret_energy":0.00},"em1data:1":{"id":1,"total_act_energy":9446.24,"total_act_ret_energy":2595.26},"eth":{"ip":null},"modbus":{},"mqtt":{"connected":true},"switch:0":{"id":0, "source":"HTTP_in", "output":true,"temperature":{"tC":31.5, "tF":88.7}},"sys":{"mac":"08F9E0E67204","restart_required":false,"time":"18:45","unixtime":1707932745,"uptime":30753,"ram_size":241888,"ram_free":116808,"fs_size":524288,"fs_free":208896,"cfg_rev":20,"kvs_rev":0,"schedule_rev":0,"webhook_rev":0,"available_updates":{"stable":{"version":"1.2.0"}},"reset_reason":3},"wifi":{"sta_ip":"192.168.8.9","status":"got ip","ssid":"K1","rssi":-60},"ws":{"connected":false}}
[shelly] TRACE 2024/02/14 18:45:45 POST http://192.168.8.9/rpc/Shelly.GetStatus
[shelly] TRACE 2024/02/14 18:45:46 {"id":0,"on":false,"src":"evcc","method":"Shelly.GetStatus"}
--
{"ble":{},"cloud":{"connected":false},"em1:0":{"id":0,"current":0.020,"voltage":221.1,"act_power":0.0,"aprt_power":4.5,"pf":0.00, "freq":50.0,"calibration":"factory"},"em1:1":{"id":1,"current":30.647,"voltage":221.1,"act_power":6756.2,"aprt_power":6810.5,"pf":0.99, "freq":50.0,"calibration":"factory [... cut ...] stable":{"version":"1.2.0"}},"reset_reason":3},"wifi":{"sta_ip":"192.168.8.9","status":"got ip","ssid":"K1","rssi":-59},"ws":{"connected":false}}
[shelly] TRACE 2024/02/14 18:45:46 POST http://192.168.8.9/rpc/Switch.GetStatus
[shelly] TRACE 2024/02/14 18:45:46 {"id":0,"on":false,"src":"evcc","method":"Switch.GetStatus"}
--
{"id":0, "source":"HTTP_in", "output":true,"temperature":{"tC":31.5, "tF":88.7}}
Power:         0W
Energy:        0.0kWh
Charge status: B
Enabled:       true
Features:      []

peterle-k, 2024-02-14:
pi@IOB-KEP:~ $ evcc -l trace meter
[main  ] INFO 2024/02/14 18:47:50 evcc 0.124.1
[main  ] INFO 2024/02/14 18:47:50 using config file: /etc/evcc.yaml
[db    ] INFO 2024/02/14 18:47:50 using sqlite database: /home/pi/.evcc/evcc.db
[db    ] TRACE 2024/02/14 18:47:50 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2024/02/14 18:47:50 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2024/02/14 18:47:50 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/02/14 18:47:50 SELECT * FROM `settings` 6 <nil>
[mqtt  ] INFO 2024/02/14 18:47:50 connecting evcc at tcp://192.168.8.164:1884
[db    ] TRACE 2024/02/14 18:47:50 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="devices" -1 <nil>
[mqtt  ] DEBUG 2024/02/14 18:47:50 tcp://192.168.8.164:1884 connected
[mqtt  ] ERROR 2024/02/14 18:47:50 tcp://192.168.8.164:1884 connection lost: EOF
[db    ] TRACE 2024/02/14 18:47:50 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="device_details" -1 <nil>
[db    ] TRACE 2024/02/14 18:47:50 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="configs" -1 <nil>
[db    ] TRACE 2024/02/14 18:47:50 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "configs" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[mqtt  ] DEBUG 2024/02/14 18:47:50 tcp://192.168.8.164:1884 connected
[db    ] TRACE 2024/02/14 18:47:50 SELECT * FROM `configs` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/02/14 18:47:50 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="config_details" -1 <nil>
[db    ] TRACE 2024/02/14 18:47:50 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "config_details" AND sql IS NOT NULL order by type = "table" desc 2 <nil>
[db    ] TRACE 2024/02/14 18:47:50 SELECT * FROM `config_details` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/02/14 18:47:50 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT ""fk_configs_details"" %" OR sql LIKE "%CONSTRAINT fk_configs_details %" OR sql LIKE "%CONSTRAINT `fk_configs_details`%" OR sql LIKE "%CONSTRAINT [fk_configs_details]%" OR sql LIKE "%CONSTRAINT 	fk_configs_details	%") -1 <nil>
[db    ] TRACE 2024/02/14 18:47:50 SELECT count(*) FROM sqlite_master WHERE type = "index" AND tbl_name = "config_details" AND name = "idx_unique" -1 <nil>
[db    ] TRACE 2024/02/14 18:47:50 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT ""fk_devices_details"" %" OR sql LIKE "%CONSTRAINT fk_devices_details %" OR sql LIKE "%CONSTRAINT `fk_devices_details`%" OR sql LIKE "%CONSTRAINT [fk_devices_details]%" OR sql LIKE "%CONSTRAINT 	fk_devices_details	%") -1 <nil>
[db    ] TRACE 2024/02/14 18:47:50 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%""device_id"" %" OR sql LIKE "%device_id %" OR sql LIKE "%`device_id`%" OR sql LIKE "%[device_id]%" OR sql LIKE "%	device_id	%") -1 <nil>
[shelly] TRACE 2024/02/14 18:47:50 GET http://192.168.8.9/shelly
[shelly] TRACE 2024/02/14 18:47:50 {"name":"Wallbox Carport ","id":"shellyproem50-08f9e0e67204","mac":"08F9E0E67204","slot":0,"model":"SPEM-002CEBEU50","gen":2,"fw_id":"20240130-115202/1.2.0-beta1-gd959786","ver":"1.2.0-beta1","app":"ProEM","auth_en":false,"auth_domain":null}
[db    ] TRACE 2024/02/14 18:47:50 SELECT * FROM `configs` WHERE `configs`.`class` = 2 ORDER BY id 0 <nil>
netz
----
[modbus] TRACE 2024/02/14 18:47:50 modbus: send 00 01 00 00 00 06 00 03 9c 89 00 02
[modbus] TRACE 2024/02/14 18:47:50 modbus: recv 00 01 00 00 00 07 00 03 04 ff ff ff ff
Power: -1W

pv
--
[modbus] TRACE 2024/02/14 18:47:50 modbus: send 00 02 00 00 00 06 00 03 9c 83 00 02
[modbus] TRACE 2024/02/14 18:47:50 modbus: recv 00 02 00 00 00 07 00 03 04 00 00 00 00
[modbus] TRACE 2024/02/14 18:47:50 modbus: send 00 03 00 00 00 06 00 03 9c 8b 00 02
[modbus] TRACE 2024/02/14 18:47:50 modbus: recv 00 03 00 00 00 07 00 03 04 00 00 00 00
Power: 0W

battery
-------
[modbus] TRACE 2024/02/14 18:47:50 modbus: send 00 04 00 00 00 06 00 03 9c 85 00 02
[modbus] TRACE 2024/02/14 18:47:50 modbus: recv 00 04 00 00 00 07 00 03 04 e0 ef ff ff
[modbus] TRACE 2024/02/14 18:47:50 modbus: send 00 05 00 00 00 06 00 03 9c 92 00 01
[modbus] TRACE 2024/02/14 18:47:50 modbus: recv 00 05 00 00 00 05 00 03 02 00 40
Power: 7953W
Soc:   64%

shelly
------
[shelly] TRACE 2024/02/14 18:47:50 POST http://192.168.8.9/rpc/Shelly.GetStatus
[shelly] TRACE 2024/02/14 18:47:50 {"id":1,"on":false,"src":"evcc","method":"Shelly.GetStatus"}
--
{"ble":{},"cloud":{"connected":false},"em1:0":{"id":0,"current":0.020,"voltage":221.3,"act_power":0.0,"aprt_power":4.5,"pf":0.00, "freq":50.0,"calibration":"factory"},"em1:1":{"id":1,"current":30.665,"voltage":221.4,"act_power":6769.3,"aprt_power":6823.1,"pf":0.99, "freq":50.0,"calibration":"factory"},"em1data:0":{"id":0,"total_act_energy":0.00,"total_act_ret_energy":0.00},"em1data:1":{"id":1,"total_act_energy":9671.54,"total_act_ret_energy":2595.26},"eth":{"ip":null},"modbus":{},"mqtt":{"connected":true},"switch:0":{"id":0, "source":"HTTP_in", "output":true,"temperature":{"tC":31.6, "tF":88.8}},"sys":{"mac":"08F9E0E67204","restart_required":false,"time":"18:47","unixtime":1707932870,"uptime":30877,"ram_size":241888,"ram_free":116776,"fs_size":524288,"fs [... cut ...] , "source":"HTTP_in", "output":true,"temperature":{"tC":31.6, "tF":88.8}},"sys":{"mac":"08F9E0E67204","restart_required":false,"time":"18:47","unixtime":1707932870,"uptime":30877,"ram_size":241888,"ram_free":116768,"fs_size":524288,"fs_free":208896,"cfg_rev":20,"kvs_rev":0,"schedule_rev":0,"webhook_rev":0,"available_updates":{"stable":{"version":"1.2.0"}},"reset_reason":3},"wifi":{"sta_ip":"192.168.8.9","status":"got ip","ssid":"K1","rssi":-61},"ws":{"connected":false}}
Power:  0W
Energy: 0.0kWh

thierolm, 2024-02-14:
Ich bastel dir mal eine customer meter config, die du ausprobieren kannst. Allerdings komme ich da erst später dazu (Valentinstag ....) ...
==> D.h. es wird später oder morgen ...

thierolm, 2024-02-14:
Eine Frage noch: Wie kommt es, dass dein Meter sowohl act (Energy imported (positive value)) als auch ret (Energy exported (negative value)) energy gemessen hat?

```json
{
    "em1data:1": {
        "id": 1,
        "total_act_energy": 6644.90,
        "total_act_ret_energy": 2595.26
    }
}
```

thierolm, 2024-02-14:
Eine erste schnelle Custom Meter Config könnte wie folgt aussehen (bitte mal testen):
```yaml
meters:
  - name: shelly
    type: custom
    power:
      source: http
      uri: http://192.168.8.9/rpc/Shelly.GetStatus)
      jq: [( .["em1:0"].act_power , .["em1:1"].act_power)] | add
```

Bitte wieder Ausgabe von
`evcc -l trace meter`
posten.

thierolm, 2024-02-14:
Habe PR #12259 erstellt. Wenn @andig diesen in den Master merged kannst du testen, ob Power und Energy für deinen EM - 50 korrekt abgebildet werden. Dann benötigst du auch keine separate Meter-Definition mehr. Macht dann alles der Charger ...

peterle-k, 2024-02-15:
Die meter Definition erzeugt einen Fehler:
```shell
Feb 15 08:21:29 IOB-KEP systemd[1]: Started evcc.
Feb 15 08:21:30 IOB-KEP evcc[646792]: [main  ] INFO 2024/02/15 08:21:30 evcc 0.124.1
Feb 15 08:21:30 IOB-KEP evcc[646792]: [main  ] INFO 2024/02/15 08:21:30 using config file: /etc/evcc.yaml
Feb 15 08:21:30 IOB-KEP evcc[646792]: [main  ] INFO 2024/02/15 08:21:30 starting ui and api at :7070
Feb 15 08:21:30 IOB-KEP evcc[646792]: [main  ] FATAL 2024/02/15 08:21:30 While parsing config: yaml: line 48: did not find expected ',' or ']'
Feb 15 08:21:30 IOB-KEP evcc[646792]: [main  ] FATAL 2024/02/15 08:21:30 will attempt restart in: 5m0s
``` 
[image]

peterle-k, 2024-02-15:
so gehts:

```shell
- name: shelly  
  type: custom  
  power:  
    source: http  
    uri: http://192.168.8.9/rpc/Shelly.GetStatus
    jq:  .["em1:1"].act_power
  energy:  
    source: http  
    uri: http://192.168.8.9/rpc/Shelly.GetStatus
    jq:  .["em1data:1"].total_act_energy
    scale: 0.001  
``` 
[image]

peterle-k, 2024-02-15:
Trace:

```shell
pi@IOB-KEP:~ $ evcc -l trace meter
[main  ] INFO 2024/02/15 08:42:10 evcc 0.124.1
[main  ] INFO 2024/02/15 08:42:10 using config file: /etc/evcc.yaml
[db    ] INFO 2024/02/15 08:42:10 using sqlite database: /home/pi/.evcc/evcc.db
[db    ] TRACE 2024/02/15 08:42:10 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2024/02/15 08:42:10 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2024/02/15 08:42:10 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/02/15 08:42:10 SELECT * FROM `settings` 6 <nil>
[mqtt  ] INFO 2024/02/15 08:42:10 connecting evcc at tcp://192.168.8.164:1884
[db    ] TRACE 2024/02/15 08:42:10 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="devices" -1 <nil>
[mqtt  ] DEBUG 2024/02/15 08:42:10 tcp://192.168.8.164:1884 connected
[db    ] TRACE 2024/02/15 08:42:10 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="device_details" -1 <nil>
[db    ] TRACE 2024/02/15 08:42:10 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="configs" -1 <nil>
[mqtt  ] ERROR 2024/02/15 08:42:10 tcp://192.168.8.164:1884 connection lost: EOF
[db    ] TRACE 2024/02/15 08:42:10 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "configs" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2024/02/15 08:42:10 SELECT * FROM `configs` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/02/15 08:42:10 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="config_details" -1 <nil>
[mqtt  ] DEBUG 2024/02/15 08:42:10 tcp://192.168.8.164:1884 connected
[db    ] TRACE 2024/02/15 08:42:10 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "config_details" AND sql IS NOT NULL order by type = "table" desc 2 <nil>
[db    ] TRACE 2024/02/15 08:42:10 SELECT * FROM `config_details` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/02/15 08:42:10 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT ""fk_configs_details"" %" OR sql LIKE "%CONSTRAINT fk_configs_details %" OR sql LIKE "%CONSTRAINT `fk_configs_details`%" OR sql LIKE "%CONSTRAINT [fk_configs_details]%" OR sql LIKE "%CONSTRAINT 	fk_configs_details	%") -1 <nil>
[db    ] TRACE 2024/02/15 08:42:10 SELECT count(*) FROM sqlite_master WHERE type = "index" AND tbl_name = "config_details" AND name = "idx_unique" -1 <nil>
[db    ] TRACE 2024/02/15 08:42:10 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT ""fk_devices_details"" %" OR sql LIKE "%CONSTRAINT fk_devices_details %" OR sql LIKE "%CONSTRAINT `fk_devices_details`%" OR sql LIKE "%CONSTRAINT [fk_devices_details]%" OR sql LIKE "%CONSTRAINT 	fk_devices_details	%") -1 <nil>
[db    ] TRACE 2024/02/15 08:42:10 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%""device_id"" %" OR sql LIKE "%device_id %" OR sql LIKE "%`device_id`%" OR sql LIKE "%[device_id]%" OR sql LIKE "%	device_id	%") -1 <nil>
[db    ] TRACE 2024/02/15 08:42:10 SELECT * FROM `configs` WHERE `configs`.`class` = 2 ORDER BY id 0 <nil>
netz
----
[modbus] TRACE 2024/02/15 08:42:10 modbus: send 00 01 00 00 00 06 00 03 9c 89 00 02
[modbus] TRACE 2024/02/15 08:42:10 modbus: recv 00 01 00 00 00 07 00 03 04 00 05 00 00
Power: 5W

pv
--
[modbus] TRACE 2024/02/15 08:42:10 modbus: send 00 02 00 00 00 06 00 03 9c 83 00 02
[modbus] TRACE 2024/02/15 08:42:10 modbus: recv 00 02 00 00 00 07 00 03 04 04 02 00 00
[modbus] TRACE 2024/02/15 08:42:10 modbus: send 00 03 00 00 00 06 00 03 9c 8b 00 02
[modbus] TRACE 2024/02/15 08:42:10 modbus: recv 00 03 00 00 00 07 00 03 04 fa 15 ff ff
Power: 2541W

battery
-------
[modbus] TRACE 2024/02/15 08:42:10 modbus: send 00 04 00 00 00 06 00 03 9c 85 00 02
[modbus] TRACE 2024/02/15 08:42:10 modbus: recv 00 04 00 00 00 07 00 03 04 eb a8 ff ff
[modbus] TRACE 2024/02/15 08:42:10 modbus: send 00 05 00 00 00 06 00 03 9c 92 00 01
[modbus] TRACE 2024/02/15 08:42:10 modbus: recv 00 05 00 00 00 05 00 03 02 00 10
Power: 5208W
Soc:   16%

shelly
------
[http  ] TRACE 2024/02/15 08:42:10 GET http://192.168.8.9/rpc/Shelly.GetStatus
[http  ] TRACE 2024/02/15 08:42:10 {"ble":{},"cloud":{"connected":false},"em1:0":{"id":0,"current":0.020,"voltage":220.8,"act_power":0.0,"aprt_power":4.5,"pf":0.00, "freq":50.0,"calibration":"factory"},"em1:1":{"id":1,"current":31.250,"voltage":220.9,"act_power":6881.9,"aprt_power":6936.5,"pf":0.99, "freq":50.0,"calibration":"factory"},"em1data:0":{"id":0,"total_act_energy":0.00,"total_act_ret_energy":0.00},"em1data:1":{"id":1,"total_act_energy":12167.64,"total_act_ret_energy":2595.27},"eth":{"ip":null},"modbus":{},"mqtt":{"connected":true},"switch:0":{"id":0, "source":"HTTP_in", "output":true,"temperature":{"tC":28.6, "tF":83.5}},"sys":{"mac":"08F9E0E67204","restart_required":false,"time":"08:42","unixtime":1707982930,"uptime":80938,"ram_size":241828,"ram_free":114472,"fs_size":524288,"fs_free":208896,"cfg_rev":20,"kvs_rev":0,"schedule_rev":0,"webhook_rev":0,"available_updates":{"stable":{"version":"1.2.0"}},"reset_reason":3},"wifi":{"sta_ip":"192.168.8.9","status":"got ip","ssid":"K1","rssi":-58},"ws":{"connected":false}}
[http  ] TRACE 2024/02/15 08:42:10 GET http://192.168.8.9/rpc/Shelly.GetStatus
[... cut ...] e":"HTTP_in", "output":true,"temperature":{"tC":28.6, "tF":83.5}},"sys":{"mac":"08F9E0E67204","restart_required":false,"time":"08:42","unixtime":1707982930,"uptime":80938,"ram_size":241828,"ram_free":114460,"fs_size":524288,"fs_free":208896,"cfg_rev":20,"kvs_rev":0,"schedule_rev":0,"webhook_rev":0,"available_updates":{"stable":{"version":"1.2.0"}},"reset_reason":3},"wifi":{"sta_ip":"192.168.8.9","status":"got ip","ssid":"K1","rssi":-58},"ws":{"connected":false}}
Power:  6882W
Energy: 12.2kWh
```

peterle-k, 2024-02-15:
total_act_energy bringt die Summe der geladenen Energie. Es ist also erforderlich, dass bei ladebeginn der Shelly Speicher zurückgesetzt wird.
Das wird so gemacht.
```
http://192.168.8.9/rpc/EM1Data.DeleteAllData?id=1
``` 
Wie integriere ich das in yaml?

thierolm, 2024-02-15:
Das brauchst du nicht, du musst das shelly meter an deinem Load-Point zusätzlich als Meter definieren. Die Ladeleistung ermittelt evcc dann selbstständig am Laodpoint.

thierolm, 2024-02-15:
Das musst du nicht zurücksetzen. Die Geladen-Anzeige am Loadpoint wird von evcc über die mitgeschriebenen Power / Energy Werte errechnet.