evcc-io/evcc #12253: Aufzeichnung zu Ladevorgang geht verloren bei Server Neustart

An-Wi, 2024-02-14:
### Describe the bug

Wird während eines Ladevorgang der (Windows)-Server, auf dem evcc läuft, aus irgendeinem Grund neu gestartet (bspw. Windows Updates in der Nacht), dann gehen die Aufzeichnungen zum aktiven Ladevorgang komplett verloren! 
Nach dem Neustart wird ein neuer Ladevorgang begonnen.

[image]

### Steps to reproduce

1. Ladevorgang starten (möglicherweise nicht zwangsweise über Ladeplanung)
2. Server, auf dem evcc läuft, neustarten ohne evcc vorher zu beenden
3. evcc startet automatisch über "Aufgabenplanung"
4. Letzter Ladevorgang nicht vorhanden!

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
```

### Log details

```text
[site  ] DEBUG 2024/02/14 03:35:05 ----
[lp-1  ] DEBUG 2024/02/14 03:35:05 charge power: 10586W
[site  ] DEBUG 2024/02/14 03:35:05 pv power: 0W
[site  ] DEBUG 2024/02/14 03:35:05 grid power: 10863W
[site  ] DEBUG 2024/02/14 03:35:05 grid currents: [15.6 15.3 15.4]A
[site  ] DEBUG 2024/02/14 03:35:05 site power: 10663W
[lp-1  ] DEBUG 2024/02/14 03:35:05 charge currents: [15.2 15.1 15.2]A
[lp-1  ] DEBUG 2024/02/14 03:35:05 detected active phases: 3p
[lp-1  ] DEBUG 2024/02/14 03:35:05 charge total import: 3001.074kWh
[lp-1  ] DEBUG 2024/02/14 03:35:05 charger status: C
[lp-1  ] DEBUG 2024/02/14 03:35:05 soc estimated: 87.49% (vehicle: 87.00%)
[lp-1  ] DEBUG 2024/02/14 03:35:05 vehicle soc: 87%
[lp-1  ] ERROR 2024/02/14 03:35:05 vehicle range: not available
[lp-1  ] DEBUG 2024/02/14 03:35:05 plan: charge 14m16s between 2024-02-14 03:35:06 +0100 CET until 2024-02-14 04:00:00 +0100 CET (power: 11040W, avg cost: 0.280)
[site  ] DEBUG 2024/02/14 03:35:25 ----
[lp-1  ] DEBUG 2024/02/14 03:35:25 charge power: 10584W
[site  ] DEBUG 2024/02/14 03:35:25 pv power: 0W
[site  ] DEBUG 2024/02/14 03:35:25 grid power: 10867W
[site  ] DEBUG 2024/02/14 03:35:25 grid currents: [15.6 15.3 15.4]A
[site  ] DEBUG 2024/02/14 03:35:25 site power: 10667W
[lp-1  ] DEBUG 2024/02/14 03:35:25 charge currents: [15.2 15.1 15.1]A
[lp-1  ] DEBUG 2024/02/14 03:35:25 detected active phases: 3p
[lp-1  ] DEBUG 2024/02/14 03:35:25 charge total import: 3001.131kWh
[lp-1  ] DEBUG 2024/02/14 03:35:25 charger status: C
[lp-1  ] DEBUG 2024/02/14 03:35:25 soc estimated: 87.55% (vehicle: 87.00%)
[lp-1  ] DEBUG 2024/02/14 03:35:25 vehicle soc: 88%
[lp-1  ] ERROR 2024/02/14 03:35:25 vehicle range: not available
[lp-1  ] DEBUG 2024/02/14 03:35:25 plan: charge 13m58s between 2024-02-14 03:35:25 +0100 CET until 2024-02-14 04:00:00 +0100 CET (power: 11040W, avg cost: 0.280)
[site  ] DEBUG 2024/02/14 03:35:45 ----
[lp-1  ] DEBUG 2024/02/14 03:35:45 charge power: 10588W
[site  ] DEBUG 2024/02/14 03:35:45 pv power: 0W
[site  ] DEBUG 2024/02/14 03:35:45 grid power: 10847W
[site  ] DEBUG 2024/02/14 03:35:45 grid currents: [15.5 15.3 15.4]A
[site  ] DEBUG 2024/02/14 03:35:45 site power: 10647W
[lp-1  ] DEBUG 2024/02/14 03:35:45 charge currents: [15.2 15.1 15.1]A
[lp-1  ] DEBUG 2024/02/14 03:35:45 detected active phases: 3p
[lp-1  ] DEBUG 2024/02/14 03:35:45 charge total import: 3001.189kWh
[lp-1  ] DEBUG 2024/02/14 03:35:45 charger status: C
[lp-1  ] DEBUG 2024/02/14 03:35:45 soc estimated: 87.61% (vehicle: 87.00%)
[lp-1  ] DEBUG 2024/02/14 03:35:45 vehicle soc: 88%
[lp-1  ] ERROR 2024/02/14 03:35:45 vehicle range: not available
[lp-1  ] DEBUG 2024/02/14 03:35:45 plan: charge 13m39s between 2024-02-14 03:35:46 +0100 CET until 2024-02-14 04:00:00 +0100 CET (power: 11040W, avg cost: 0.280)
[site  ] DEBUG 2024/02/14 03:36:05 ----
[lp-1  ] DEBUG 2024/02/14 03:36:05 charge power: 10587W
[site  ] DEBUG 2024/02/14 03:36:05 pv power: 0W
[site  ] DEBUG 2024/02/14 03:36:05 grid power: 10869W
[site  ] DEBUG 2024/02/14 03:36:05 grid currents: [15.5 15.3 15.5]A
[... 855 lines cut ...]
[lp-1  ] ERROR 2024/02/14 04:00:02 vehicle range: not available
[lp-1  ] DEBUG 2024/02/14 04:00:02 limitSoc reached: 90.0% > 90%
[lp-1  ] DEBUG 2024/02/14 04:00:02 charger: guard elapse
[lp-1  ] DEBUG 2024/02/14 04:00:02 guard timer inactive
[site  ] DEBUG 2024/02/14 04:00:18 ----
[lp-1  ] DEBUG 2024/02/14 04:00:18 charge power: 0W
[site  ] DEBUG 2024/02/14 04:00:18 pv power: 0W
[site  ] DEBUG 2024/02/14 04:00:18 grid power: 141W
[site  ] DEBUG 2024/02/14 04:00:18 grid currents: [0.401 0.376 0.572]A
[site  ] DEBUG 2024/02/14 04:00:18 site power: -59W
[lp-1  ] DEBUG 2024/02/14 04:00:18 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2024/02/14 04:00:18 charge total import: 3002.986kWh
[lp-1  ] DEBUG 2024/02/14 04:00:18 charger status: B
[lp-1  ] DEBUG 2024/02/14 04:00:18 next soc poll remaining time: 59m42s
[lp-1  ] DEBUG 2024/02/14 04:00:18 limitSoc reached: 90.0% > 90%
[site  ] DEBUG 2024/02/14 04:00:38 ----
```

### What type of operating system are you running?

Linux

### Version

v0.124.1

andig, 2024-02-14:
Um zu sehen, was in der DB landet oder nicht braucht es `db: trace`. Bitte melden wenn es ein passendes Log gibt.

closed 2024-02-14 by andig

An-Wi, 2024-02-15:
Ok, kein Problem - dachte ich zunächst, weil mir das Verhalten im letzten Jahr schon ein paar Mal aufgefallen war.
**_ABER:_** es scheint differenzierter zu sein...

Habe soeben 2 verschiedene Tests gemacht ("PV+Min"- und "Plan"-Laden) mit jeweiligem Server-Neustart dazwischen.
Alle 3 Ladevorgänge sind vollständig aufgezeichnet, so wie es sein soll:
(Vorgang 1 -> Neustart -> Vorgang 2 -> Neustart -> Vorgang 3) ... perfekt
[image]
Server-Neustarts waren **08:50:00** und **09:00:00**.

Ich habe `db: trace` jetzt dauerhaft an. Sobald ich das Phänomen erneut entdecke, packe ich das Log hier rein.