evcc-io/evcc #12264: Vestel EVC04 wrong decimal offset for total charge import

plieven, 2024-02-15:
### Describe the bug

The total kWh consumed by the wallbox carry a wrong decimal offset. EVCC reports 117.600 kWh while 11760.0 would be correct. Maybe the decimal offset of the mobbus register is incorrect.

### Steps to reproduce

Connect Vestel EVC04 Wallbox to evcc via Modbus TCP.

### Configuration details

```yaml
chargers:
- type: template
  template: vestel 
  id: 255  
  host: 192.168.178.137  
  port: 502  
  modbus: tcpip  
  name: wallbox1
```

### Log details

```text
[main  ] INFO 2024/02/15 10:13:51 evcc 0.124.2 (f564a9b5)
[main  ] INFO 2024/02/15 10:13:51 using config file: /etc/evcc.yaml
[main  ] INFO 2024/02/15 10:13:51 starting ui and api at :7070
[db    ] INFO 2024/02/15 10:13:51 using sqlite database: /root/.evcc/evcc.db
[lp-1  ] DEBUG 2024/02/15 10:13:51 charge total import: 117.600kWh
[site  ] INFO 2024/02/15 10:13:51 site config:
[site  ] INFO 2024/02/15 10:13:51   meters:      grid ✓ pv ✓ battery ✗
[site  ] INFO 2024/02/15 10:13:51     grid:      power ✓ energy ✗ currents ✗
[site  ] INFO 2024/02/15 10:13:51     pv 1:      power ✓ energy ✗ currents ✗
[site  ] INFO 2024/02/15 10:13:51     pv 2:      power ✓ energy ✗ currents ✗
[lp-1  ] INFO 2024/02/15 10:13:51 loadpoint 1:
[lp-1  ] INFO 2024/02/15 10:13:51   mode:        off
[lp-1  ] INFO 2024/02/15 10:13:51   charger:     power ✓ energy ✓ currents ✓ phases ✗ wakeup ✗
[lp-1  ] INFO 2024/02/15 10:13:51   meters:      charge ✓
[lp-1  ] INFO 2024/02/15 10:13:51     charge:    power ✓ energy ✓ currents ✓
[lp-1  ] DEBUG 2024/02/15 10:13:51 phase timer inactive
[lp-1  ] DEBUG 2024/02/15 10:13:51 pv timer inactive
[site  ] DEBUG 2024/02/15 10:13:51 ----
[lp-1  ] DEBUG 2024/02/15 10:13:51 charge power: 5W
[site  ] DEBUG 2024/02/15 10:13:52 pv power: 803W
[site  ] DEBUG 2024/02/15 10:13:52 grid power: 1892W
[site  ] DEBUG 2024/02/15 10:13:52 site power: 1892W
[lp-1  ] DEBUG 2024/02/15 10:13:52 charge voltages: [229 233 233]V
[lp-1  ] DEBUG 2024/02/15 10:13:52 detected connected phases: 3p
[lp-1  ] DEBUG 2024/02/15 10:13:52 charge currents: [0.063 0.018 0.018]A
[lp-1  ] DEBUG 2024/02/15 10:13:52 charge total import: 117.600kWh
[lp-1  ] DEBUG 2024/02/15 10:13:52 charger status: B
[lp-1  ] INFO 2024/02/15 10:13:52 car connected
```

### What type of operating system are you running?

Linux

### Version

0.124.2 (f564a9b5)

andig, 2024-02-15:
> EVCC reports 117.600 kWh while 11760.0 would be correct

Are you saying that it is by factor 10 to large?

plieven, 2024-02-15:
Assuming that the „.“ (dot) is the decimal separator, it is by factor 100 too small.Looking at the Modbus specificaton [1] of the box the unit of the meter register is 0.1kWh. Maybe the author of the plugin believed it was Wh as in the register for the session energy.As the vestel box is a sponsor token feature I think there is no public source for the plugin, right? Otherwise I could check this myself and send a PR.[1] https://api.library.loxone.com/downloader/file/722/EVC04_Modbus_RTU_Specification_v2_12.04.2021.pdfVon meinem iPhone gesendet

andig, 2024-02-15:
Looks like the implementation is correct, result in kWh:

```go
// TotalEnergy implements the api.MeterEnergy interface
func (wb *Vestel) TotalEnergy() (float64, error) {
	b, err := wb.conn.ReadInputRegisters(vestelRegTotalEnergy, 2)
	if err != nil {
		return 0, err
	}

	return float64(binary.BigEndian.Uint32(b)) / 10, err
}

```

plieven, 2024-02-15:
Guess I have to check whats the raw output from the box. Can you recommend a tool to query Modbus/TCP from the cli?

premultiply, 2024-02-15:
`evcc meter --log trace`

plieven, 2024-02-15:
evcc charger --log trace

it seems that the higher 16-bits are always 0 when querying 32 bit 
registers:

[vestel] TRACE 2024/02/15 21:42:04 modbus: send 00 01 00 00 00 06 ff 03 
07 d2 00 01
[vestel] TRACE 2024/02/15 21:42:04 modbus: recv 00 01 00 00 00 05 ff 03 
02 00 00
[vestel] TRACE 2024/02/15 21:42:04 modbus: send 00 02 00 00 00 06 ff 04 
03 fc 00 02
[vestel] TRACE 2024/02/15 21:42:04 modbus: recv 00 02 00 00 00 07 ff 04 
04 00 00 0c f0
[vestel] TRACE 2024/02/15 21:42:04 modbus: send 00 03 00 00 00 06 ff 04 
04 0c 00 02
[vestel] TRACE 2024/02/15 21:42:04 modbus: recv 00 03 00 00 00 07 ff 04 
04 00 00 04 99
[vestel] TRACE 2024/02/15 21:42:04 modbus: send 00 04 00 00 00 06 ff 04 
03 f0 00 01
[vestel] TRACE 2024/02/15 21:42:04 modbus: recv 00 04 00 00 00 05 ff 04 
02 39 73
[vestel] TRACE 2024/02/15 21:42:04 modbus: send 00 05 00 00 00 06 ff 04 
03 f2 00 01
[vestel] TRACE 2024/02/15 21:42:04 modbus: recv 00 05 00 00 00 05 ff 04 
02 00 12
[vestel] TRACE 2024/02/15 21:42:04 modbus: send 00 06 00 00 00 06 ff 04 
03 f4 00 01
[vestel] TRACE 2024/02/15 21:42:04 modbus: recv 00 06 00 00 00 05 ff 04 
02 00 12
[vestel] TRACE 2024/02/15 21:42:04 modbus: send 00 07 00 00 00 06 ff 04 
03 f6 00 01
[vestel] TRACE 2024/02/15 21:42:04 modbus: recv 00 07 00 00 00 05 ff 04 
02 00 e3
[vestel] TRACE 2024/02/15 21:42:04 modbus: send 00 08 00 00 00 06 ff 04 
03 f8 00 01
[vestel] TRACE 2024/02/15 21:42:04 modbus: recv 00 08 00 00 00 05 ff 04 
02 00 f1
[vestel] TRACE 2024/02/15 21:42:04 modbus: send 00 09 00 00 00 06 ff 04 
03 fa 00 01
[vestel] TRACE 2024/02/15 21:42:04 modbus: recv 00 09 00 00 00 05 ff 04 
02 00 ef
[vestel] TRACE 2024/02/15 21:42:04 modbus: send 00 0a 00 00 00 06 ff 04 
03 ec 00 01
[vestel] TRACE 2024/02/15 21:42:04 modbus: recv 00 0a 00 00 00 05 ff 04 
02 00 03
[vestel] TRACE 2024/02/15 21:42:04 modbus: send 00 0b 00 00 00 06 ff 04 
03 e9 00 01
[vestel] TRACE 2024/02/15 21:42:04 modbus: recv 00 0b 00 00 00 05 ff 04 
02 00 01
[vestel] TRACE 2024/02/15 21:42:04 modbus: send 00 0c 00 00 00 06 ff 04 
03 ec 00 01
[vestel] TRACE 2024/02/15 21:42:04 modbus: recv 00 0c 00 00 00 05 ff 04 
02 00 03
[vestel] TRACE 2024/02/15 21:42:04 modbus: send 00 0d 00 00 00 06 ff 04 
03 e9 00 01
[vestel] TRACE 2024/02/15 21:42:04 modbus: recv 00 0d 00 00 00 05 ff 04 
02 00 01
[vestel] TRACE 2024/02/15 21:42:04 modbus: send 00 0e 00 00 00 06 ff 04 
05 de 00 02
[vestel] TRACE 2024/02/15 21:42:04 modbus: recv 00 0e 00 00 00 07 ff 04 
04 00 00 01 1e
[vestel] TRACE 2024/02/15 21:42:04 modbus: send 00 0f 00 00 00 06 ff 04 
05 e4 00 02
[vestel] TRACE 2024/02/15 21:42:04 modbus: recv 00 0f 00 00 00 07 ff 04 
04 00 00 01 45

I will ask Vestel for a firmware update.Message ID: 
***@***.***>