evcc-io/evcc #12268: Polestar with 0.124.2: Error messages "User not authenticated" but working

msp23, 2024-02-15:
### Describe the bug

After Update to 124.2 the SoC and API Information from Polestart are visible in evcc again. Great! Love it!
I only get a lot of ERROR and WARINGS. Again all information in sync with polestar web/app and up to date. 

elli Wallbox TG: vehicle range: Message: Post \"https://pc-api.polestar.com/eu-north-1/mystar-v2\": Message: User not authenticated, Locations: [], Extensions: map[code:UNAUTHENTICATED exception:map[locations:[map[column:49 line:1]] message:User not authenticated path:[refreshAuthToken] stacktrace:[AuthenticationError: User not authenticated at ULi (/var/task/src/functions/auth/handler.js:1322:2695784) at GLi (/var/task/src/functions/auth/handler.js:1322:2697052) at refreshAuthToken (/var/task/src/functions/auth/handler.js:1322:2724075) at Lbr.r.resolve (/var/task/src/functions/auth/handler.js:288:6635) at WEe (/var/task/src/functions/auth/handler.js:84:5200) at x3 (/var/task/src/functions/auth/handler.js:84:4784) at cnr (/var/task/src/functions/auth/handler.js:84:4308) at jEe (/var/task/src/functions/auth/handler.js:84:2354) at f (/var/task/src/functions/auth/handler.js:288:41070) at Vfr (/var/task/src/functions/auth/handler.js:288:40241)]] serviceName:auth], Locations: [], Extensions: map[code:request_error]

elli Wallbox TG: vehicle soc: Message: Post \"https://pc-api.polestar.com/eu-north-1/mystar-v2\": Message: User not authenticated, Locations: [], Extensions: map[code:UNAUTHENTICATED exception:map[locations:[map[column:49 line:1]] message:User not authenticated path:[refreshAuthToken] stacktrace:[AuthenticationError: User not authenticated at ULi (/var/task/src/functions/auth/handler.js:1322:2695784) at GLi (/var/task/src/functions/auth/handler.js:1322:2697052) at refreshAuthToken (/var/task/src/functions/auth/handler.js:1322:2724075) at Lbr.r.resolve (/var/task/src/functions/auth/handler.js:288:6635) at WEe (/var/task/src/functions/auth/handler.js:84:5200) at x3 (/var/task/src/functions/auth/handler.js:84:4784) at cnr (/var/task/src/functions/auth/handler.js:84:4308) at jEe (/var/task/src/functions/auth/handler.js:84:2354) at f (/var/task/src/functions/auth/handler.js:288:41070) at Vfr (/var/task/src/functions/auth/handler.js:288:40241)]] serviceName:auth], Locations: [], Extensions: map[code:request_error] (ignored by estimator)

### Steps to reproduce

1. update to 0.124.2
2.
3.
...

### Configuration details

```yaml
.
```

### Log details

```text
.
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.124.2

andig, 2024-02-15:
> Error messages "User not authenticated" but working

Funny. Could you share credentials with info@evcc.io? Are you using 2FA?

demichve, 2024-02-16:
Same problem. SoC seems to work, range is not updated anymore.
And a lot of these unauthenticated messages.

andig, 2024-02-16:
I cannot repro this. Please send credentials to info@evcc.io

OFre12, 2024-02-16:
Exactly as Wolfgang Sh described it.
Unfortunately, the SoC does not update when charging.

andig, 2024-02-16:
Any amount of comments will not make a difference. We need a full trace log from start to problem.

andig, 2024-02-17:
As mentioned above: needs polestar at trace…

msp23, 2024-02-17:
Will provide Polestar login credentials to info @ tomorrow.

demichve, 2024-02-17:
And I will provide the Polestar trace. Most probably tomorrow morning, I have to wait until tomorrow to reproduce it.

demichve, 2024-02-18:
@andig 
Here you can find the trace for polestar.
Feb 17, 19:09 it worked.
When I connected my polestar again this morning (Feb 18, 8:23) without restarting evcc, the error appears.

`Feb 17 19:09:28 openhabian systemd[1]: Started evcc.
Feb 17 19:09:28 openhabian evcc[16766]: [main  ] INFO 2024/02/17 19:09:28 evcc 0.124.2
Feb 17 19:09:28 openhabian evcc[16766]: [main  ] INFO 2024/02/17 19:09:28 using config file: /etc/evcc.yaml
Feb 17 19:09:36 openhabian evcc[16766]: [polestar] TRACE 2024/02/17 19:09:36 GET https://polestarid.eu.polestar.com/as/authorization.oauth2?access_type=offline&client_id=polmystar&redirect_uri=https%3A%2F%2Fwww.polestar.com%2Fsign-in-callback&response_type=code&scope=openid+profile+email+customer%3Aattributes&state=HMEHklav4Ans0jvD
Feb 17 19:09:36 openhabian evcc[16766]: [polestar] TRACE 2024/02/17 19:09:36 POST https://polestarid.eu.polestar.com/as/WZaqLEQxaY/resume/as/authorization.ping?client_id=polmystar
Feb 17 19:09:36 openhabian evcc[16766]: [polestar] TRACE 2024/02/17 19:09:36 pf.pass=***&pf.username=***
Feb 17 19:09:36 openhabian evcc[16766]: [polestar] TRACE 2024/02/17 19:09:36 POST https://pc-api.polestar.com/eu-north-1/auth
Feb 17 19:09:37 openhabian evcc[16766]: [polestar] TRACE 2024/02/17 19:09:37 {"query":"query getAuthToken($code:String!){getAuthToken(code: $code){access_token,refresh_token,expires_in}}","variables":{"code":"NrNG7sKHGl2mOnHjrsS_1IcK-IXKtpXWK9AqUQaD"},"operationName":"getAuthToken"}
Feb 17 19:09:37 openhabian evcc[16766]: --
Feb 17 19:09:37 openhabian evcc[16766]: {"data":{"getAuthToken":{"access_token":"eyJhbGciOiJSUzI1NiIsImtpZCI6IkdjcTBoZ1NVWGRoLW9SYklBbW05MmFncVNmOF9SUzI1NiIsInBpLmF0bSI6ImwifQ.eyJzY29wZSI6Im9wZW5pZCBwcm9maWxlIGVtYWlsIGN1c3RvbWVyOmF0dHJpYnV0ZXMiLCJhdXRob3JpemF0aW9uX2RldGFpbHMiOltdLCJjbGllbnRfaWQiOiJwb2xteXN0YXIiLCJncm50aWQiOiI2Y3hUVG94ZkhsZjBNb1NEQ1cyMWkwVkdiT3RxZkZVMCIsImlzcyI6Imh0dHBzOi8vcG9sZXN0YXJpZC5ldS5wb2xlc3Rhci5jb20iLCJhdWQiOiJwb2xteXN0YXIiLCJhY3IiOiJ1cm46dm9sdm9pZDphYWw6YnJvbnplOmFueSIsInN1YiI6ImY5ZmJmYjhiLTU2MTItNDY1ZS04ZGE4LWIzZGY0MmU0NzVmMyIsImV4cCI6MTcwODE5Njk3N30.cpud_Qetnt_ev0D3MmS9wcV9WTH0SAkuKrUnVC0yGEfCuB-4ajnKwQWkV4U6ZuaHIPXrdDO64f7oeGn1H_hY92SrmPSamAYrX-zgG_DTL26hgJjgiaYe8f8xfNL8cIFJGVPZ0sr9UfsNyJb7lGE7cdpl1Qekr1-2qvv0qAdeWO3aHPLnDxDKkbrEF4pqKlMnlMiXWU7cTYsI1LDHUtwEgekvuqxVqJf7CRoV-31JTldbSkKMT1LYAxBvW_wB8SOA_1YLQdPYPVrxesHuATFGiPU6la4ZI6dTY5tiIYOgObSCsOkFiD9PCWkiojc8Yx0S3Z9OzByDgw273Bbq-zguQA","refresh_token":"EnWxu4gieI3lfmf3daxWVg49Q9bu3hyIrYhoPnjI3Y","expires_in":3599}}}
Feb 17 19:09:37 openhabian evcc[16766]: [polestar] TRACE 2024/02/17 19:09:37 POST https://pc-api.polestar.com/eu-north-1/my-star
Feb 17 19:09:38 openhabian evcc[16766]: [polestar] TRACE 2024/02/17 19:09:38 {"query":"query getCars{getConsumerCarsV2{vin,internalVehicleIdentifier}}","operationName":"getCars"}
Feb 17 19:09:38 openhabian evcc[16766]: --
Feb 17 19:09:38 openhabian evcc[16766]: {"data":{"getConsumerCarsV2":[{"vin":"***","internalVehicleIdentifier":"b566965d-2a89-48f5-b89c-6cd98332fb10"}]}}
Feb 17 19:09:39 openhabian evcc[16766]: [polestar] TRACE 2024/02/17 19:09:39 POST https://pc-api.polestar.com/eu-north-1/mystar-v2
Feb 17 19:09:39 openhabian evcc[16766]: [polestar] TRACE 2024/02/17 19:09:39 {"query":"query GetBatteryData($vin:String!){getBatteryData(vin: $vin){batteryChargeLevelPercentage,chargerConnectionStatus,chargingStatus,estimatedChargingTimeToFullMinutes,estimatedDistanceToEmptyKm,eventUpdatedTimestamp{iso}}}","variables":{"vin":"***"},"operationName":"GetBatteryData"}
Feb 17 19:09:39 openhabian evcc[16766]: --
Feb 17 19:09:39 openhabian evcc[16766]: {"data":{"getBatteryData":{"batteryChargeLevelPercentage":67.0,"chargerConnectionStatus":"CHARGER_CONNECTION_STATUS_DISCONNECTED","chargingStatus":"CHARGING_STATUS_IDLE","estimatedChargingTimeToFullMinutes":1320,"estimatedDistanceToEmptyKm":310,"eventUpdatedTimestamp":{"iso":"2024-02-17T17:00:27.000Z"}}}}
Feb 17 19:10:58 openhabian evcc[16766]: [polestar] TRACE 2024/02/17 19:10:58 POST https://pc-api.polestar.com/eu-north-1/mystar-v2
Feb 17 19:10:59 openhabian evcc[16766]: [polestar] TRACE 2024/02/17 19:10:59 {"query":"query GetBatteryData($vin:String!){getBatteryData(vin: $vin){batteryChargeLevelPercentage,chargerConnectionStatus,chargingStatus,estimatedChargingTimeToFullMinutes,estimatedDistanceToEmptyKm,eventUpdatedTimestamp{iso}}}","variables":{"vin":"***"},"operationName":"GetBatteryData"}
Feb 17 19:10:59 openhabian evcc[16766]: --
Feb 17 19:10:59 openhabian evcc[16766]: {"data":{"getBatteryData":{"batteryChargeLevelPercentage":67.0,"chargerConnectionStatus":"CHARGER_CONNECTION_STATUS_DISCONNECTED","chargingStatus":"CHARGING_STATUS_IDLE","estimatedChargingTimeToFullMinutes":1320,"estimatedDistanceToEmptyKm":310,"eventUpdatedTimestamp":{"iso":"2024-02-17T17:00:27.000Z"}}}}
Feb 17 19:11:58 openhabian evcc[16766]: [polestar] TRACE 2024/02/17 19:11:58 POST https://pc-api.polestar.com/eu-north-1/mystar-v2
Feb 17 19:11:59 openhabian evcc[16766]: [polestar] TRACE 2024/02/17 19:11:59 {"query":"query GetBatteryData($vin:String!){getBatteryData(vin: $vin){batteryChargeLevelPercentage,chargerConnectionStatus,chargingStatus,estimatedChargingTimeToFullMinutes,estimatedDistanceToEmptyKm,eventUpdatedTimestamp{iso}}}","variables":{"vin":"***"},"operationName":"GetBatteryData"}
Feb 17 19:11:59 openhabian evcc[16766]: --
[... cut ...] .resolve (/var/task/src/functions/auth/handler.js:288:6635)     at WEe (/var/task/src/functions/auth/handler.js:84:5200)     at x3 (/var/task/src/functions/auth/handler.js:84:4784)     at cnr (/var/task/src/functions/auth/handler.js:84:4308)     at jEe (/var/task/src/functions/auth/handler.js:84:2354)     at f (/var/task/src/functions/auth/handler.js:288:41070)     at Vfr (/var/task/src/functions/auth/handler.js:288:40241)]] serviceName:auth], Locations: [], Extensions: map[code:request_error]
`

andig, 2024-02-18:
Will take a look. Maybe the refresh token expires, too?

WolfgangSh, 2024-02-18:
new test: 
8:50 reset evcc, start loading in min+pv (no sun) all ok
        stop loading
9:30 start loading again, all ok
        stop loading
10:00 start loading again -> errors occures

andig, 2024-02-18:
Seems that a token refresh can fail:

```
Feb 18 08:23:39 openhabian evcc[16766]: [polestar] TRACE 2024/02/18 08:23:39 {"query":"query refreshAuthToken($token:String!){refreshAuthToken(token: $token){access_token,refresh_token,expires_in}}","variables":{"token":"EnWxu4gieI3lfmf3daxWVg49Q9bu3hyIrYhoPnjI3Y"},"operationName":"refreshAuthToken"}
Feb 18 08:23:39 openhabian evcc[16766]: --
Feb 18 08:23:39 openhabian evcc[16766]: {"errors":[{"message":"User not authenticated","path":["refreshAuthToken"],"extensions":{"code":"UNAUTHENTICATED","exception":{"stacktrace":["AuthenticationError: User n
```

In that case we'll just need to login again. Will prepare a PR.

andig, 2024-02-26:
Kann morgen im Nightly getestet werden!

WolfgangSh, 2024-02-27:
hallo andig, falls eine Rückmeldung interessant ist:
Der Fehler kam bisher verläßlich nach Ablauf des AuthToken
expiers_in=3599 nach Neustart von evcc, sobald eine Ladeaktivität
gestartet wurde. Mit der nightly-Version tritt er bei mir nicht mehr auf 🙂.

Danke für die investierte Zeit und viel Erfolg euch weiter!
Wolfgang