evcc-io/evcc #12272: max charge current 6A: vehicle not capable of current control

peterle-k, 2024-02-15:
### Describe the bug

Ich bin mir nicht sicher ob das ein Bug ist: 
Wenn man den Min. Ladestrom auf 6 A stellt:
[image]
dann kommt über die Abfrage 
```
sudo journalctl -fau evcc

Feb 15 14:28:47 IOB-KEP evcc[729207]: [lp-1  ] ERROR 2024/02/15 14:28:47 max charge current 6A: vehicle not capable of current control
``` 
dieser Error und es wird nicht geladen. 

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: info
levels:
#  site: debug
#  lp-2: debug
#  ioniq: trace
  cache: error

# unique installation id
plant: b2f71f52f022982de7773b21abc7934ed502dea2271d1b958f374a2ef9d21b39

interval: 10s # control cycle interval

meters:
# Netz
- type: template
  template: e3dc
  usage: grid
  host: 192.168.8.210
  port: 11516
  name: netz
# PV Leistung
- type: template
  template: e3dc
  usage: pv
  host: 192.168.8.210
  port: 11516
  name: pv
# Batterie
- type: template
  template: e3dc
  usage: battery
  host: 192.168.8.210
  port: 11516
  name: battery
# Shelly pro 50em im Carport
- name: shellyem50
  type: custom
  power:
    source: http
    uri: http://192.168.8.9/rpc/Shelly.GetStatus
    jq:  .["em1:1"].act_power
[... 176 lines cut ...]
[lp-1  ] DEBUG 2024/02/15 14:38:43 site power -8365W <= 0W enable threshold
[lp-1  ] DEBUG 2024/02/15 14:38:43 pv enable in 0s
[lp-1  ] DEBUG 2024/02/15 14:38:43 pv enable timer elapsed
[lp-1  ] ERROR 2024/02/15 14:38:43 max charge current 6A: vehicle not capable of current control
[site  ] DEBUG 2024/02/15 14:38:53 ----
[lp-1  ] DEBUG 2024/02/15 14:38:53 charge power: 0W
[lp-2  ] DEBUG 2024/02/15 14:38:53 charge power: 2W
[site  ] DEBUG 2024/02/15 14:38:53 pv power: 9464W
[site  ] DEBUG 2024/02/15 14:38:53 battery soc: 99%
[site  ] DEBUG 2024/02/15 14:38:53 battery power: 0W
[site  ] DEBUG 2024/02/15 14:38:53 grid power: -8739W
[site  ] DEBUG 2024/02/15 14:38:53 site power: -8439W
[lp-2  ] DEBUG 2024/02/15 14:38:53 vehicle odometer: 53778km
[lp-2  ] DEBUG 2024/02/15 14:38:53 charge total import: 0.348kWh
[lp-2  ] DEBUG 2024/02/15 14:38:53 charger status: B
[lp-2  ] DEBUG 2024/02/15 14:38:53 pv charge current: 36.7A = 0A + 36.7A (-8439W @ 1p)
```

### What type of operating system are you running?

Linux

### Version

0.124.2

litithio, 2024-02-15:
Nein, das ist kein Bug. Da Tesla das API geändert hat, ist es derzeit noch nicht möglich, die Ladung von Teslas am TWC3 zu steuern, da das Auto in evcc noch nicht steuerbar ist. Ist aber in Arbeit.

peterle-k, 2024-02-15:
oha ... dann muss ich auf die alte Version zurükgehen, da hats ja getan

andig, 2024-02-15:
https://github.com/evcc-io/evcc/issues/11846

closed 2024-02-15 by andig