evcc-io/evcc #12275: UI: Darstellung Balken Eigenverbrauch

fuxae, 2024-02-15:
### Describe the bug

Hallo, 

@naltatis mir ist aufgefallen, dass der Balken im UI des Eigenverbrauchs abgeschnitten wird, wenn das Verhältnis zwischen Eigenverbrauch und Einspeisung zu groß wird. Die Zahl wird dann abgeschnitten. Siehe Screenshot. 

Getestet unter Chrome, Samsung Browser und auch die HA App. Überall das gleiche Verhalten. 

Es würde eventuell Sinn machen dem Bereich eine minimale Breite zu geben, sodass es zu keiner Abschneidung kommt, wenn das Verhältnis zu groß wird?

Gruß Fuxae

[image]

### Steps to reproduce

N/A

### Configuration details

```yaml
N/A
```

### Log details

```text
N/A
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.124.1

naltatis, 2024-02-15:
Ok, interessant. Eigentlich wird die Zahl ausgeblendet, wenn nicht mehr genug Platz da ist. Mindestbreite machen wir nicht, da sonst das Verhältnis nicht mehr passt. Ich schau mal rein. Alles auf Android, richtig?

fuxae, 2024-02-15:
Ausblenden wäre natürlich auch eine Möglichkeit, ja. Sieht nur so etwas komisch aus. 

Korrekt, alles auf Android.

naltatis, 2024-02-15:
Wert-Verstecken wenn zu klein war in der Tat gar nicht eingebau. Hätte schwören können, dass wir das hatten. Aber gut. Vmtl verwechsele ich das mit den Icons. Nun gibts nen Fix. Danke fürs Melden. 👆

PR #12280 (2024-02-15): Visualization: hide values if not enough space
fixes #12275
closed 2024-02-15 by andig