evcc-io/evcc #12276: schaltbare Steckdose im "static switch Modus" mit externem Charge Meter; keine Leistungsanzeige

VolkerK62, 2024-02-15:
Der Use case ist nicht unbedingt evcc Kerngeschäft, aber die Auswertung der Ladevorgänge hat mich auf die Idee gebracht.
Von Großverbrauchern soll die Nutzungsdauer, die Energiemenge und der Verbauch gemessen werden.
Dazu ist der Loadpoint immer eingeschaltet (schnell) und die Erfassung der Nutzung (Ladevorgang) erfolgt mit Hilfe von `standbypower` (positiver Wert).
Das funktioniert super, wenn die Schaltsteckdose auf messen kann.
Solange kein Strom fließt, wird "ladebereit" angezeigt, fließt Strom ist "Ladevorgang aktiv" und der Verbauch, die Dauer und die Energiemenge wird erfasst.

Kann die Schaltsteckdose aber nicht messen und wird deshalb ein externes Meter am Loadpoint konfiguriert, funktioniert das Umschalten auf "Ladevorgang aktiv" nicht mehr. Die Anzeige bleibt bei "ladebereit". Der Verbrauch wird zwar angezeigt, aber die Dauer und somit auch die Energiemenge wird nicht mehr erfasst. 
Setzt man `standbypower` auf einen negativen Wert, erfolgt zwar die Umschaltung auf "Ladevorgang aktiv", aber der Verbrauch wird nicht angezeigt, sondern stattdessen nur der Wert, der bei `standbypower` eingetragen ist.
Wünschenswert wäre, dass in dem Fall auch der tatsächliche Verbrauch erfasst würde.
Ist das evtl ein Bug?
Denn in der Doku steht es eigentlich so drin.
https://docs.evcc.io/docs/devices/chargers#schaltbare-steckdosen
[image]

andig, 2024-02-21:
> Kann die Schaltsteckdose aber nicht messen und wird deshalb ein externes Meter am Loadpoint konfiguriert, funktioniert das Umschalten auf "Ladevorgang aktiv" nicht mehr.

Das liegt daran, dass die Steckdose selbst diese Leistung nicht sieht.

> Wünschenswert wäre, dass in dem Fall auch der tatsächliche Verbrauch erfasst würde.

Im Prinzip würde das bedeuten, dass ein am Ladepunkt konfigurierter Zähler einen eingebauten vollständig ersetzen sollte. Möglich, aber aktuell nicht umgesetzt.

> Von Großverbrauchern soll die Nutzungsdauer, die Energiemenge und der Verbauch gemessen werden.

Das ist tatsächlich nicht unser Anwendungsfeld. Mangels Priorität wontfix, PR welcome falls es jemand versuchen möchte :/

VolkerK62, 2024-05-01:
>Im Prinzip würde das bedeuten, dass ein am Ladepunkt konfigurierter Zähler einen eingebauten vollständig ersetzen sollte. Möglich, aber aktuell nicht umgesetzt.

Hallo @andig  ich hoffe ich habe das richtig verstanden.
Ich komme hierauf nochmal zurück, weil genau das aktuell möglich ist. 
Oder ist es in dem Fall ein Unterschied, ob der Charger eine Wallbox oder eine Schaltsteckdose ist?

Meine Wallbox hat einen eingebauten Zähler. Da aber die Gesamt-Leistungsanzeige immer kleiner war, als die Summe der Phasenleistungen habe ich den Zähler als custom nachgebaut (power= Summe der Phasenleistungen) und dem Loadpoint zugewiesen. Funktioniert tadellos.
Der eingebaute Zähler wird also durch den zugewiesenen ersetzt.

VolkerK62, 2024-06-14:
@thierolm  hast du evtl eine Idee, wie man das lösen könnte, dass bei Switchsocket mit externem Meter und positiver standbypower der Status korrekt von "B" auf "C" geschaltet wird?

andig, 2024-06-15:
> Das funktioniert super, wenn die Schaltsteckdose auf messen kann.

Ich würde sagen das ist works as designed. Die Schaltsteckdose kennt den externen Zähler nicht und kann daher auch ihren Status nicht ändern. Wenn das jetzt kein Massenphänomän ist würde ich das so lassen.

VolkerK62, 2024-06-16:
>Wenn das jetzt kein Massenphänomän ist würde ich das so lassen.

Die Nutzergruppe wächst stetig.
Bei Wärmepumpen wird der SG-Kontakt gesteuert, bei Heizstäben ein Schütz ... immer wenn dann auch der tatsächliche Verbrauch gemessen werden soll, tritt dieses Problem auf.
 
Die beiden Beispiele hatte ich noch in Erinnerung ... es gab aber noch ein paar mehr.
https://github.com/evcc-io/evcc/issues/10220#issuecomment-1937867096
https://github.com/evcc-io/evcc/discussions/14329

Nimm es doch ins Backlog auf und ich werde weitere Fälle "sammeln".

premultiply, 2024-06-16:
Ich hätte jetzt auch erwartet dass man _jedem_ Loadpoint bereits ein externes Meter zuweisen kann welches ein ggf. vorhandenes internes Meter ersetzt.
Bei "normalen" Ladepunkten klappt das ja auch bestens. Warum sollte das bei Schaltsteckdosen nicht funktionieren?

andig, 2024-06-16:
Das meter sieht nur der Loadpoint. Der Status kommt aber vom Charger.

VolkerK62, 2024-07-03:
keine Ahnung, ob es in den letzen Monaten eine (unabsichtliche?) Änderung gab, aber jetzt scheint es zu funktionieren.
https://github.com/evcc-io/evcc/discussions/14318#discussioncomment-9950091
Bei negativer standbypower wird die gemessene Leistung angezeigt.

Ich mach hier zu.

closed 2024-07-03 by VolkerK62