evcc-io/evcc #12294: Tesla Proxy for TWC3 does not wake vehicle from sleep

yipikaye, 2024-02-16:
### Describe the bug

Vehicle control now works with the new proxy. However when the car goes to sleep the proxy commands do not wake the vehicle. 

### Steps to reproduce

1. let the car go to sleep and switched EVCC to fast or minpv and that forces the error for me

Trace logs attached as per discusison here: https://github.com/evcc-io/evcc/issues/11846 
...
[evcc.log](https://github.com/evcc-io/evcc/files/14310223/evcc.log)

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: info
levels:
  cache: error

# unique installation id
plant: 8asdfdsaf
sponsortoken: asdfasdf
telemetry: true

interval: 15s # control cycle interval

hems:
  type: sma
  allowcontrol: false # set true to allow SHM controlling charger in PV modes

meters:
- name: grid
  type: template
  template: shelly-1pm
  usage: grid
  host: 192.168.86.115
  user: sdfa
  password: asdf
  channel: 0 # optional
  #- name: grid
  #  type: custom
  #  power:
  #  source: const
  # value: 0

chargers:
- type: template
  template: twc3 
  host: 192.168.86.106
  name: TeslaWallCharger

loadpoints:
- title: Carport
  charger: TeslaWallCharger
  mode: pv

vehicles:

site:
  title: Lyrebirds Nest
  meters:
    grid: grid
    pv:
    battery:
  residualPower: 300

tariffs:
  currency: AUD # three letter ISO-4217 currency code (default EUR)
  grid:
    type: fixed
    price: 0.66
  feedin:
    type: fixed
    price: 0.00

# push messages
messaging:
  events:
    start: # charge start event
      title: Charge started
      msg: Started charging in ${mode} mode at ${vehicleSoc:%.0f}% SOC with ${pvPower:%.1fk}kWh 
    stop: # charge stop event
      title: Charge finished
      msg: Finished charging at ${vehicleSoc:%.0f}% SOC with ${chargedEnergy:%.1fk}kWh in ${chargeDuration}
    connect: # vehicle connect event
      title: Car connected
      msg: ${vehicleName} connected at ${pvPower:%.1fk}kWh and ${vehicleSoc:%.0f}% SOC
    disconnect: # vehicle connected event
      title: Car disconnected
      msg: ${vehicleName} disconnected after ${connectedDuration} at ${vehicleSoc:%.0f}% SOC
    soc: # vehicle soc update event
      title: Soc updated
      msg: Battery charged to ${vehicleSoc:%.0f}%
    guest: # vehicle could not be identified
      title: Unknown vehicle
      msg: Unknown vehicle, guest connected?
  services:
   - type: pushover
     app: asdfasdfa
     recipients: 
      - asdfasdf

influx:
  url: http://localhost:8086
  database: evcc
  # user:
  # password:
```

### Log details

```text
attached
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.124.2 (777cb7b7)

andig, 2024-02-16:
Missing tesla trace :(

yipikaye, 2024-02-16:
how do i do that, that was the log, i got rid of my tokens from the log but the rest is the whole trace log

haroldboom, 2024-02-16:
Trace attached

Feb 16 11:28:04 testrig systemd[1]: evcc.service: Consumed 38.996s CPU time.
Feb 16 11:28:04 testrig systemd[1]: Started evcc.
Feb 16 11:28:04 testrig evcc[2137]: [main  ] INFO 2024/02/16 11:28:04 evcc 0.124.2 (777cb7b7)
Feb 16 11:28:04 testrig evcc[2137]: [main  ] INFO 2024/02/16 11:28:04 using config file: /etc/evcc.yaml
Feb 16 11:28:04 testrig evcc[2137]: [main  ] INFO 2024/02/16 11:28:04 starting ui and api at :7070
Feb 16 11:28:06 testrig evcc[2137]: [db    ] INFO 2024/02/16 11:28:06 using sqlite database: /var/lib/evcc/evcc.db
Feb 16 11:28:06 testrig evcc[2137]: [db    ] TRACE 2024/02/16 11:28:06 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
Feb 16 11:28:06 testrig evcc[2137]: [db    ] TRACE 2024/02/16 11:28:06 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
Feb 16 11:28:06 testrig evcc[2137]: [db    ] TRACE 2024/02/16 11:28:06 SELECT * FROM `settings` LIMIT 1 -1 <nil>
Feb 16 11:28:06 testrig evcc[2137]: [db    ] TRACE 2024/02/16 11:28:06 SELECT * FROM `settings` 4 <nil>
Feb 16 11:28:06 testrig evcc[2137]: [db    ] TRACE 2024/02/16 11:28:06 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="devices" -1 <nil>
Feb 16 11:28:06 testrig evcc[2137]: [db    ] TRACE 2024/02/16 11:28:06 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="device_details" -1 <nil>
Feb 16 11:28:06 testrig evcc[2137]: [db    ] TRACE 2024/02/16 11:28:06 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="configs" -1 <nil>
Feb 16 11:28:06 testrig evcc[2137]: [db    ] TRACE 2024/02/16 11:28:06 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "configs" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
Feb 16 11:28:06 testrig evcc[2137]: [db    ] TRACE 2024/02/16 11:28:06 SELECT * FROM `configs` LIMIT 1 -1 <nil>
Feb 16 11:28:06 testrig evcc[2137]: [db    ] TRACE 2024/02/16 11:28:06 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="config_details" -1 <nil>
Feb 16 11:28:06 testrig evcc[2137]: [db    ] TRACE 2024/02/16 11:28:06 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "config_details" AND sql IS NOT NULL order by type = "table" desc 2 <nil>
Feb 16 11:28:06 testrig evcc[2137]: [db    ] TRACE 2024/02/16 11:28:06 SELECT * FROM `config_details` LIMIT 1 -1 <nil>
Feb 16 11:28:06 testrig evcc[2137]: [db    ] TRACE 2024/02/16 11:28:06 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT ""fk_configs_details"" %" OR sql LIKE "%CONSTRAINT fk_configs_details %" OR sql LIKE "%CONSTRAINT `fk_configs_details`%" OR sql LIKE "%CONSTRAINT [fk_configs_details]%" OR sql LIKE "%CONSTRAINT         fk_configs_details        %") -1 <nil>
Feb 16 11:28:06 testrig evcc[2137]: [db    ] TRACE 2024/02/16 11:28:06 SELECT count(*) FROM sqlite_master WHERE type = "index" AND tbl_name = "config_details" AND name = "idx_unique" -1 <nil>
Feb 16 11:28:06 testrig evcc[2137]: [db    ] TRACE 2024/02/16 11:28:06 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT ""fk_devices_details"" %" OR sql LIKE "%CONSTRAINT fk_devices_details %" OR sql LIKE "%CONSTRAINT `fk_devices_details`%" OR sql LIKE "%CONSTRAINT [fk_devices_details]%" OR sql LIKE "%CONSTRAINT         fk_devices_details        %") -1 <nil>
Feb 16 11:28:06 testrig evcc[2137]: [db    ] TRACE 2024/02/16 11:28:06 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%""device_id"" %" OR sql LIKE "%device_id %" OR sql LIKE "%`device_id`%" OR sql LIKE "%[device_id]%" OR sql LIKE "%        device_id        %") -1 <nil>
Feb 16 11:28:06 testrig evcc[2137]: [db    ] TRACE 2024/02/16 11:28:06 SELECT * FROM `configs` WHERE `configs`.`class` = 2 ORDER BY id 0 <nil>
Feb 16 11:28:06 testrig evcc[2137]: [db    ] TRACE 2024/02/16 11:28:06 SELECT * FROM `configs` WHERE `configs`.`class` = 1 ORDER BY id 0 <nil>
Feb 16 11:28:06 testrig evcc[2137]: [db    ] TRACE 2024/02/16 11:28:06 SELECT * FROM `configs` WHERE `configs`.`class` = 3 ORDER BY id 0 <nil>
Feb 16 11:28:06 testrig evcc[2137]: [tesla-proxy] TRACE 2024/02/16 11:28:06 POST https://auth.tesla.com/oauth2/v3/token
Feb 16 11:28:07 testrig evcc[2137]: [tesla-proxy] TRACE 2024/02/16 11:28:07 client_id=***&grant_type=refresh_token&refresh_token=***
Feb 16 11:28:07 testrig evcc[2137]: --
Feb 16 11:28:07 testrig evcc[2137]: {"access_token":"removed","expires_in":28800,"token_type":"Bearer"}
Feb 16 11:28:07 testrig evcc[2137]: [tesla-proxy] TRACE 2024/02/16 11:28:07 GET https://tesla.evcc.io/api/1/vehicles
Feb 16 11:28:08 testrig evcc[2137]: [tesla-proxy] TRACE 2024/02/16 11:28:08 {"response":[{"id":3744513744581445,"vehicle_id":2252134536587332,"vin":"LRWYHCFS3NC428511","color":null,"access_type":"OWNER","display_name":"Blobby","option_codes":null,"granular_access":{"hide_private":false},"tokens":["3ddfb388ab1e37f1","27e4478d1069b778"],"state":"asleep","in_service":false,"id_s":"3744513744581445","calendar_enabled":true,"api_version":71,"backseat_token":null,"backseat_token_updated_at":null,"ble_autopair_enrolled":false}],"count":1}
Feb 16 11:28:08 testrig evcc[2137]: [twc3  ] TRACE 2024/02/16 11:28:08 GET http://192.168.1.73/api/1/vitals
[... cut ...]
Feb 16 11:29:54 testrig evcc[2137]: [tesla-proxy] TRACE 2024/02/16 11:29:54 POST https://tesla.evcc.io/api/1/vehicles/LRWYHCFS3NC428511/command/set_charging_amps
Feb 16 11:29:56 testrig evcc[2137]: [tesla-proxy] TRACE 2024/02/16 11:29:56 {"charging_amps": 32}
Feb 16 11:29:56 testrig evcc[2137]: --
Feb 16 11:29:56 testrig evcc[2137]: {"response":null,"error":"vehicle unavailable: vehicle is offline or asleep","error_description":""}
Feb 16 11:29:56 testrig evcc[2137]: [lp-1  ] ERROR 2024/02/16 11:29:56 max charge current 32A: 500 Internal Server Error

yipikaye, 2024-02-16:
@andig is there an email address i Can message the tesla logs (include tokens)

andig, 2024-02-16:
@haroldboom could you do an 

    evcc vehicle

when it's asleep? Interesting that the Tesla proxy returns an HTTP 500 when asleep- pretty sure the Fleet API doesn't do that. Could you also try to configure the vehicle once more as `tesla` (not `tesla-proxy`) and do the same test again?

andig, 2024-02-16:
@yipikaye there's always info@evcc.io

andig, 2024-02-16:
Furthermore it seems that retrieving data is different from sending commands:

```
[tesla-proxy] TRACE 2024/02/16 11:28:24 GET https://tesla.evcc.io/api/1/vehicles/LRWYHCFS3NC428511/vehicle_data
[tesla-proxy] TRACE 2024/02/16 11:28:24 {"response":null,"error":"vehicle unavailable: vehicle is offline or asleep","error_description":""}
[lp-1  ] ERROR 2024/02/16 11:28:24 vehicle odometer: asleep

[tesla-proxy] TRACE 2024/02/16 11:29:54 POST https://tesla.evcc.io/api/1/vehicles/LRWYHCFS3NC428511/command/set_charging_amps
[tesla-proxy] TRACE 2024/02/16 11:29:56 {"charging_amps": 32}
--
{"response":null,"error":"vehicle unavailable: vehicle is offline or asleep","error_description":""}
[lp-1  ] ERROR 2024/02/16 11:29:56 max charge current 32A: 500 Internal Server Error
```

I.e. we need to handle the HTTP 500 for the latter case.

haroldboom, 2024-02-16:
> @haroldboom could you do an
> 
> ```
> evcc vehicle
> ```
> 
> when it's asleep? Interesting that the Tesla proxy returns an HTTP 500 when asleep- pretty sure the Fleet API doesn't do that. Could you also try to configure the vehicle once more as `tesla` (not `tesla-proxy`) and do the same test again?

See attached outputs, I also did tesla-command
[vehicle_command.txt](https://github.com/evcc-io/evcc/files/14315755/vehicle_command.txt)
[vehicle_tesla.txt](https://github.com/evcc-io/evcc/files/14315756/vehicle_tesla.txt)
[vehicle_proxy.txt](https://github.com/evcc-io/evcc/files/14315757/vehicle_proxy.txt)

andig, 2024-02-16:
> Is that normal?

No. Needs full trace log in new issue.

andig, 2024-02-16:
Can we try two more things please? Send stop charge commands when vehicle is asleep for command and proxy versions, again with trace log. See cli options for how to.

haroldboom, 2024-02-16:
> Can we try two more things please? Send stop charge commands when vehicle is asleep for command and proxy versions, again with trace log. See cli options for how to.

Hi Andy, sorry I cant find the cli option for stop charging. Could you point me in the right direction?

yipikaye, 2024-02-17:
Thanks @haroldboom for doing the hard yards on the log collection. My micro raspberry pi doesn't like it much lol

andig, 2024-02-17:
@haroldboom if you run

    evcc vehicle --help

you can see the options, so it becomes

    evcc vehicle --stop

again, please for command and proxy.

haroldboom, 2024-02-17:
Hey Andy, sorry not sure why my outputs were blank. Here they are attached again:
[stop_tesla.txt](https://github.com/evcc-io/evcc/files/14318328/stop_tesla.txt)
[stop_command.txt](https://github.com/evcc-io/evcc/files/14318329/stop_command.txt)
[stop_proxy.txt](https://github.com/evcc-io/evcc/files/14318330/stop_proxy.txt)

Also my token had expired again. It seems to expire when I have the PC shutdown for a few hours. I will create an issue when I get a chance

andig, 2024-02-17:
> Also my token had expired again.

Again: this needs a full trace log. And you should stop using the old `tesla`. It won't work, maybe that's what's killing the token. The other issue is clear, will provide a fix.

haroldboom, 2024-02-17:
> > Also my token had expired again.
> 
> Again: this needs a full trace log. And you should stop using the old `tesla`. It won't work, maybe that's what's killing the token. The other issue is clear, will provide a fix.

You are probably right it only started happening when I have been playing around with my config. Once this issue is gone I will see if the token expires still, I am thinking it won’t

andig, 2024-02-17:
Can be tested in tomorrows nightly.

haroldboom, 2024-02-18:
> Can be tested in tomorrows nightly.

Hey Andy has the nightly been pushed?

infinus, 2024-02-18:
@haroldboom I had an update it looks like it was run manually.

andig, 2024-02-18:
Last nightly was broken. Should work now, hopefully. Make sure to read https://github.com/evcc-io/evcc/pull/12277 and set `control: true`.

TheJDJ13, 2024-02-18:
I'm still getting the 500 error. Did I add the 'control: true' properly?

vehicles:
- type: template
   template: tesla-proxy
   title: My Car
   accessToken:  atoken
   refreshToken:  rtoken
   vin: vin
   capacity: 82
   name: TM3
   control: true

andig, 2024-02-18:
Do you have a trace log for tesla?

haroldboom, 2024-02-18:
Same error on my end, trace attached. @TheJDJ13 template should be just 'tesla', 'tesla-proxy' is defunct (from the docs).

[trace.txt](https://github.com/evcc-io/evcc/files/14324366/trace.txt)

andig, 2024-02-18:
@infinus control requires the nightly version. If you cant use that please be patient until the next release.

TheJDJ13, 2024-02-18:
> Same error on my end, trace attached. @TheJDJ13 template should be just 'tesla', 'tesla-proxy' is defunct (from the docs).
> 
> [trace.txt](https://github.com/evcc-io/evcc/files/14324366/trace.txt)

Thanks haroldboom. I just corrected it. The issue remains though.

infinus, 2024-02-18:
@andig I'm on latest nightly, I included the version in the log, the GUI will fail to start. 

`evcc 0.124.2 (777cb7b7)`

TheJDJ13, 2024-02-18:
> @andig I'm believe I'm on latest nightly I updated right before trying, I included the version in the log, the GUI will fail to start.
> 
> `evcc 0.124.2 (777cb7b7)`

I'm on evcc 0.124.2 (4d0c46ef)

0.124.2.+1708273124

infinus, 2024-02-18:
@TheJDJ13 @andig  My apologies, I updated prior to testing and my docker client reported it was up to date but clearly it was not!

yipikaye, 2024-02-18:
testing today but its so overcast i will have to simulate tests

It seems my token has expired after the upgrade, is this normal?

yipikaye, 2024-02-18:
hmm not working at all now, car asleep or awake. 

```
Feb 19 09:41:37 raspberrypi evcc[16457]: [lp-1  ] ERROR 2024/02/19 09:41:37 max charge current 5A: vehicle not capable of current control
Feb 19 09:41:52 raspberrypi evcc[16457]: [lp-1  ] ERROR 2024/02/19 09:41:52 max charge current 5A: vehicle not capable of current control
Feb 19 09:42:07 raspberrypi evcc[16457]: [lp-1  ] ERROR 2024/02/19 09:42:07 max charge current 5A: vehicle not capable of current control
Feb 19 09:42:22 raspberrypi evcc[16457]: [lp-1  ] ERROR 2024/02/19 09:42:22 max charge current 5A: vehicle not capable of current control
```

haroldboom, 2024-02-19:
> hmm not working at all now, car asleep or awake.
> 
> The app can get the SOC off the car but cannot control the car.
> 
> ```
> Feb 19 09:41:37 raspberrypi evcc[16457]: [lp-1  ] ERROR 2024/02/19 09:41:37 max charge current 5A: vehicle not capable of current control
> Feb 19 09:41:52 raspberrypi evcc[16457]: [lp-1  ] ERROR 2024/02/19 09:41:52 max charge current 5A: vehicle not capable of current control
> Feb 19 09:42:07 raspberrypi evcc[16457]: [lp-1  ] ERROR 2024/02/19 09:42:07 max charge current 5A: vehicle not capable of current control
> Feb 19 09:42:22 raspberrypi evcc[16457]: [lp-1  ] ERROR 2024/02/19 09:42:22 max charge current 5A: vehicle not capable of current control
> ```

Mine is working fine controlling the car once its awake but just cant wake. Is control set in your config?

yipikaye, 2024-02-19:
Maybe not, I see a new parameter in the GUI. I will delete and recreate, do I set it to true?

pezet1910, 2024-02-19:
„I didn't get any value to paste into that last field and the data wasn't clear so i set it to "true"
Ja, „true“ ist richtig.

andig, 2024-02-19:
@TheJDJ13 thanks for the log. I can see the error reason is in a different place that I've checked sofar. Will fix this today.

andig, 2024-02-19:
Closed in https://github.com/evcc-io/evcc/commit/8a385f4e5d59034263d12e6415ef47b626bdf64f, new nightly in 30min.

closed 2024-02-19 by andig

haroldboom, 2024-02-19:
> Closed in [8a385f4](https://github.com/evcc-io/evcc/commit/8a385f4e5d59034263d12e6415ef47b626bdf64f), new nightly in 30min.
[trace.txt](https://github.com/evcc-io/evcc/files/14329121/trace.txt)

hey Andy, nightly seems to have the same issue

TheJDJ13, 2024-02-19:
> @TheJDJ13 thanks for the log:
> 
> ```
> {"response":null,"error":"vehicle unavailable: vehicle is offline or asleep","error_description":""}
> ```
> 
> I can see the error reason is in a different place that I've checked sofar. Will fix this today.

I had not sent a log so thanks to whoever really supplied that.

TheJDJ13, 2024-02-19:
> > Closed in [8a385f4](https://github.com/evcc-io/evcc/commit/8a385f4e5d59034263d12e6415ef47b626bdf64f), new nightly in 30min.
> 
> hey Andy, nightly seems to have the same issue [trace.txt](https://github.com/evcc-io/evcc/files/14329127/trace.txt)

Unfortunately I see the same. Still getting the error.

andig, 2024-02-19:
Are you running https://github.com/evcc-io/evcc/releases/tag/0.124.4?

TheJDJ13, 2024-02-19:
Yes I am on 0.124.4.

andig, 2024-02-19:
Please share trace log.

haroldboom, 2024-02-19:
Hi Andy, I had to update my evcc with the 124.4 deb manually as I had no updates available (via apt update) after the 124.2 nightly however the issue persists. Please see my trace log below.
[trace.txt](https://github.com/evcc-io/evcc/files/14336713/trace.txt)

yipikaye, 2024-02-20:
I can confirm I still get 500 internal server error when waking the vehicle from sleep

andig, 2024-02-20:
Sorry for the trouble. I‘ll code a test to see where this still comes from.

yipikaye, 2024-02-20:
No worries mate, we appreciate all you do. There can't be much of the old API code left now 😅

Stacykid, 2024-02-20:
I have been running it for about 2 hours now with the current 124.4 and have not used the Tesla app to put the car to sleep. After 2 hours I opened the Tesla app and the car didn't seem to sleep either.

So far I haven't received any error messages and when I switch back and forth between Fast and Min+PV this also works. 
As there is only a little sun today, I have set it down to 2A min, but it regulates back and forth between 1.5 and 5/6KW.

lits101, 2024-02-20:
I'm still getting Error 500 when the vehicle is asleep under 0.124.4. Waking the vehicle with the Tesla app prior to commencing charging in evcc resolves the issue.

andig, 2024-02-20:
...another nightly in 30min

TheJDJ13, 2024-02-20:
I don't need to charge right now and the forecast here is for rain all day, but I was able to see my car wake up when changing from solar to min-solar.

0.124.4. (97eebcc7)

haroldboom, 2024-02-20:
Latest nightly is able to wake the vehicle for charging. Will test today to be sure but looks good @andig

yipikaye, 2024-02-20:
Its so foggy here today I cannot see my front gate so I cannot really test the solar part but I let the car go to sleep then set it to min+ solar. There were a few failures (while the car woke up) then on a retry charging started! :) 

So very pleased with this outcome :) Thank you @andig

Stacykid, 2024-02-21:
Here again a statement before the last nightly, so only with the official 124.4.

I set EVCC to PV only and plugged it into the car in the morning when it was still dark. After the Tesla had fallen asleep, I set it to Min+PV and it couldn't wake up the car (error 500...) until I opened the Tesla app and woke it up.

Conclusion: with 124.4, the Tesla must not currently fall asleep, otherwise it will not regulate. Once it is awake, it will stay that way due to the constant regulation.

This only concludes the current official version, as I said, without the last Nightly!

silencer2008, 2024-10-07:
Hi all, has this been fixed ? I seem to have the same issue where evcc unable to wake vehicle up. 
Running Nightly build on HA add-on. 
trace attached
[evcc-20241007-093141-trace.log](https://github.com/user-attachments/files/17271825/evcc-20241007-093141-trace.log)

andig, 2024-10-07:
@silencer2008 without looking at the log: what specific wakeup are you missing, i.e. which situation (using TWC3? BLE proxy?)?

silencer2008, 2024-10-07:
Hi @andig , wakeup using TWC3, this is the error in the trace.
{"response":null,"error":"vehicle unavailable: vehicle is offline or asleep","error_description":""}

Car only starts charging after using app to wakeup

andig, 2024-10-07:
Question is not answered. In which situation? Restart of evcc? Charge plan? What's your config?

andig, 2024-10-07:
Moved to https://github.com/evcc-io/evcc/discussions/16545