evcc-io/evcc #12302: Batteriesteuerung Huawei LUNA

mucki12, 2024-02-16:
Hallo,

ich würde mich sehr über die Umsetzung der Batteriesteuerung für Huawei LUNA freuen.

Wenn ich das richtig überblicke, dann basierten die meisten bisherigen Vorschläge auf der Regelung von Basiseinstellungen wie End-of-discharge SoC und Maximum charge/discharge power.

Für die Umsetzung stellt Huawei allerdings separate Möglichkeiten zur Verfügung, welche die o. a. Basiseinstellungen nicht verändern und zudem auch nur temporär wirken.

Zunächst noch einmal die gesamte Modbus Dokumentation:
https://www.photovoltaikforum.com/core/attachment/315028-solar-inverter-modbus-interface-definitions-pdf/

Umsetzung **batterydischargecontrol** (ich hoffe ihr wisst was ich meine):

An:
1. Modbus Adresse 47100 (Nr. 131) auf "2: Discharge"
2. Modbus Adresse 47083 (Nr. 125) auf "1440" (=Minuten / Max-Wert 1440 als Vorschlag - nach Ablauf der  Zeit wird 1. wieder automatisch rückgängig gemacht = auf Standard "0: Stop" gesetzt!)
3. Modbus Adresse 47249 (Nr. 142) auf "0" (LUNA wird nicht entladen)

Aus:
1. Modbus Adresse 47100 (Nr. 131) auf "0: Stop" 

Dieses wurde bereits im Vorfeld von mir getestet und hat exakt die beschriebene Wirkung. 

Noch einmal zur Sicherheit:
Nein - der oben beschriebene Vorschlag hat keinen Einfluss auf die Basiseinstellungen des Nutzers. 

Nächste mögliche Ausbaustufe: Umsetzung **forcecharge** (für günstige Stromkosten Preise etc.):

An:
1. Modbus Adresse 47087 (Nr. 128) auf "1: Enable" (AC-Laden grundsätzlich einschalten)
2. Modbus Adresse 47100 (Nr. 131) auf "1: Charge"
3. Modbus Adresse 47083 (Nr. 125) auf "1440" (=Minuten / als Vorschlag - nach Ablauf der  Zeit wird 1. wieder automatisch rückgängig gemacht = auf Standard "0: Stop" gesetzt!)
4. Modbus Adresse 47249 (Nr. 142) auf "2500" (als fester Wert / unterstützt jeder LUNA)
Alternativ zu 4.:  Modbus Adresse 37007 (Nr. 6) auslesen und bei 47249 (Nr. 142) eintragen

Aus:
1. Modbus Adresse 47100 (Nr. 131) auf "0: Stop" 

Ja - das ist noch einmal eine Schippe drauf. Aber auch hier sind die Änderungen ebenfalls nur temporär. Alternativ kann auch statt einem Zeitablauf ein SoC-Zielwert genutzt werden bzw. in Kombination. Allerdings sehe ich vor dem Hintergrund der Nutzung (Laden nur zu günstigen Zeiten) den Nutzen noch nicht.

Würde mich freuen, wenn das umgesetzt werden würde und natürlich auch, wenn andere Huawei Nutzer unterstützen würden.

Beteiligte Modbus Register als Screenshot:
[image]

[image]

[image]

Optional:
[image]

markusr, 2024-02-19:
Hier Links zur Huawei-Python-Implementierung:

https://github.com/wlcrs/huawei_solar
bzw. die Register-Definitionen sind in
https://github.com/wlcrs/huawei-solar-lib

Dort kann man schön die Werte / Logik nachschauen, z.B:
```forcible_charge``` --> https://github.com/wlcrs/huawei_solar/blob/main/services.py#L264

andig, 2024-02-21:
@mucki12 wir brauchen normal, hold und charge. hold finde ich oben nicht? Und ganz wichtig: hast Du probiert ob das funktioniert?

mucki12, 2024-02-21:
@andig 

**normal:** (hierauf wird ansonsten nach Zeitablauf von hold/charge eh wieder umgestellt)
Modbus Adresse 47100 (Nr. 131) auf "0: Stop"

**hold:**
Modbus Adresse 47100 (Nr. 131) auf "2: Discharge"
Modbus Adresse 47083 (Nr. 125) auf "1440" (=Minuten / Max-Wert 1440 als Vorschlag - nach Ablauf der Zeit wird 1. wieder automatisch rückgängig gemacht = auf Standard "0: Stop" gesetzt!)
Modbus Adresse 47249 (Nr. 142) auf "0" (LUNA wird nicht entladen)

**charge:**
Modbus Adresse 47087 (Nr. 128) auf "1: Enable" (AC-Laden grundsätzlich einschalten)
Modbus Adresse 47100 (Nr. 131) auf "1: Charge"
Modbus Adresse 47083 (Nr. 125) auf "1440" (=Minuten / als Vorschlag - nach Ablauf der Zeit wird 1. wieder automatisch rückgängig gemacht = auf Standard "0: Stop" gesetzt!)
Modbus Adresse 47247 (Nr. 141) auf "2500" (als fester Wert / unterstützt jeder LUNA)
Alternativ zu 4.: Modbus Adresse 37007 (Nr. 6) auslesen und bei 47247 (Nr. 141) eintragen

Funktioniert alles wie gewünscht und durch das Setzen des Minuten-Parameters auch noch gefahrlos.

Getestet mit den korrespondierenden Einstellungen 
a) direkt im Webfrontend und 
b) über Home Assistant Funktion. 
Fazit: funktioniert wir gewünscht. Hatte ich mit @premultiply  hier schon mal vorbesprochen: https://github.com/evcc-io/evcc/discussions/10806

Vielleicht zur Veranschaulichung noch einmal die o. a. identischen Parameter aus dem Webfrontend:
[image]

andig, 2024-02-23:
@mucki12 das sieht überschaubar aus. Welches Template nutzt du? Dann würde ich es da mal zu Testzwecken zur Verfügung stellen.

mucki12, 2024-02-23:
@andig Vielen Dank für die Rückmeldung.
Nutze das template: huawei-dongle-powersensor 

Kurze Verständnisfrage: Müsste ich dann eine evcc Version kompilieren *zitter*?
:-)

closed 2024-02-25 by andig

mucki12, 2024-02-25:
Zur Sicherheit auch hier noch mal die Rückfrage:
Verstehe ich das richtig, dass in der geplanten Umsetzung die benötigten Modbus Register für hold und charge alle 30 Sekunden neu geschrieben werden? Vielleicht habe ich das auch falsch verstanden...

premultiply, 2024-02-26:
Ja, bitte testen.

mucki12, 2024-02-26:
Schade. Hatte noch gehofft, dass ich das falsch verstanden hätte :-)

Also einen Watchdog finde ich grundsätzlich klasse und und eine super Idee. Daher vielen Dank für die Mühe hier weiter zu denken. Allerdings halte ich einen 30 Sekundentakt für den schreibenden Zugriff für deutlich zu ambitioniert. Wir Huawei Nutzer sind ja schon froh, wenn der sDongle bei den normalen Abfragen alle 30 Sekunden nicht aus den Fugen gerät :-)

Da ich die Anfrage hier ja gestellt habe, kann ich das natürlich gerne mal kurz (!) in einer für Abfragen nicht ganz so relevanten Zeit testen. Dauerhaft nutzen käme für mich mit einem 30 Sekundentakt allerdings nicht in Frage (würde ich auch keinem zu raten) und daher wäre es wahrscheinlich besser, wenn das noch jemand zusätzlich testet, der das dann auch zukünftig nutzen wird.

Reicht es für den Kurztest die letzte nightly zu installieren und mit der gleichen Konfiguration zu arbeiten?

premultiply, 2024-02-26:
Bitte einfach mal testen.

mucki12, 2024-02-26:
Ums Testen will ich mich ja nicht drücken :-) Habe mal die aktuelle nightly gezogen: evcc 0.124.4 (2df084af) und die evcc.yaml unverändert gelassen (also Nutzung template: huawei-dongle-powersensor).

Fehler beim Start:
[evcc-nightly 2.csv](https://github.com/evcc-io/evcc/files/14401396/evcc-nightly.2.csv)

andig, 2024-02-26:
Logfiles als CSV sind... MIST. Soweit ich sehe fehlt der interessante Teil der Meldung.

mucki12, 2024-02-26:
Da gebe ich dir recht. Bin allerdings gerade nicht zu Hause und hatte daher nur begrenzte Möglichkeiten :-)
Reiche ich dann heute Abend oder morgen nach. 

Wie kann ich das Log für euch interessanter gestalten:
```
log: debug
levels:
  cache: error
```

auf 
log: trace 
umstellen?

mucki12, 2024-02-26:
Vielleicht habe ich auch den Huawei Dongle unterschätzt. Ich habe die geplante Watchdog Logik mal in HA nachgebaut und das hat der sDongle ganz gut verkraftet. Hatte eigentlich erwartet, dass er mit Schreibbefehlen alle 30 Sekunden höher ausgelastet wäre.
Allerdings fehlten bei diesem Test noch die zusätzlichen Abfragen von evcc. Das schaue ich mir dann morgen mit der nächsten nightly an.

premultiply, 2024-02-28:
`evcc meter --help` 😉

`evcc meter my_battery_name -b normal|hold|charge`

mucki12, 2024-02-28:
@premultiply 
Sehr gerne - hoffe das Format ist so passend:

evcc meter battery -b normal:
```
[main  ] INFO 2024/02/28 12:04:49 evcc 0.124.4 (45ac7c81)
[main  ] INFO 2024/02/28 12:04:49 using config file: /etc/evcc.yaml
[db    ] INFO 2024/02/28 12:04:49 using sqlite database: /root/.evcc/evcc.db
[db    ] TRACE 2024/02/28 12:04:49 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2024/02/28 12:04:49 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2024/02/28 12:04:49 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/02/28 12:04:49 SELECT * FROM `settings` 8 <nil>
[mqtt  ] INFO 2024/02/28 12:04:49 connecting evcc-693918641 at tcp://192.168.2.189:1883
[mqtt  ] DEBUG 2024/02/28 12:04:49 tcp://192.168.2.189:1883 connected
[db    ] TRACE 2024/02/28 12:04:49 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="devices" -1 <nil>
[db    ] TRACE 2024/02/28 12:04:49 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="device_details" -1 <nil>
[db    ] TRACE 2024/02/28 12:04:49 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="configs" -1 <nil>
[db    ] TRACE 2024/02/28 12:04:49 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "configs" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2024/02/28 12:04:49 SELECT * FROM `configs` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/02/28 12:04:49 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="config_details" -1 <nil>
[db    ] TRACE 2024/02/28 12:04:49 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "config_details" AND sql IS NOT NULL order by type = "table" desc 2 <nil>
[db    ] TRACE 2024/02/28 12:04:49 SELECT * FROM `config_details` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/02/28 12:04:49 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT ""fk_configs_details"" %" OR sql LIKE "%CONSTRAINT fk_configs_details %" OR sql LIKE "%CONSTRAINT `fk_configs_details`%" OR sql LIKE "%CONSTRAINT [fk_configs_details]%" OR sql LIKE "%CONSTRAINT     fk_configs_details  %") -1 <nil>
[db    ] TRACE 2024/02/28 12:04:49 SELECT count(*) FROM sqlite_master WHERE type = "index" AND tbl_name = "config_details" AND name = "idx_unique" -1 <nil>
[db    ] TRACE 2024/02/28 12:04:49 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT ""fk_devices_details"" %" OR sql LIKE "%CONSTRAINT fk_devices_details %" OR sql LIKE "%CONSTRAINT `fk_devices_details`%" OR sql LIKE "%CONSTRAINT [fk_devices_details]%" OR sql LIKE "%CONSTRAINT     fk_devices_details  %") -1 <nil>
[db    ] TRACE 2024/02/28 12:04:49 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%""device_id"" %" OR sql LIKE "%device_id %" OR sql LIKE "%`device_id`%" OR sql LIKE "%[device_id]%" OR sql LIKE "%       device_id       %") -1 <nil>
[db    ] TRACE 2024/02/28 12:04:49 SELECT * FROM `configs` WHERE `configs`.`class` = 2 ORDER BY id 0 <nil>
[modbus] TRACE 2024/02/28 12:04:50 modbus: send 00 01 00 00 00 06 01 06 b7 fc 00 00
[modbus] TRACE 2024/02/28 12:04:51 modbus: recv 00 01 00 00 00 06 01 06 b7 fc 00 00
```

evcc meter battery -b hold:

```
[main  ] INFO 2024/02/28 12:07:36 evcc 0.124.4 (45ac7c81)
[main  ] INFO 2024/02/28 12:07:36 using config file: /etc/evcc.yaml
[db    ] INFO 2024/02/28 12:07:37 using sqlite database: /root/.evcc/evcc.db
[db    ] TRACE 2024/02/28 12:07:37 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2024/02/28 12:07:37 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2024/02/28 12:07:37 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/02/28 12:07:37 SELECT * FROM `settings` 8 <nil>
[mqtt  ] INFO 2024/02/28 12:07:37 connecting evcc-1078861679 at tcp://192.168.2.189:1883
[mqtt  ] DEBUG 2024/02/28 12:07:37 tcp://192.168.2.189:1883 connected
[db    ] TRACE 2024/02/28 12:07:37 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="devices" -1 <nil>
[db    ] TRACE 2024/02/28 12:07:37 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="device_details" -1 <nil>
[db    ] TRACE 2024/02/28 12:07:37 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="configs" -1 <nil>
[db    ] TRACE 2024/02/28 12:07:37 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "configs" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2024/02/28 12:07:37 SELECT * FROM `configs` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/02/28 12:07:37 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="config_details" -1 <nil>
[db    ] TRACE 2024/02/28 12:07:37 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "config_details" AND sql IS NOT NULL order by type = "table" desc 2 <nil>
[db    ] TRACE 2024/02/28 12:07:37 SELECT * FROM `config_details` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/02/28 12:07:37 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT ""fk_configs_details"" %" OR sql LIKE "%CONSTRAINT fk_configs_details %" OR sql LIKE "%CONSTRAINT `fk_configs_deta [... cut ...]
[modbus] TRACE 2024/02/28 12:08:38 modbus: send 00 05 00 00 00 0b 01 10 b7 ef 00 02 04 00 00 00 01
[modbus] TRACE 2024/02/28 12:08:39 modbus: recv 00 05 00 00 00 03 01 90 04
[main  ] FATAL 2024/02/28 12:08:39 set battery mode: modbus: exception '4' (server device failure), function '16'
```

Nach meinem Laienverständnis scheinbar noch ein Fehler bei charge. Weiß aber auch nicht wie ich das in der Praxis testen kann und ob er trotzdem laden würde. Nochmal kurz die Logik mit HA nachgespielt und da lädt er.

mucki12, 2024-03-02:
Sorry falls ich nerven sollte @premultiply , aber brauchst du noch weitere Infos/Logs?

mucki12, 2024-03-06:
@rucksman 
Das HA Add-on Modbus Proxy lief hier eine zeitlang, aber im Moment benutze ich evcc selber als Modbus Proxy. Läuft bisher ziemlich rund.

snookerap, 2024-03-11:
Sorry für die dumme Frage...
Aber, was geht hiermit nun? Kann ich damit aktiv meine Batterie bei günstigen Strom beladen?

VG Christian

mucki12, 2024-03-11:
@snookerap 
Hausspeicher mit günstigem Netzstrom laden geht im evcc grundsätzlich noch nicht, aber mit dem Huawei Template ist man schon mal vorbereitet.
Was jetzt schon geht ist die Entladesperre bei dem Modus Schnell, Planladen und smarten Netzladen.

snookerap, 2024-03-12:
@mucki12  ích muss hier nochmal fragen bzgl. Modbus Proxy.
Wie sieht den dein evcc.yaml unter "meters" aus?
Wie gehst du auf den WR drauf, über den Dongle oder über Wifi-AP?
Ich würde auch gern evcc als Proxy laufen lassen, zur Zeit habe ich den Modbus-Proxy als Addon laufen, bin aber nicht so zufrieden damit (immer wieder TimeOuts)

mucki12, 2024-03-13:
Lieber nicht in diesem Issue.

snookerap, 2024-03-13:
okay, habe ein einen Beitrag in Discussions erstellt. Konnte dich aber nicht erwähnen.

Sammy990, 2024-03-14:
Eine Frage, welches Template muss ich verwenden damit die Batteriesteuerung im UI sichtbar wird?

Verwende aktuell huawei-sun2000-rs485 und hier taucht die Battery Control in der UI nicht auf...

Danke

P.S. Bei Bedarf erstelle ich gerne eine neue Frage mit Config Auszug

andig, 2024-03-14:
Siehe https://github.com/evcc-io/evcc/pull/12449