evcc-io/evcc #12305: Niro EV only 7kW?

apographon, 2024-02-16:
### Describe the bug

I need to figure out where my issue root cause is.
My Niro EV dies only charge on 7.2kW since it was in the Garage today.
Charging Speed in the Vehicle is set to 100%
Config of the easee is set to 16A.
EVCC put on „schnell“
Max.Ladestrom is set to 16A (default)

Did I miss anything to check?

### Steps to reproduce

1. Start Charging
2.
3.
...

### Configuration details

```yaml
phases: 3
mincurrent: 6
maxcurrent: 16
```

### Log details

```text
None
```

### What type of operating system are you running?

HomeAssistant Add-on

Hofyyy, 2024-02-17:
Did you already charge the car on your wallbox with 11kw without evcc? I ask because you compare against different loadpoints. 

I would test first without evcc. souns like 2 phases are connected …

apographon, 2024-02-17:
Fair questions.
Yes, the easee was serving 16A/3p in the past.
Charging yesterday was 10A/3p.
Public charging station was 16A/3p. So I can exclude the Vehicle.

closed 2024-02-17 by andig