evcc-io/evcc #12312: Docker Mount Point

razaqq, 2024-02-17:
By default the `evcc.yaml` is mounted into `/etc/evcc.yaml`, this makes it hard to mount into the container with compose as file mounts arent well supported compared to directory mounts.

For example doing this:
```
    volumes:
      - ./config/evcc.yaml:/etc/evcc.yaml
```
will yield the following
```
Error response from daemon: failed to create task for container: failed to create shim task: OCI runtime create failed: runc create failed: unable to start container process: error during container init: error mounting "/opt/evcc/config/evcc.yaml" to rootfs at "/etc/evcc.yaml": mount /opt/evcc/config/evcc.yaml:/etc/evcc.yaml (via /proc/self/fd/6), flags: 0x5000: not a directory: unknown: Are you trying to mount a directory onto a file (or vice-versa)? Check if the specified host path exists and is the expected type
```

It will always create a directory named `evcc.yaml` for example.

The solution to create the config already and then mounting the file directly as a bind mount:
```
volumes:
      - type: bind
        source: ./config/evcc.yaml
        target: /etc/evcc.yaml
```

This is inconvenient, makes initial configuration hard and can be done much simpler.

I would suggest instead mounting it into `/etc/evcc/evcc.yaml` by default.
Additionally i would suggest adding an environment variable `EVCC_CONFIG_LOCATION` to give the user control over the mount point.

This way we can instead mount `./config:/etc/evcc` as a directory and let evcc create a default config.

andig, 2024-02-17:
Not sure I copy. I can happily do this:

```
  evcc:
    container_name: evcc
    image: evcc/evcc:nightly
    command: evcc --metrics --profile
    ports:
      - 7070:7070
    volumes:
      - /volume1/data/docker/evcc/evcc.yaml:/etc/evcc.yaml
```

razaqq, 2024-02-17:
Are you using a docker volume or mounting from file system?

Specifying the full path to the local file for me still creates a directory `evcc.yaml`