evcc-io/evcc #12317: ERROR battery mode: modbus: exception '4' (server device failure), function '4'

Gandolf90, 2024-02-17:
### Describe the bug

Seit dem letzten update evcc v0.124.2 bekomme ich immer folgende meldung :

"battery mode: modbus: exception '4' (server device failure), function '4'"

habe mal log aktiviert im docker container

folgende config ( siehe config anhang ) habe ich für meine QCells Wechselrichter
Wallbox ist Tesla3er / Tesla model 3 als auto

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
meters:
  - type: template
    template: solax
    id: 1
    host: {ip}
    port: 502
    usage: grid
    modbus: tcpip
    name: grid
  - type: template
    template: solax
    id: 1
    host: {ip}
    port: 502
    usage: pv
    modbus: tcpip
    name: pv
  - type: template
    template: solax
    id: 1
    host: {ip}
    port: 502
    usage: battery
    modbus: tcpip
    name: battery
```

### Log details

```text
/app # evcc meter --log error,modbus:trace --battery-mode hold --battery-mode-wait 30s
[main  ] INFO 2024/02/17 12:57:02 evcc 0.124.2
[main  ] INFO 2024/02/17 12:57:02 using config file: /etc/evcc.yaml
[modbus] TRACE 2024/02/17 12:58:02 modbus: send 00 01 00 00 00 06 01 06 00 20 00 00
[modbus] TRACE 2024/02/17 12:58:02 modbus: recv 00 01 00 00 00 07 01 04 04 1e e9 00 05
[main  ] FATAL 2024/02/17 12:58:02 set battery mode: modbus: exception '4' (server device failure), function '4'
```

### What type of operating system are you running?

Docker container

### Version

v0.124.2

andig, 2024-02-17:
/cc @ioniks

andig, 2024-02-21:
/cc @docsnyder

Looking a bit like https://github.com/evcc-io/evcc/issues/11227, only neither FC 6 nor FC 4 are supported. No idea :(

bombero4482, 2024-02-22:
Ich habe leider das gleiche Problem. Hat jemand eine Lösung?
Liebe Grüße

andig, 2024-02-22:
Scheint als bräuchten wir mal einen Solax Anwender bei dem es funktioniert um die Unterschiede zu finden.

andig, 2024-02-23:
@Gandolf90 kannst du bitte gleichen Test nochmal mit alter funktionierender Version machen?

bombero4482, 2024-02-24:
Moin. Also bei mir funktioniert jetzt wieder alles einwandfrei mit der Version 0.124.4. Habe den Raspberry neu installiert und die PV Anlage einmal neu gestartet. Seitdem klappt‘s wieder. Home Assistant greift bei mir auch noch auf den Modbus zu. Den hatte ich während der Neuinstallation ausgeschaltet. 
Vielleicht hilft das ja ein bisschen weiter. 
Liebe Grüße

andig, 2024-02-24:
Scheint also am Gerät zu liegen? Dann machen wir hier mal zu. Mangels weiterer Diagnose wäre ohnehin nix zu tun.

closed 2024-02-24 by andig

Gandolf90, 2024-02-26:
so bin lange nicht mehr dazu gekommen 
ich starte mal den Wechselrichter neu und hoffe das dadurch das Problem behoben wird
bin bei version 0.124.4.

doctordsingh, 2024-07-28:
Hat jemand eine Lösung mittlerweile?

Manfredgm, 2024-12-31:
Ich hatte die Fehlermeldung auch mit Solax, beim Schnell-Laden mit Evcc wurde die Hausbatterie nicht gesperrt, dabei kam dann die Fehlermeldung. 
Ich habe dann im Evcc im Menu Hausbatterie den Wert von Prorisiere Hausbatterie bis sie 0% erreicht hat geändert auf eine Wert von 90%. Seitdem wird die Hausbatterie beim Schnell Laden gesperrt und die Fehlermeldung ist ausgeblieben. (Anmerkung: Hausbatterie war auf 90% geladen, als das Auto laden gestartet wurde). Meine Erfahrung, vielleicht hilft es bei euch ebenfalls.

niniob, 2025-03-29:
In der Vergangenheit hat die Einstellung „_Verhindere Entladung im Schnell-Modus und bei geplantem Laden._“ ohne Probleme funktioniert, aber jetzt hat es nicht mehr geklappt, die Batterie wurde entladen und es kam die obige Fehlermeldung in evcc. Ein Update von evcc auf die aktuelle Version 202.1 hat nichts geändert, Fehler war immer noch da. Schlussendlich konnte ich es aber sehr einfach über die Solax App fixen. Nachdem ich dort die aktuellen Einstellungen des Wechselrichter ausgelesen hatte (nichts geändert) war das Problem weg. D.h. einfach Solax App öffnen, dann auf Geräte > X3-Hybrid-G4 > Ferneinstellungen > Einstellungen (Default Passwort 0000) und dann mal alle Einstellungen durchklicken. Wie gesagt, Anzeige in der Solax App war ausreichend, geändert habe ich nichts. Vielleicht hilft die Info jemand anderem mit dem gleichen Problem.