evcc-io/evcc #12326: Tesla kommuniziert nicht mit TWC3.

pezet1910, 2024-02-18:
### Describe the bug

Die evcc.yaml Datei lt.[ ](https://docs.evcc.io/docs/devices/vehicles#tesla) geändert. Leider keine Kommunikation zwischen Tesla und TWC3.

### Steps to reproduce

1. evcc starten
2. von PV auf Min+PV oder Schnell Umschelten
3. Fehlermeldung
...

### Configuration details

```yaml
vehicles:
- name: my_car
  type: template
  template: tesla
  title: TESLA
  accessToken: ey1234
  refreshToken: EU_1234
  capacity: 82
```

### Log details

```text
[site  ] DEBUG 2024/02/18 10:51:19 ----                                          
[lp-1  ] DEBUG 2024/02/18 10:51:19 charge power: 0W                              
[site  ] DEBUG 2024/02/18 10:51:19 pv power: 704W                                
[site  ] DEBUG 2024/02/18 10:51:19 battery soc: 25%                              
[site  ] DEBUG 2024/02/18 10:51:19 battery power: -10W                           
[site  ] DEBUG 2024/02/18 10:51:19 grid power: 765W                              
[site  ] DEBUG 2024/02/18 10:51:19 battery has priority at soc 25% (< 90%)       
[site  ] DEBUG 2024/02/18 10:51:19 site power: 1265W                             
[lp-1  ] DEBUG 2024/02/18 10:51:19 charge voltages: [0 4.4 2.8]V                 
[lp-1  ] DEBUG 2024/02/18 10:51:19 charge currents: [0 0.3 0.3]A                 
[lp-1  ] DEBUG 2024/02/18 10:51:19 charger status: B                             
[main  ] ERROR 2024/02/18 10:51:19 vehicle status: asleep                        
[lp-1  ] DEBUG 2024/02/18 10:51:19 pv charge current: 0A = 0A + -1.83A (1265W @ 3
p)                                                                               
[lp-1  ] ERROR 2024/02/18 10:51:19 max charge current 6A: vehicle not capable of 
current control
```

### What type of operating system are you running?

Docker container

### Version

0.124.2

mucki12, 2024-02-18:
Dachte die Änderungen für die Ladesteuerung mit dem neuen Proxy sind bisher nur in der Nightly enthalten?

KevinGruber, 2024-02-18:
Hätte ich auch gesagt, jedoch scheint es im nightly auch Probleme zu geben -> habe gerade in die Diskussion gepostet

https://github.com/evcc-io/evcc/discussions/12143#discussioncomment-8506422

pezet1910, 2024-02-18:
> Dachte die Änderungen für die Ladesteuerung mit dem neuen Proxy sind bisher nur in der Nightly enthalten?

Das hat bis gestern auch funktioniert (nightly). Seit ca. 20 Stunden kann ich weder mit "nightly" noch mit "latest" Version die Applikation zum Laufen bringen.

andig, 2024-02-18:
Das geht mit tesla template aktuell nur im nightly. Im Release ist es tesla-proxy. Leider kann die Doku das noch nicht differenzieren, da müssen wir dran arbeiten.

pezet1910, 2024-02-18:
Ist die nightly Version seit gestern verändert worden? Meine Konfiguration, die gestern sehr gut funktioniert hat, macht heute Probleme.

andig, 2024-02-18:
Sorry. The nightly build is corrupt regarding Tesla. Will fix asap.

andig, 2024-02-18:
Nightly should be fixed in 30 minutes. See https://github.com/evcc-io/evcc/pull/12277 for config changes

closed 2024-02-18 by andig

laenglea, 2024-02-23:
Bei mir dasselbe. Sollte das schon behoben sein? Aktuellste nightly mit includiertem fix mag (noch) nicht steuern

mucki12, 2024-02-23:
Selbst die Final steuert schon. Meine das aktuell nur noch ein Problem mit dem Standby des Teslas in Verbindung mit einem Ladeplan gefixt wird.

mucki12, 2024-02-23:
Kenne deine Konfiguration ja nicht, aber tippe auf "control: true" :-)

laenglea, 2024-02-23:
Genau das war es.
control: true

Vielen Dank

andig, 2024-02-23:
Ist ab heute im Nightly auch nciht mehr notwendig- wir haben das vereinfacht.