evcc-io/evcc #12333: Deye 3p hybrid inverter add active battery control

deadrabbit87, 2024-02-18:
**Describe the solution you'd like**
Deye is using Timeslots (called Time of use TOU) which can be used to control the battery soc in specified time slots. 

See also page 33 in the [manual](https://www.deyeinverter.com/deyeinverter/2022/04/12/manual-sun-(5-12)k-sg04lp3-au-deye-1.pdf). 

It is possible to change these values via modbus. If the timeslot 1 is set like 00:00 - 23:59, the battery won't discharge any more if the Register `166` is set to the current state of the battery. See also the attached modbus register description. 

**Additional context**
Checks with `mbpoll` where sucessfull, setting soc for timeslot 1 to 100%
```
sudo mbpoll -1 -a 1 -b 9600 -r 166 -0 -P none -t 4:int  /dev/ttyACM0 100
mbpoll 1.0-0 - FieldTalk(tm) Modbus(R) Master Simulator
Copyright © 2015-2019 Pascal JEAN, https://github.com/epsilonrt/mbpoll
This program comes with ABSOLUTELY NO WARRANTY.
This is free software, and you are welcome to redistribute it
under certain conditions; type 'mbpoll -w' for details.

Protocol configuration: Modbus RTU
Slave configuration...: address = [1]
                        start reference = 166, count = 1
Communication.........: /dev/ttyACM0,       9600-8N1
                        t/o 1.00 s, poll rate 1000 ms
Data type.............: 32-bit integer (little endian), output (holding) register table

Written 1 references.

```

[Modbus_Deye.pdf](https://github.com/evcc-io/evcc/files/14323686/Modbus_Deye.pdf)

deadrabbit87, 2024-02-23:
https://github.com/evcc-io/evcc/pull/12415

closed 2024-02-24 by andig

Blowfly69, 2025-01-03:
Hi Filip, 
you don't need to change settings at the inverter. 
- Is a recent evcc release installed?
- During the installation routine ("evcc configure"), did you use select the "deye hybrid 3p" template? 
- Modbus connection to the inverter is working, i.e. evcc shows home battery SOC and grid power etc.?
If so, you should be able to select the "avoid using home battery during fast charging" button (or similar, label translated from German, don't have the english version running). 
If you activate the "experimental" features in the configuration menu and have a flexible energy tariff, you additionally can use smart charging of your home battery (see second tab in the home battery menu). 

However, please be adviced that in the current version of the template, your TOU table is partially overwritten by new values to enable/disable discharging. This is still a bug from my point of view which needs to be resolved.

andig, 2025-01-03:
Please move to a discussion- this PR is implemented and closed. Thank you!