evcc-io/evcc #12336: Tronity API: Unexpected status: 401 (Unauthorized)

xbaun, 2024-02-18:
### Describe the bug

When I reconnect the car after some time, evcc can no longer access the tronity api and logs the following error message:

```
[lp-1  ] ERROR 2024/02/17 11:42:33 vehicle soc: unexpected status: 401 (Unauthorized)
````

I suspect that the access token has expired and the token is not getting refetched. The expiration duration of a tronity access token is one hour.

I think the error is, that the expiration time is not correctly applied to the oauth token when the api response is getting parsed. The `oauth2.Token` struct tag for the Expiry field is "expiry" and not matching the response field "expires_in" of the [tronity api](https://app.tronity.tech/docs#operation/AuthController_authentication).

I tried to fix this, but I am not sure if this is the right location to do it: https://github.com/xbaun/evcc/commit/8413183640759c8de3e7c2d37392686ec8db5997

Maybe someone can reproduce this error and verify my solution?

### Steps to reproduce

When using Tronity, reconnecting the car after more than one hour should trigger the error.

### Configuration details

```yaml
vehicles:
  - type: template
    template: tronity
    title: TRONITY Extension
    clientid: XXXX
    clientsecret: XXXX
    capacity: 50
    name: ev5
```

### Log details

```text
[lp-1  ] INFO 2024/02/17 10:06:33 car connected
[lp-1  ] ERROR 2024/02/17 10:06:33 vehicle soc: unexpected status: 401 (Unauthorized)
[lp-1  ] ERROR 2024/02/17 10:06:39 vehicle odometer: unexpected status: 401 (Unauthorized)
[lp-1  ] INFO 2024/02/17 10:06:53 start charging ->
[lp-1  ] ERROR 2024/02/17 10:06:54 vehicle soc: unexpected status: 401 (Unauthorized)
```

### What type of operating system are you running?

Linux

### Version

0.124.1

andig, 2024-02-18:
> think the error is, that the expiration time is not correctly applied to the oauth token when the api response is getting parsed. The oauth2.Token struct tag for the Expiry field is "expiry" and not [...]

That‘s standard go and working fine. Closing due to missing trace log.

closed 2024-02-18 by andig

xbaun, 2024-02-19:
Hi @andig , thanks for your quick response.

You are right it is pretty standard oauth stuff. As far as I can oversee the token retrieval in evcc is a [custom implementation](https://github.com/xbaun/evcc/blob/master/util/oauth/tokensource.go#L35) when [evcc token](https://github.com/xbaun/evcc/blob/master/vehicle/tronity.go#L98) is not used. 

This custom token source implementation doesn’t respect the retrieved `expires_in` field. When using `evcc token` command to retrieve a new access token, then the [oauth2 implementation](https://github.com/xbaun/evcc/blob/master/cmd/token_tronity.go#L33) is used that respects the expires_in field of the response and sets the `Expiry` field of `oauth2.Token`.

While investigating this bug, I also noticed that the scopes of the tronity oauth config needs an update: https://github.com/xbaun/evcc/blob/master/vehicle/tronity/auth.go#L17

Further, the current `evcc token` command cannot handle a tronity vehicle template:
> [main  ] FATAL 2024/02/19 09:20:32 vehicle type 'template' does not support token authentication

I would be happy to help here, but would need some help from you as to whether my thoughts are correct.

I will also provide a Trace Log when I can reproduce this behaviour the next time.

andig, 2024-02-19:
> This custom token source implementation doesn’t respect the retrieved expires_in field.

You are making assumptions.

> I will also provide a Trace Log when I can reproduce this behaviour the next time.

Full config and trace log please.

xbaun, 2024-03-03:
@andig Have you had a chance to look at the trace? I fixed it by extracting the expiry time from the tronity jwt and pass it to oauth2 token impl. Since then it runs without out the 401 failure. 

Here my solution for the error I described above: https://github.com/xbaun/evcc/commit/36882aa2242a9d4593639260d12f0b8f4c3a2787#diff-fe96e494ec6bd226373c84fca0dbd17402636e4965738a601a2319bb31600f73R167

Can you reopen the issue?

andig, 2024-03-03:
@xbaun deine Lösung führt einfach dazu, dass kein token refresh gemacht wird. Wenn der dann später kommt tritt das gleiche Problem wieder auf. Anders gesagt: das ist keine Lösung :O

andig, 2024-03-03:
Duplikat: https://github.com/evcc-io/evcc/issues/12599