evcc-io/evcc #12344: Ladestand während Ladevorgang falsch

langepil, 2024-02-19:
### Describe the bug

Ich benutze einen go-e Charger und als Fahrzeug einen Opel Mokka-e über die Tronity Schnittstelle.

Leider berechnet evcc während des Ladevorgangs den SOC des Autos falsch.

Als Capacity bei der Tronity Schnittstelle habe ich 48kWh eingestellt. Gestartet habe ich bei 51%, geladen habe ich 7,5kWh. D.h. der Ladestand hätte rechnerisch um ca. 16%, also auf 67% steigen müssen. Im evcc ist er aber "nur" auf 59% gestiegen.

Woran kann das liegen?

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
chargers:
- type: template
  template: go-e-v3 
  host: 10.10.10.63  
  name: wallbox5

vehicles:
- type: template
  template: tronity 
  title: Opel Mokka-e - Tronity  
  icon: car  
  clientid: XXXXX
  clientsecret: XXXXX  
  vin: VXKUK??????????  
  capacity: 48  
  phases: 3  
  mode: now  
  minCurrent: 6  
  maxCurrent: 16  
  name: ev4

loadpoints:
- title: Carport
  charger: wallbox5
  vehicle: ev4
  mode: now
```

### Log details

```text
-
```

### What type of operating system are you running?

Docker container

### Version

0.124.2

closed 2024-02-19 by andig

langepil, 2024-02-19:
@andig 
Kannst du mich bitte kurz aufschlagen, was ich falsch mache, oder woran es liegt?

StefanSchoof, 2024-02-19:
Am besten liest du einmal die Text in dem Bug Formular: https://github.com/evcc-io/evcc/issues/new?assignees=&labels=&projects=&template=bug_report.yaml

> NOTE Due to amount of issues raised we will close incomplete issues (unclear description what the issue actually is, missing log file, unsupported platform) without further comments.