evcc-io/evcc #12356: residualPower setting configurable per month

Till--H, 2024-02-19:
**Is your feature request related to a problem? Please describe.**
To maximize PV use with the  `residualPower` parameter, I need to reconfigure the parameter and restart EVCC every month or so during winter/spring to adjust it according to the expected sun energy and my minimum phases.

**Describe the solution you'd like**
I‘d love to supply a list of month in the config that describes for every month how much residualPower should be assumed, e.g.

```yaml
residualPower:
  - start: 01.01
     end: 31.03
     value: -1000
  - start: 01.10
     end: 31.12
     value: -800
```

**Describe alternatives you've considered**
- Syntax could be simpler to just accept a month name and its value
- Could add another value to just handle the month override (makes backwards compatibility eaiser)
- Could add the value to the UI instead to make changing it easier
- Could add a solar forecast to dynamically set the value for a given month (fully automates it, but very error prone as well)

**Additional context**
If this is interesting and has chances to be accepted, I could try to build this myself

andig, 2024-02-19:
I agree this is beyond scope of what we want to do.

closed 2024-02-19 by andig

Till--H, 2024-02-21:
Thanks for pointing me to the API. To whom it may be interesting: I added an automation in Homeassistant to set this API according to a fixed schedule, there might be smarter ways (e.g., by using solar forecast of the day):

In my evcc configuration.yaml, I added
```yaml
rest_command:
  set_residual_power:
    url: "http://EVCC-URL:7070/api/residualpower/{{ res_power }}"
    method: "post"

automations:
 - alias: EVCC Residual Power
   description: "Sets the residual power in EVCC"
   trigger:
     - platform: time
       at: "01:00:00"
   action:
     - service: rest_command.set_residual_power
       data_template:
         res_power: >
           {% set month = now().month %}
           {% if month in [1, 12] %}
             -3000
           {% elif month in [2, 11] %}
             -2500
           {% elif month in [3, 10] %}
             -2000
           {% else %}
             0
           {% endif %}
```

This sets the residual power every day to automatically take care of a restarting add-on or Homeassistant.