evcc-io/evcc #12367: Keine Verbindung mehr zum VW e-UP! nach Update von 0.124.1 auf 0.124.4

Balou1986, 2024-02-20:
### Describe the bug

Nach dem Update von 0.124.1 auf 0.124.4 wird mein VW e-up! nur noch als (Offline) im EVCC angezeigt.

Reboot, systemctl restart evcc, ect. brachte auch keine Besserung.
In der VW App wird der Status des Fahrzeuges ganz normal angezeigt.

### Steps to reproduce

1. sudo apt-get update
2. sudo apt-get upgrade
3. sudo apt-get autoremove
4. evcc -l debug vehicle

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: xxx

interval: 10s # control cycle interval

sponsortoken: xxx

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: true

meters:
- type: template
  template: senec-home 
  usage: grid  
  host: 192.168.178.116  
  schema: http
  name: EnFluRi
- type: template
  template: senec-home 
  usage: pv  
  host: 192.168.178.116  
  schema: http
  name: Fronius_Symo_8_2
- type: template
  template: senec-home 
  usage: battery  
  host: 192.168.178.116  
  schema: http
  name: Senec_Home_V1_Li_10

chargers:
- type: template
  template: go-e 
  host: 192.168.178.237  
  name: go-e_Home_fix
- name: TP-Link_H110
  type: template
  template: tplink
  host: 192.168.178.226 # IP-Adresse oder Hostname
  standbypower: 15 # Leistung oberhalb des angegebenen Wertes wird als Ladeleistung gewertet (Optional)
  integrateddevice: true 
  icon: bike

loadpoints:
- title: go e charger
  charger: go-e_Home_fix
  vehicle: VW e-UP
  mode: off
  phases: 0
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: true
  priority: 1
- title: Giant E-Bikes
  charger: TP-Link_H110
  mode: off
  phases: 1
  mincurrent: 1
  maxcurrent: 2
  resetOnDisconnect: true
  priority: 2

site:
  title: Teck46
  meters:
    grid: EnFluRi
    pv:
    - Fronius_Symo_8_2
    battery:
    - Senec_Home_V1_Li_10
  prioritySoc: 80 # Hausbatterie bekommt bis zum Soc 80% Priorität beim laden
  bufferSoc: 90 # Hausbatterie wird oberhalb Soc 90% als Puffer genutzt

tariffs:
  currency: EUR # (default EUR)
  grid:
    # static grid price
    type: fixed
    price: 0.xxx # [currency]/kWh

  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.xx # [currency]/kWh

#  co2:
#    type: grünstromindex
#    zip: 33428

#vehicles:
#  - name: VW e-UP
#    type: template
#    template: offline
#    title: VW e-UP! # Wird in der Benutzeroberfläche angezeigt (Optional)
#    capacity: 32 # Akkukapazität in kWh (Optional)
#    phases: 2
    
vehicles:
  - name: VW e-UP
    type: template
    template: vw
    title: VW e-UP! # Wird in der Benutzeroberfläche angezeigt (Optional)
    user: xxx # Benutzerkonto (bspw. E-Mail Adresse, User Id, etc.)
    password: xxx # Passwort des Benutzerkontos (bei führenden Nullen bitte in einfache Hochkommata setzen)
    vin: WVWZZxxx # Erforderlich, wenn mehrere Fahrzeuge des Herstellers vorhanden sind (Optional)
    capacity: 32 # Akkukapazität in kWh (Optional)
    phases: 2
```

### Log details

```text
main  ] INFO 2024/02/20 11:57:09 evcc 0.124.4
[main  ] INFO 2024/02/20 11:57:09 using config file: /etc/evcc.yaml
[main  ] INFO 2024/02/20 11:57:09 starting ui and api at :7070
[db    ] INFO 2024/02/20 11:57:10 using sqlite database: /home/ralf/.evcc/evcc.db
[main  ] ERROR 2024/02/20 11:57:12 creating vehicle VW_e-UP failed: cannot create vehicle type 'template': cannot create vehicle type 'vw': missing state
[lp-1  ] WARN 2024/02/20 11:57:12 deprecated: minCurrent setting is ignored, please remove
[lp-1  ] WARN 2024/02/20 11:57:12 deprecated: maxcurrent setting is ignored, please remove
[lp-2  ] WARN 2024/02/20 11:57:12 deprecated: minCurrent setting is ignored, please remove
[lp-2  ] WARN 2024/02/20 11:57:12 deprecated: maxcurrent setting is ignored, please remove
[lp-2  ] WARN 2024/02/20 11:57:12 deprecated: phases setting is ignored, please remove
[lp-1  ] DEBUG 2024/02/20 11:57:13 charge total import: 860.205kWh
[lp-2  ] DEBUG 2024/02/20 11:57:14 charge total import: 0.001kWh
[site  ] WARN 2024/02/20 11:57:14 battery configured but residualPower is missing or <= 0 (add residualPower: 100 to site), see https://docs.evcc.io/en/docs/reference/configuration/site#residualpower
[site  ] INFO 2024/02/20 11:57:14 site config:
[site  ] INFO 2024/02/20 11:57:14   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2024/02/20 11:57:14     grid:      power ✓ energy ✗ currents ✗
[site  ] INFO 2024/02/20 11:57:14     pv 1:      power ✓ energy ✗ currents ✗
[site  ] INFO 2024/02/20 11:57:14     battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✗
[site  ] INFO 2024/02/20 11:57:14   vehicles:
[site  ] INFO 2024/02/20 11:57:14     vehicle 1: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[lp-1  ] INFO 2024/02/20 11:57:14 loadpoint 1:
[lp-1  ] INFO 2024/02/20 11:57:14   mode:        off
[lp-1  ] INFO 2024/02/20 11:57:14   charger:     power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
[lp-1  ] INFO 2024/02/20 11:57:14   meters:      charge ✓
[lp-1  ] INFO 2024/02/20 11:57:14     charge:    power ✓ energy ✓ currents ✓
[lp-2  ] INFO 2024/02/20 11:57:14 loadpoint 2:
[lp-2  ] INFO 2024/02/20 11:57:14   mode:        off
[lp-2  ] INFO 2024/02/20 11:57:14   charger:     power ✓ energy ✓ currents ✗ phases ✗ wakeup ✗
[lp-2  ] INFO 2024/02/20 11:57:14   meters:      charge ✓
[lp-2  ] INFO 2024/02/20 11:57:14     charge:    power ✓ energy ✓ currents ✗
[lp-1  ] DEBUG 2024/02/20 11:57:14 phase timer inactive
[lp-1  ] DEBUG 2024/02/20 11:57:14 pv timer inactive
[lp-1  ] INFO 2024/02/20 11:57:14 vehicle updated: unknown -> VW_e-UP! (offline)
[lp-2  ] DEBUG 2024/02/20 11:57:14 phase timer inactive
[lp-2  ] DEBUG 2024/02/20 11:57:14 pv timer inactive
[site  ] WARN 2024/02/20 11:57:14 interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval
[server] FATAL 2024/02/20 11:57:14 unlinkat /tmp/evcc: operation not permitted
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.124.4

andig, 2024-02-20:
> cannot create vehicle type 'vw': missing state

Da klappt das Login nicht. Schau mal ob neue Terms&Conditions zu akzeptieren sind.

closed 2024-02-20 by andig

Balou1986, 2024-02-20:
Hi,
danke für die schnelle Antwort.
Ich habe mich gerade bei myVolkswagen angemeldet.
Alles ganz normal, keine Meldung, nichts was ich akzeptieren musste oder ähnliches.

Aber, jetzt funktioniert alles wieder.
Scheinbar hatte VW nur Sehnsucht nach mir...

Sorry, nächstes mal bin ich Schlauer und probiere mich erst einmal anzumelden.