evcc-io/evcc #12371: Tariffs: add time-dependent charges (peak and off-peak)

andig, 2024-02-20:
Refs https://github.com/evcc-io/evcc/discussions/12331#discussioncomment-8508798

Maergsche, 2024-03-12:
Mit: 0.124.9+1710209577

Fehlermeldung,:
`[ main  ] ERROR 2024/03/12 05:42:10 failed configuring grid tariff:
cannot create tariff type 'energinet': 1 error(s) decoding:

* '' has invalid keys: zones`

Konfiguration:
```
tariffs:
  currency: DKK
  grid:
    type: energinet
    region: dk1
    charges: 0.966
    zones:
      - days: Mo-So
        hours: 0-6
        price: 0.11012
      - days: Mo-So
        hours: 6-17
        price: 0.33035
      - days: Mo-So
        hours: 17-21
        price: 0.99105
      - days: Mo-So
        hours: 21-0
        price: 0.33035
    tax: 0.25
  feedin:
    type: energinet
    region: dk1
    charges: -0.01
    tax: 0.0```

hokkeist, 2024-07-02:
Hi, Any news about this issue, would help. Since in Finland there can also be different prices for night and daytime tariffs. So that would be Entso-E + additional tariffs based on zones.

danielraffel, 2025-01-02:
It would be great to support seasonal distinctions (summer vs. winter, defined by specific months/days) for California utilities that charge homes based on an Electric/Solar energy plan with fixed seasonal rates (no daily changes):

**Winter Season: Oct 1-May 30**
- Peak (highest price) – 4 p.m. to 9 p.m is $0.XX
- Partial-Peak – between 3 p.m and 4pm, and between 9 p.m and 12 a.m is $0.XX
- Off-Peak (lowest price) – between 12 a.m and 3 p.m is $0.XX

**Summer Season: June 1-September 30**
- Peak (highest price) – 4 p.m. to 9 p.m is $0.XX
- Partial-Peak – between 3 p.m and 4pm, and between 9 p.m and 12 a.m is $0.XX
- Off-Peak (lowest price) – between 12 a.m and 3 p.m is $0.XX

andig, 2025-01-02:
This has been enabled with https://github.com/evcc-io/evcc/pull/17414. You can now add a formula that takes the slot‘s timestamp into account and modify charges/price accordingly.

andig, 2025-01-02:
See https://docs.evcc.io/docs/devices/tariffs#b%C3%B6rsenpreise-mit-zeitkomponente

closed 2025-01-02 by andig