evcc-io/evcc #12373: Problem with modbus TCP meters after upgrade do 0.124.4

bilbolodz, 2024-02-20:
### Describe the bug

I'm using some SDM630 meters and connected via HF5111B modbus TCP converter. In version 0.123.7 all was working fine. After upgrade to 0.124.4 I'd to change configuration file (old format not supported any more) and now evcc can't connect to any of my meters with error:
"charge meter: read failed: read tcp 192.168.105.100:49550->192.168.105.4:502: i/o timeout"
Old working config file was:
```
meters:
- type: template
  template: fronius-solarapi-v1 
  usage: pv  
  host: 192.168.105.252  
  name: Fronius
- name: PGE
  type: modbus
  model: SDM # SDM630
  uri: 192.168.105.4:502
  rtu: false # rs485 device connected using ethernet adapter
  id: 1
- name: Dom
  type: modbus
  model: SDM # SDM630
  uri: 192.168.105.4:502
  rtu: false # rs485 device connected using ethernet adapter
  id: 2
  #power: Power # default value, optionally override
  #energy: Sum # default value, optionally override
- name: Klimatyzacja
  type: modbus
  model: SDM230 # SDM630
  uri: 192.168.105.4:502
  rtu: false # rs485 device connected using ethernet adapter
  id: 4
- name: Ladowarka
  type: modbus
  model: SDM # SDM630
  uri: 192.168.105.4:502
  rtu: false # rs485 device connected using ethernet adapter
  id: 3
```
New (not working) config file is:
```
meters:
- type: template
  template: fronius-solarapi-v1 
  usage: pv
  host: 192.168.105.252
  name: Fronius
- name: PGE
  type: template
  template: eastron
  modbus: rs485tcpip
  host: 192.168.105.4 # Hostname
  port: 502 # Port
  #rtu: false # rs485 device connected using ethernet adapter
  id: 1
- name: Dom
  type: template
  template: eastron
  modbus: rs485tcpip
  host: 192.168.105.4 # Hostname
  port: 502 # Port
  #rtu: false # rs485 device connected using ethernet adapter
  id: 2
- name: Klimatyzacja
  type: template
  template: eastron
  modbus: rs485tcpip
  host: 192.168.105.4 # Hostname
  port: 502 # Port
  #rtu: false # rs485 device connected using ethernet adapter
  id: 4
- name: Ladowarka
  type: template
  template: eastron
  modbus: rs485tcpip
  host: 192.168.105.4 # Hostname
  port: 502 # Port
  #rtu: false # rs485 device connected using ethernet adapter
  id: 3
```
How to fix it?

### Steps to reproduce

Install evcc version 0.124.4

### Configuration details

```yaml
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: info
levels:
  cache: error
  site: info
  lp-1: info
  ID.3: debug
  eebus: info

# unique installation id
plant: removed

interval: 60s # control cycle interval

meters:
- type: template
  template: fronius-solarapi-v1 
  usage: pv
  host: 192.168.105.252
  name: Fronius
- name: PGE
  type: template
  template: eastron
  modbus: rs485tcpip
  host: 192.168.105.4 # Hostname
  port: 502 # Port
  #rtu: false # rs485 device connected using ethernet adapter
  id: 1
- name: Dom
  type: template
  template: eastron
  modbus: rs485tcpip
  host: 192.168.105.4 # Hostname
  port: 502 # Port
  #rtu: false # rs485 device connected using ethernet adapter
  id: 2
- name: Klimatyzacja
  type: template
  template: eastron
  modbus: rs485tcpip
  host: 192.168.105.4 # Hostname
  port: 502 # Port
  #rtu: false # rs485 device connected using ethernet adapter
  id: 4
- name: Ladowarka
  type: template
  template: eastron
  modbus: rs485tcpip
  host: 192.168.105.4 # Hostname
  port: 502 # Port
  #rtu: false # rs485 device connected using ethernet adapter
  id: 3
chargers:
- type: template
  template: elliconnect 
  ski: removed
  ip: 192.168.105.241  
  name: wallbox4

vehicles:
- type: template
  template: id 
  title: ID.3  
  user: removed  
  password: removed  
  capacity: 51  
  phases: 2  
  icon: car  
  cache: 15m  
  #minSoc: 20  
  #targetSoc: 80  
  name: VW-ID3
  vin: removed  
loadpoints:
- title: Lisia
  charger: wallbox4
  meter: Ladowarka
  mode: pv
  phases: 0

site:
  title: Lisia
  meters:
    grid: PGE
    pv:
    - Fronius
    aux:
    - Klimatyzacja
    - Dom

eebus:
  certificate:
      private: |
          -----BEGIN EC PRIVATE KEY-----

          -----END EC PRIVATE KEY-----
      public: |
          -----BEGIN CERTIFICATE-----
          -----END CERTIFICATE-----

modbusproxy:
  #  - port: 5200
  #    uri: solar-edge:502
  #    # rtu: true
  #    # readonly: true

tariffs:
  currency: PLN # three letter ISO-4217 currency code (default EUR)
  grid:
    type: fixed
    price: 0.4218 # EUR/kWh
    zones:
      - days: Mo-Fr
        hours: 6-22
        price: 0.5999 # EUR/kWh

messaging:
  events:
    start: # charge start event
      title: Charge started
      msg: Started charging in "${mode}" mode
    stop: # charge stop event
      title: Charge finished
      msg: Finished charging ${chargedEnergy:%.1fk}kWh in ${chargeDuration}.
    connect: # vehicle connect event
      title: Car connected
      msg: "Car connected at ${pvPower:%.1fk}kW PV"
    disconnect: # vehicle connected event
      title: Car disconnected
      msg: Car disconnected after ${connectedDuration}
    soc: # vehicle soc update event
      title: Soc updated
      msg: Battery charged to ${vehicleSoc:%.0f}%
    guest: # vehicle could not be identified
      title: Unknown vehicle
      msg: Unknown vehicle, guest connected?
  services:
    - type: email
      uri: smtp://xxxxx
influx:
  url: http://192.168.105.133:8086
  database: evcc 
  token: removed
  org: removed

mqtt:
  broker: 192.168.105.22:1883
  topic: evcc # root topic for publishing, set empty to disable
```

### Log details

```text
Feb 20 17:19:59 evcc evcc[1261]: [site  ] ERROR 2024/02/20 17:19:59 grid meter: read failed: read tcp 192.168.105.100:36886->192.168.105.4:502: i/o timeout
Feb 20 17:19:59 evcc evcc[1261]: [site  ] ERROR 2024/02/20 17:19:59 grid meter: read failed: read tcp 192.168.105.100:36886->192.168.105.4:502: i/o timeout
Feb 20 17:20:29 evcc evcc[1261]: [lp-1  ] ERROR 2024/02/20 17:20:29 charge meter: read failed: read tcp 192.168.105.100:51420->192.168.105.4:502: i/o timeout
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.124.4

andig, 2024-02-20:
The modbus library was last updated 5 months ago. Can you identify the exact release that has brought this issue? There is nothing else I could suggest for diagnosis. i/o timeout is exactly that...

bilbolodz, 2024-02-20:
OK I will try to check which version is broken

bilbolodz, 2024-02-20:
It looks that problem stars in version 0.123.9. It requires new config format and NOT connecting to my modbus TCP adapter.
Reversing to version 0.123.7 (and old config) heals problem.

andig, 2024-02-20:
.8 works too? Could you share modbus trace logs for both when running

    evcc meter

andig, 2024-02-20:
Tentatively marking as bug

bilbolodz, 2024-02-20:
.8 generate other errors but it read meter. What do you mean "forward your grid meter port"?

andig, 2024-02-20:
Port forwarding.

bilbolodz, 2024-02-20:
```
evcc meter
[main  ] INFO 2024/02/20 22:22:22 evcc 0.123.7
[main  ] INFO 2024/02/20 22:22:22 using config file: /root/evcc.yaml
[db    ] INFO 2024/02/20 22:22:22 using sqlite database: /root/.evcc/evcc.db
[mqtt  ] INFO 2024/02/20 22:22:22 connecting evcc-193857410 at tcp://192.168.105.22:1883
[mqtt  ] DEBUG 2024/02/20 22:22:22 tcp://192.168.105.22:1883 connected
[eebus ] INFO 2024/02/20 22:22:22 Local SKI:  bfe624a1d5c92cd51fa43ea0414ccd6a945d1df8
[eebus ] DEBUG 2024/02/20 22:22:22 mdns: announce
[eebus ] DEBUG 2024/02/20 22:22:22 mdns: using zeroconf
[eebus ] DEBUG 2024/02/20 22:22:22 starting websocket server on :4712
[eebus ] DEBUG 2024/02/20 22:22:22 error during websocket server starting: listen tcp :4712: bind: address already in use
Fronius
-------
Power: 0W

PGE
---
Power: 8713W

Dom
---
Power: 1503W

Klimatyzacja
------------
Power: 22W

Ladowarka
---------
Power: 7129W

[eebus ] DEBUG 2024/02/20 22:22:22 mdns: stop announcement
```

```
evcc meter
[main  ] INFO 2024/02/20 22:24:56 evcc 0.123.8
[main  ] INFO 2024/02/20 22:24:56 using config file: /root/evcc.yaml
[db    ] INFO 2024/02/20 22:24:56 using sqlite database: /root/.evcc/evcc.db
[mqtt  ] INFO 2024/02/20 22:24:56 connecting evcc-1203291675 at tcp://192.168.105.22:1883
[mqtt  ] DEBUG 2024/02/20 22:24:56 tcp://192.168.105.22:1883 connected
[eebus ] INFO 2024/02/20 22:24:56 Local SKI:  bfe624a1d5c92cd51fa43ea0414ccd6a945d1df8
[eebus ] DEBUG 2024/02/20 22:24:56 mdns: announce
[eebus ] DEBUG 2024/02/20 22:24:56 mdns: using zeroconf
[eebus ] DEBUG 2024/02/20 22:24:56 starting websocket server on :4712
[eebus ] DEBUG 2024/02/20 22:24:56 error during websocket server starting: listen tcp :4712: bind: address already in use
Fronius
-------
Power: 0W

PGE
---
Power: 8711W

Dom
---
Power: 1499W

Klimatyzacja
------------
Power: 24W

Ladowarka
---------
Power: 7130W

[eebus ] DEBUG 2024/02/20 22:24:56 mdns: stop announcement

```

bilbolodz, 2024-02-20:
```
Feb 20 22:27:35 evcc evcc[3724]: [main  ] INFO 2024/02/20 22:27:35 evcc 0.123.9
Feb 20 22:27:35 evcc evcc[3724]: [main  ] INFO 2024/02/20 22:27:35 using config file: /etc/evcc.yaml
Feb 20 22:27:35 evcc evcc[3724]: [main  ] INFO 2024/02/20 22:27:35 starting ui and api at :7070
Feb 20 22:27:35 evcc evcc[3724]: [db    ] INFO 2024/02/20 22:27:35 using sqlite database: /var/lib/evcc/evcc.db
Feb 20 22:27:35 evcc evcc[3724]: [mqtt  ] INFO 2024/02/20 22:27:35 connecting evcc-1517020724 at tcp://192.168.105.22:1883
Feb 20 22:27:35 evcc evcc[3724]: [eebus ] INFO 2024/02/20 22:27:35 Local SKI:  bfe624a1d5c92cd51fa43ea0414ccd6a945d1df8
Feb 20 22:27:39 evcc evcc[3724]: [lp-1  ] WARN 2024/02/20 22:27:39 phases not configured, assuming 3p
Feb 20 22:27:49 evcc evcc[3724]: [lp-1  ] ERROR 2024/02/20 22:27:49 charge total import: read failed: read tcp 192.168.105.100:42848->192.168.105.4:502: i/o timeout
Feb 20 22:27:49 evcc evcc[3724]: [site  ] INFO 2024/02/20 22:27:49 site config:
Feb 20 22:27:49 evcc evcc[3724]: [site  ] INFO 2024/02/20 22:27:49   meters:      grid ✓ pv ✓ battery ✗
Feb 20 22:27:49 evcc evcc[3724]: [site  ] INFO 2024/02/20 22:27:49     grid:      power ✓ energy ✓ currents ✓
Feb 20 22:27:49 evcc evcc[3724]: [site  ] INFO 2024/02/20 22:27:49     pv 1:      power ✓ energy ✗ currents ✗
Feb 20 22:27:49 evcc evcc[3724]: [site  ] INFO 2024/02/20 22:27:49   vehicles:
Feb 20 22:27:49 evcc evcc[3724]: [site  ] INFO 2024/02/20 22:27:49     vehicle 1: range ✓ finish ✓ status ✓ climate ✓ wakeup ✓
Feb 20 22:27:49 evcc evcc[3724]: [lp-1  ] INFO 2024/02/20 22:27:49 loadpoint 1:
Feb 20 22:27:49 evcc evcc[3724]: [lp-1  ] INFO 2024/02/20 22:27:49   mode:        pv
Feb 20 22:27:49 evcc evcc[3724]: [lp-1  ] INFO 2024/02/20 22:27:49   charger:     power ✗ energy ✗ currents ✗ phases ✗ wakeup ✗
Feb 20 22:27:49 evcc evcc[3724]: [lp-1  ] INFO 2024/02/20 22:27:49   meters:      charge ✓
Feb 20 22:27:49 evcc evcc[3724]: [lp-1  ] INFO 2024/02/20 22:27:49     charge:    power ✓ energy ✓ currents ✓
Feb 20 22:27:49 evcc evcc[3724]: [main  ] INFO 2024/02/20 22:27:49 new version available: 0.124.4
```

bilbolodz, 2024-02-20:
I can offer you ssh access to machine running evcc. Please share your public ssh key and preferred user name.

bilbolodz, 2024-02-20:
MBMD version 0.13 is working fine with my meters

andig, 2024-02-21:
@bilbolodz my pubkey is `ssh-ed25519 AAAAC3NzaC1lZDI1NTE5AAAAIGi8rUJarKZuWnZNOnhtcXZCrPskbF7HOcig6nkjWlU2`

andig, 2024-02-21:
@bilbolodz we'd need trace logs, not debug:

    evcc meter --log trace,db:error

for last good and first bad version.

bilbolodz, 2024-02-21:
Working:

```
evcc meter --log trace,db:error
[main  ] INFO 2024/02/21 09:05:25 evcc 0.123.8
[main  ] INFO 2024/02/21 09:05:25 using config file: /root/evcc.yaml
[mqtt  ] INFO 2024/02/21 09:05:25 connecting evcc-1087818029 at tcp://192.168.105.22:1883
[mqtt  ] DEBUG 2024/02/21 09:05:25 tcp://192.168.105.22:1883 connected
[eebus ] INFO 2024/02/21 09:05:25 Local SKI:  bfe624a1d5c92cd51fa43ea0414ccd6a945d1df8
[eebus ] DEBUG 2024/02/21 09:05:25 mdns: announce
[eebus ] DEBUG 2024/02/21 09:05:25 mdns: using zeroconf
[eebus ] DEBUG 2024/02/21 09:05:25 starting websocket server on :4712
[eebus ] DEBUG 2024/02/21 09:05:25 error during websocket server starting: listen tcp :4712: bind: address already in use
Fronius
-------
[http  ] TRACE 2024/02/21 09:05:25 GET http://192.168.105.252/solar_api/v1/GetPowerFlowRealtimeData.fcgi
[http  ] TRACE 2024/02/21 09:05:25 {
   "Body" : {
      "Data" : {
         "Inverters" : {
            "1" : {
               "DT" : 105,
               "E_Day" : 490.30001831054688,
               "E_Total" : 27541200,
               "E_Year" : 224626.109375,
               "P" : 427
            }
         },
         "Site" : {
            "E_Day" : 490.30001831054688,
            "E_Total" : 27541200,
            "E_Year" : 224626.109375,
            "Meter_Location" : "load",
            "Mode" : "vague-meter",
            "P_Akku" : null,
            "P_Grid" : 236.71679945535072,
            "P_Load" : -663.71679945535072,
            "P_PV" : 427,
            "rel_Autonomy" : 64.33466809193294,
            "rel_SelfConsumption" : 100
         },
         "Version" : "12"
      }
   },
   "Head" : {
      "RequestArguments" : {},
      "Status" : {
         "Code" : 0,
         "Reason" : "",
         "UserMessage" : ""
      },
      "Timestamp" : "2024-02-21T09:06:49+01:00"
   }
}
Power: 427W

PGE
---
[modbus] TRACE 2024/02/21 09:05:25 modbus: send 00 01 00 00 00 06 01 04 00 34 00 02
[modbus] TRACE 2024/02/21 09:05:26 modbus: recv 00 01 00 00 00 07 01 04 04 43 8a 2b 16
Power: 276W

Dom
---
[modbus] TRACE 2024/02/21 09:05:26 modbus: send 00 02 00 00 00 06 02 04 00 34 00 02
[modbus] TRACE 2024/02/21 09:05:26 modbus: recv 00 02 00 00 00 07 02 04 04 44 24 7e b5
Power: 658W

Klimatyzacja
------------
[modbus] TRACE 2024/02/21 09:05:26 modbus: send 00 03 00 00 00 06 04 04 00 0c 00 02
[modbus] TRACE 2024/02/21 09:05:26 modbus: recv 00 03 00 00 00 07 04 04 04 41 b1 d5 e0
Power: 22W

Ladowarka
---------
[modbus] TRACE 2024/02/21 09:05:26 modbus: send 00 04 00 00 00 06 03 04 00 34 00 02
[modbus] TRACE 2024/02/21 09:05:26 modbus: recv 00 04 00 00 00 07 03 04 04 00 00 00 00
Power: 0W

[eebus ] DEBUG 2024/02/21 09:05:26 mdns: stop announcement
```

Not working:

```
evcc meter --log trace,db:error
[main  ] INFO 2024/02/21 09:10:11 evcc 0.123.9
[main  ] INFO 2024/02/21 09:10:11 using config file: /root/evcc.yaml
[mqtt  ] INFO 2024/02/21 09:10:11 connecting evcc-696670209 at tcp://192.168.105.22:1883
[mqtt  ] DEBUG 2024/02/21 09:10:11 tcp://192.168.105.22:1883 connected
[eebus ] INFO 2024/02/21 09:10:11 Local SKI:  bfe624a1d5c92cd51fa43ea0414ccd6a945d1df8
[eebus ] DEBUG 2024/02/21 09:10:11 mdns: announce
[eebus ] DEBUG 2024/02/21 09:10:11 mdns: using zeroconf
[main  ] FATAL 2024/02/21 09:10:11 cannot create meter 'PGE': invalid meter type: modbus
```

```
evcc meter --log trace,db:error
[main  ] INFO 2024/02/21 09:11:42 evcc 0.124.4
[main  ] INFO 2024/02/21 09:11:42 using config file: /root/evcc.yaml
[mqtt  ] INFO 2024/02/21 09:11:42 connecting evcc-1575318572 at tcp://192.168.105.22:1883
[mqtt  ] DEBUG 2024/02/21 09:11:42 tcp://192.168.105.22:1883 connected
[eebus ] INFO 2024/02/21 09:11:42 Local SKI:  bfe624a1d5c92cd51fa43ea0414ccd6a945d1df8
[eebus ] DEBUG 2024/02/21 09:11:42 mdns: announce
[eebus ] DEBUG 2024/02/21 09:11:42 mdns: using zeroconf
[eebus ] DEBUG 2024/02/21 09:11:42 starting websocket server on :4712
[eebus ] DEBUG 2024/02/21 09:11:42 error during websocket server starting: listen tcp :4712: bind: address already in use
[main  ] FATAL 2024/02/21 09:11:42 cannot create meter 'PGE': invalid meter type: modbus
```
PGE config:

```
meters:
[..]
- name: PGE
  type: template
  template: eastron
  modbus: rs485tcpip
  host: 192.168.105.4 # Hostname
  port: 502 # Port
  #rtu: false # rs485 device connected using ethernet adapter
  id: 1
```

mdkeil, 2024-02-21:
> [main  ] FATAL 2024/02/21 09:11:42 cannot create meter 'PGE': invalid meter type: modbus

could it be that evvc 0.124.4 continues to use your old config?

bilbolodz, 2024-02-21:
BINGO!
`using config file: /root/evcc.yaml`
But I've modified: /etc/evcc.yaml

After removing  /root/evcc.yaml 

It's working a little better but still I've minor problems:

```
evcc meter --log trace,db:error
[main  ] INFO 2024/02/21 09:33:15 evcc 0.124.4
[main  ] INFO 2024/02/21 09:33:15 using config file: /etc/evcc.yaml
[mqtt  ] INFO 2024/02/21 09:33:15 connecting evcc-897135371 at tcp://192.168.105.22:1883
[mqtt  ] DEBUG 2024/02/21 09:33:15 tcp://192.168.105.22:1883 connected
[eebus ] INFO 2024/02/21 09:33:15 Local SKI:  bfe624a1d5c92cd51fa43ea0414ccd6a945d1df8
Fronius
-------
[http  ] TRACE 2024/02/21 09:33:15 GET http://192.168.105.252/solar_api/v1/GetPowerFlowRealtimeData.fcgi
[http  ] TRACE 2024/02/21 09:33:15 {
   "Body" : {
      "Data" : {
         "Inverters" : {
            "1" : {
               "DT" : 105,
               "E_Day" : 638.60003662109375,
               "E_Total" : 27541300,
               "E_Year" : 224774.40625,
               "P" : 607
            }
         },
         "Site" : {
            "E_Day" : 638.60003662109375,
            "E_Total" : 27541300,
            "E_Year" : 224774.40625,
            "Meter_Location" : "load",
            "Mode" : "vague-meter",
            "P_Akku" : null,
            "P_Grid" : 31.388382508398763,
            "P_Load" : -638.38838250839876,
            "P_PV" : 607,
            "rel_Autonomy" : 95.083183941245082,
            "rel_SelfConsumption" : 100
         },
         "Version" : "12"
      }
   },
   "Head" : {
      "RequestArguments" : {},
      "Status" : {
         "Code" : 0,
         "Reason" : "",
         "UserMessage" : ""
      },
      "Timestamp" : "2024-02-21T09:34:39+01:00"
   }
}
Power: 607W

PGE
---
[modbus] TRACE 2024/02/21 09:33:15 modbus: send 01 04 00 34 00 02 30 05
[modbus] TRACE 2024/02/21 09:33:25 modbus: send 01 04 00 48 00 02 f1 dd
[modbus] TRACE 2024/02/21 09:33:35 modbus: send 01 04 00 06 00 02 91 ca
Power:          read failed: read tcp 192.168.105.100:33172->192.168.105.4:502: i/o timeout
Energy:         read failed: read tcp 192.168.105.100:60796->192.168.105.4:502: i/o timeout
Current L1..L3: read failed: read tcp 192.168.105.100:54542->192.168.105.4:502: i/o timeout

Dom
---
[modbus] TRACE 2024/02/21 09:33:45 modbus: send 02 04 00 34 00 02 30 36
[modbus] TRACE 2024/02/21 09:33:55 modbus: send 02 04 00 48 00 02 f1 ee
[modbus] TRACE 2024/02/21 09:34:05 modbus: send 02 04 00 06 00 02 91 f9
Power:          read failed: read tcp 192.168.105.100:35738->192.168.105.4:502: i/o timeout
Energy:         read failed: read tcp 192.168.105.100:50044->192.168.105.4:502: i/o timeout
Current L1..L3: read failed: read tcp 192.168.105.100:33016->192.168.105.4:502: i/o timeout

Klimatyzacja
------------
[modbus] TRACE 2024/02/21 09:34:15 modbus: send 04 04 00 34 00 02 30 50
[modbus] TRACE 2024/02/21 09:34:25 modbus: send 04 04 00 48 00 02 f1 88
[modbus] TRACE 2024/02/21 09:34:35 modbus: send 04 04 00 06 00 02 91 9f
Power:          read failed: read tcp 192.168.105.100:58514->192.168.105.4:502: i/o timeout
Energy:         read failed: read tcp 192.168.105.100:47390->192.168.105.4:502: i/o timeout
Current L1..L3: read failed: read tcp 192.168.105.100:43664->192.168.105.4:502: i/o timeout

Ladowarka
---------
[modbus] TRACE 2024/02/21 09:34:45 modbus: send 03 04 00 34 00 02 31 e7
[modbus] TRACE 2024/02/21 09:34:55 modbus: send 03 04 00 48 00 02 f0 3f
[modbus] TRACE 2024/02/21 09:35:05 modbus: send 03 04 00 06 00 02 90 28
Power:          read failed: read tcp 192.168.105.100:58450->192.168.105.4:502: i/o timeout
Energy:         read failed: read tcp 192.168.105.100:43656->192.168.105.4:502: i/o timeout
Current L1..L3: read failed: read tcp 192.168.105.100:41286->192.168.105.4:502: i/o timeout
```

mdkeil, 2024-02-21:
you can try to change your old meter-config:  

look at #11747 

change: `type : modbus` to `type: mbmd`

andig, 2024-02-21:
Sounds like this has been resolved :)

bilbolodz, 2024-02-21:
I think it NOT resolved because new version generates errors:

```
PGE
---
[modbus] TRACE 2024/02/21 09:33:15 modbus: send 01 04 00 34 00 02 30 05
[modbus] TRACE 2024/02/21 09:33:25 modbus: send 01 04 00 48 00 02 f1 dd
[modbus] TRACE 2024/02/21 09:33:35 modbus: send 01 04 00 06 00 02 91 ca
Power:          read failed: read tcp 192.168.105.100:33172->192.168.105.4:502: i/o timeout
Energy:         read failed: read tcp 192.168.105.100:60796->192.168.105.4:502: i/o timeout
Current L1..L3: read failed: read tcp 192.168.105.100:54542->192.168.105.4:502: i/o timeout
```
old version is working fine

bilbolodz, 2024-02-21:
> you can try to change your old meter-config:
> 
> look at #11747
> 
> change: `type : modbus` to `type: mbmd`

I don't quite get it. How to change these
```
- name: PGE
  type: template
  template: eastron
  modbus: rs485tcpip
  host: 192.168.105.4 # Hostname
  port: 502 # Port
  #rtu: false # rs485 device connected using ethernet adapter
  id: 1
```
 to "old-meter config"?

andig, 2024-02-21:
Ok, I see:

```
PGE
---
[modbus] TRACE 2024/02/21 09:05:25 modbus: send 00 01 00 00 00 06 01 04 00 34 00 02
[modbus] TRACE 2024/02/21 09:05:26 modbus: recv 00 01 00 00 00 07 01 04 04 43 8a 2b 16
Power: 276W
```

vs

```
PGE
---
[modbus] TRACE 2024/02/21 09:33:15 modbus: send 01 04 00 34 00 02 30 05
[modbus] TRACE 2024/02/21 09:33:25 modbus: send 01 04 00 48 00 02 f1 dd
[modbus] TRACE 2024/02/21 09:33:35 modbus: send 01 04 00 06 00 02 91 ca
Power:          read failed: read tcp 192.168.105.100:33172->192.168.105.4:502: i/o timeout
```

Looks like TCP vs RTU, not sure. Still waiting for remote access ;)

andig, 2024-02-21:
According to config, he's using templates:

    template: eastron

These are already mapped to mbmd.

mdkeil, 2024-02-21:
> According to config, he's using templates:
> 
> ```
> template: eastron
> ```
> 
> These are already mapped to mbmd.

yes, but in his working old config (v0.123.7) he used the `type: modbus`

bilbolodz, 2024-02-21:
> Looks like TCP vs RTU, not sure. Still waiting for remote access ;)

Email sent

andig, 2024-02-21:
> HF5111B modbus TCP converter

@bilbolodz I guess you are talking Modbus TCP to the converter? The you'll need 

    modbus: tcpip

to make that clear. Pretty sure that this has *never* worked before ;)

closed 2024-02-21 by andig

bilbolodz, 2024-02-21:
Yep it's working. Thank you very much

michiproep, 2024-04-24:
well, I don't know if this is completly resolved.
I haven't noticed that before but I'm gettings this multiple times a day now. v0.125.0
So, it's not happening all the time but at least 10 to 15 times each day. Otherwise it's working fine.
This is my config:
```
- name: Stromnetz       # grid bzw. I/O-Stromnetz  (kW)
    type: template
    template: eastron
    usage: grid
    modbus: tcpip
    id: 1
    host: 192.168.1.47 # Hostname
    port: 502 # Port
```