evcc-io/evcc #12386: Planer: 100% charging until car stops charging

mathiasmora, 2024-02-21:
**Is your feature request related to a problem? Please describe.**
This winter I used the planner with dynamic electricity prices a lot to charge my Model Y with LFP battery. In order to maintain the maximum performance of the battery, Tesla recommends charging the vehicle to 100% once a week. For this reason I often set the planner to 100%. However, I noticed that when the planner is set to 100%, the car does not load until the end of charging. As soon as 100% is reached, charging is stopped by EVCC and charging is not continued until the vehicle finishes charging.

**Describe the solution you'd like**
When the planer is set to 100% the charging should be activ until the vehicle stops the charging "Ladevorgang abgeschlossen". 

**Example**
- Charging was stopped at 100% from EVCC with an active planer and target 100% at 07:14
- I continuted charging by changing in EVCC to "Schnell" and charging continues until 07:25. Then it was ended by the car
[image]
[image]

andig, 2024-02-21:
We can't process incomplete issues

closed 2024-02-21 by andig

andig, 2024-02-21:
Evtl. bringt das die Lösung: https://github.com/evcc-io/evcc/pull/12392

hrncvk, 2024-02-21:
> Evtl. bringt das die Lösung: #12392

Beschreibt genau das selbe verhalten wie ich. Ich öffne gleich das logging über das terminal und übermittelt das log file morgen abend. planung geplant für heute nacht