evcc-io/evcc #12397: Tesla: Finish setup button on add key page doesnt do anything

Diondk, 2024-02-22:
### Describe the bug

I am a sponsor, added evcc keycard to my tesla through: https://www.tesla.com/_ak/tesla.evcc.io
When i scan the QR code i do get the do you want to add the key to your tesla in the app that works fine. But when i click finish setup the page refresh and nothing happens

At this moment i am not able to control my charging for my tesla

### Steps to reproduce

1. go to https://www.tesla.com/_ak/tesla.evcc.io
2. scan qr code, add key to your tesla in the tesla app
3. click finish and nothing happens
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error
  tesla: trace

# unique installation id
plant: xxx

interval: 10s # control cycle interval
sponsortoken: xxx
   
    
meters:
- name: pv2
  type: custom
  power:
      source: http
      uri: http://192.168.1.221/e
      method: GET # default HTTP method
      headers:
        - content-type: application/json
      insecure: true # set to true to trust self-signed certificates
      jq: .[0].ps0 # parse response json
      scale: 1.0 # floating point factor applied to result, e.g. for kW to W conversion
      timeout: 10s # timeout in golang duration format, see https://golang.org/pkg/time/#ParseDuration
- name: dsmr
  type: custom
  power:
    source: calc
    add:
    - source: http
      uri: http://192.168.1.249/api/v2/consumption/electricity-live
      method: GET # default HTTP method
      headers:
      - content-type: application/json
      - X-AUTHKEY: xxx
      insecure: true # set to true to trust self-signed certificates
      jq: .|{currently_delivered}| join(" ")
      scale: 1
      timeout: 10s
    - source: http
      uri: http://192.168.1.249/api/v2/consumption/electricity-live
      method: GET # default HTTP method
      headers:
      - content-type: application/json
      - X-AUTHKEY: xxx
      insecure: true # set to true to trust self-signed certificates
      jq: .|{currently_returned}| join(" ")
      scale: -1
      timeout: 10s
  energy:
    source: calc
    add:
    - source: http
      uri: http://192.168.1.249/api/v2/consumption/today
      method: GET # default HTTP method
      headers:
      - content-type: application/json
      - X-AUTHKEY: xxx
      insecure: true # set to true to trust self-signed certificates
      jq: .|{electricity_merged}| join(" ")
      scale: 1
      timeout: 10s
    - source: http
      uri: http://192.168.1.249/api/v2/consumption/today
      method: GET # default HTTP method
      headers:
      - content-type: application/json
      - X-AUTHKEY: xxx
      insecure: true # set to true to trust self-signed certificates
      jq: .|{electricity_returned_merged}| join(" ")
      scale: -1
      timeout: 10s
  currents:
  - source: http
    uri: http://192.168.1.249/api/v2/datalogger/dsmrreading?limit=1&ordering=-timestamp
    method: GET # default HTTP method
    headers:
    - content-type: application/json
    - X-AUTHKEY: xxx
    insecure: true # set to true to trust self-signed certificates
    jq: .results[]|{phase_power_current_l1}| join(" ")
    scale: 1
    timeout: 10s
  - source: http
    uri: http://192.168.1.249/api/v2/datalogger/dsmrreading?limit=1&ordering=-timestamp
    method: GET # default HTTP method
    headers:
    - content-type: application/json
    - X-AUTHKEY: xxx
    insecure: true # set to true to trust self-signed certificates
    jq: .results[]|{phase_power_current_l2}| join(" ")
    scale: 1
    timeout: 10s
  - source: http
    uri: http://192.168.1.249/api/v2/datalogger/dsmrreading?limit=1&ordering=-timestamp
    method: GET # default HTTP method
    headers:
    - content-type: application/json
    - X-AUTHKEY: xxx
    insecure: true # set to true to trust self-signed certificates
    jq: .results[]|{phase_power_current_l3}| join(" ")
    scale: 1
    timeout: 10s

chargers:
- type: template
  template: twc3 
  host: 192.168.1.173  
  name: wallbox3

loadpoints:
- title: Oprit
  charger: wallbox3
  mode: off
  phases: 3
  #mincurrent: 2
  #maxcurrent: 16
#  resetOnDisconnect: false

site:
  - title: Huis
    meters:
      grid: dsmr # grid meter reference
      pv: # (pvs = veraltet)
        - pv2  # first pv meter reference
vehicles:
  - name: my_car
    type: template
    template: tesla-command
    title: Executor # Wird in der Benutzeroberfläche angezeigt (Optional)
    accessToken: xxxx
    refreshToken: xxx
    vin: xx
    control:
    #onIdentify: # set defaults when vehicle is identified
    mode: pv # enable PV-charging when vehicle is identified

tariffs:
  currency: EUR # three letter ISO-4217 currency code (default EUR)
  grid:
    type: tibber
    token: "xxx" 
  # grid:
    #either static grid price (or price zones)
    # type: fixed
    # price: 0.58 # EUR/kWh
    # zones:
      # - days: Mo-Fr
        # hours: 07-21
        # price: 0.73 # EUR/kWh
      # - days: Sa,So
        # price: 0.58 # EUR/kWh
#tariffs:
#  grid:
#    # either
#    type: tibber
#    token: "xxxx" # access token
mqtt:
  broker: 192.168.1.245:1883
  topic: evcc # root topic for publishing, set empty to disable publishing
  # clientid: foo
  user: evcc
  password: evcc
```

### Log details

```text
2024-02-19T05:25:47.394609+00:00 evcc evcc[4969]: [lp-1  ] ERROR 2024/02/19 05:25:47 vehicle soc: Get "https://fleet-api.prd.eu.vn.cloud.tesla.com/api/1/vehicles/xxx6/vehicle_data": dial tcp: lookup fleet-api.prd.eu.vn.cloud.tesla.com: i/o timeout
2024-02-19T05:25:47.407366+00:00 evcc evcc[4969]: [lp-1  ] WARN 2024/02/19 05:25:47 charger out of sync: expected disabled, got enabled
2024-02-19T05:25:47.407448+00:00 evcc evcc[4969]: [lp-1  ] DEBUG 2024/02/19 05:25:47 limitSoc reached: 91.0% > 90%
2024-02-19T05:25:47.407480+00:00 evcc evcc[4969]: [lp-1  ] ERROR 2024/02/19 05:25:47 charger disable: vehicle not capable of start/stop
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.124.4

mucki12, 2024-02-22:
control: true

Diondk, 2024-02-22:
Let me try that, still doesnt change the fact that the Finish setup just refresh the page, where i am not sure if thats valid behavior.

mucki12, 2024-02-22:
Ok, but you can check the required key directly in the Tesla.

naltatis, 2024-02-23:
If I haven't missed some tesla docs we currently don't have any option to show a success screen or react to the "Finish Setup" button. But this seems to be a known issue: https://github.com/teslamotors/vehicle-command/issues/129

I'll close this for now, since it's not critical and we depend on tesla.

closed 2024-02-23 by naltatis