evcc-io/evcc #12410: Smart loading takes Batterie 🔋

lordvalium, 2024-02-23:
### Describe the bug

[image]

According to the string we see above the car charging bar „smart loading active Lower 17cent…“

it should take grid.

but it discharges my battery 

does someone have similar problems ? 

### Steps to reproduce

1. attach car 
2. smart load setting price 
3. modus PV 
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique id lncobxtiv

interval: 30s # control cycle interval

Token: lukulu 

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: true

modbusproxy:
  - port: 5021
    device: /dev/serial/by-id/usb-FTDI_FT232R_USB_UART_AQ043VSX-if00-port0
    baudrate: 9600
    comset: "8N1"

meters:
- type: template
  template: deye-hybrid 
  id: 1  
  device: /dev/ttyUSB0  
  baudrate: 9600  
  comset: 8N1  
  usage: grid  
  modbus: rs485serial  
  name: grid1
- type: template
  template: deye-hybrid 
  id: 1  
  device: /dev/ttyUSB0  
  baudrate: 9600  
  comset: 8N1  
  usage: pv  
  modbus: rs485serial  
  name: pv2
- type: template
  template: deye-hybrid 
  id: 1  
  device: /dev/ttyUSB0  
  baudrate: 9600  
  comset: 8N1  
  usage: battery  
  modbus: rs485serial  
  name: battery3

chargers:
- type: template
  template: phoenix-ev-eth 
  id: 255  
  host: 192.168.188.45  
  port: 502  
  modbus: tcpip  
  name: wallbox6

vehicles:
- type: template
  template: hyundai 
  title: IONIQ  
  user: lulu.org  
  password: clcificici 
  capacity: 38  
  language: de  
  name: ev4
- type: template
  template: hyundai 
  title: Tucson  
  user: lulu.de  
  password: gonxrov 
  capacity: 13  
  language: de  
  name: ev5

loadpoints:
- title: Car Port
  charger: wallbox6
  mode: pv
  phases: 1
  mincurrent: 6
  maxcurrent: 16
 
 
site:
  title: My home
  meters:
    grid: grid1
    pv:
    - pv2
    battery: 
    - battery3

tariffs:
  grid:
    type: tibber
    token: "qxY8U6wHI-kn18Fcoobfzpiveziv24gHa1KE3LJYpHo" 
    
# mqtt message broker
mqtt:
  broker: 192.168.188.46:1883
  topic: evcc # root topic for publishing, set empty to disable publishing
  user: mqtt-user
  password:
```

### Log details

```text
Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2024/02/23 06:37:39 evcc 0.124.4
[main  ] INFO 2024/02/23 06:37:39 using config file: /config/evcc.yaml
[main  ] INFO 2024/02/23 06:37:39 starting ui and api at :7070
[db    ] INFO 2024/02/23 06:37:39 using sqlite database: /data/evcc.db
[mqtt  ] INFO 2024/02/23 06:37:39 connecting evcc-1586811942 at tcp://192.168.188.46:1883
[mqtt  ] DEBUG 2024/02/23 06:37:39 tcp://192.168.188.46:1883 connected
[proxy-5021] DEBUG 2024/02/23 06:37:39 modbus proxy for /dev/serial/by-id/usb-FTDI_FT232R_USB_UART_AQ043VSX-if00-port0 listening at :5021
[lp-1  ] WARN 2024/02/23 06:37:44 deprecated: minCurrent setting is ignored, please remove
[lp-1  ] WARN 2024/02/23 06:37:44 deprecated: maxcurrent setting is ignored, please remove
[lp-1  ] WARN 2024/02/23 06:37:44 deprecated: phases setting is ignored, please remove
[site  ] WARN 2024/02/23 06:37:44 battery configured but residualPower is missing or <= 0 (add residualPower: 100 to site), see https://docs.evcc.io/en/docs/reference/configuration/site#residualpower
[site  ] INFO 2024/02/23 06:37:44 site config:
[site  ] INFO 2024/02/23 06:37:44   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2024/02/23 06:37:44     grid:      power ✓ energy ✓ currents ✓
[site  ] INFO 2024/02/23 06:37:44     pv 1:      power ✓ energy ✓ currents ✗
[site  ] INFO 2024/02/23 06:37:44     battery 1: power ✓ energy ✓ currents ✗ soc ✓ capacity ✗
[site  ] INFO 2024/02/23 06:37:44   vehicles:
[site  ] INFO 2024/02/23 06:37:44     vehicle 1: range ✓ finish ✓ status ✓ climate ✗ wakeup ✓
[site  ] INFO 2024/02/23 06:37:44     vehicle 2: range ✓ finish ✓ status ✓ climate ✗ wakeup ✓
[lp-1  ] INFO 2024/02/23 06:37:44 loadpoint 1:
[lp-1  ] INFO 2024/02/23 06:37:44   mode:        pv
[lp-1  ] INFO 2024/02/23 06:37:44   charger:     power ✗ energy ✗ currents ✗ phases ✗ wakeup ✗
[lp-1  ] INFO 2024/02/23 06:37:44   meters:      charge ✗
[site  ] DEBUG 2024/02/23 06:37:44 set buffer soc: 90
[site  ] DEBUG 2024/02/23 06:37:44 set smart cost limit: 0.17
[site  ] DEBUG 2024/02/23 06:37:44 set priority soc: 80
[lp-1  ] DEBUG 2024/02/23 06:37:44 phase timer inactive
[lp-1  ] DEBUG 2024/02/23 06:37:44 pv timer inactive
[site  ] DEBUG 2024/02/23 06:37:44 ----
[lp-1  ] DEBUG 2024/02/23 06:37:44 charge power: 0W
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

Current version: 0.124.4

mucki12, 2024-02-23:
deye-hybrid? Are you sure the ACTIVE battery control is supported? Can't find this for deye in the docs:
[image]

mucki12, 2024-02-23:
deye-hybrid? Are you sure that ACTIVE battery control is supported? Can't find this in the docs:
[image]

GrimmiMeloni, 2024-02-23:
Exactly. Battery Control is not supported for this hardware (yet).

GrimmiMeloni, 2024-02-23:
see also #12333

closed 2024-02-23 by GrimmiMeloni

lordvalium, 2024-02-23:
And why do we need battery control for smart charging ? Won’t it be passed from grid directly to the Wallbox ? Von meinem iPhone gesendet

mucki12, 2024-02-23:
No. Your PV system recognizes an electricity demand and covers it. It cannot know who has this electricity demand and covers it with everything possible (grid+battery). The desired result can only be achieved if the battery is actively prevented from discharging.