evcc-io/evcc #12427: Tesla laden <5A

mucki12, 2024-02-23:
### Describe the bug

Erst noch einmal vielen Dank für die vielen Änderungen, die durch die neue Tesla Fleet API notwendig waren.

Mit der vorherigen Owners API, einem TWC3 und einem Tesla war es mit evcc möglich das Fahrzeug in einem Bereich von 1A bis 16A zu laden. Mit Umsetzung der Fleet API werden scheinbar "nur" noch Ladevorgänge >=5A unterstützt.

Screenshots und Log sind angehängt. Wenn detaillierte Logs benötigt werden bitte kurz Bescheid geben. Ich kann derzeit nicht beurteilen, ob die derzeitige Tesla Umsetzung überhaupt für <5A vorgesehen ist.

Mit dem Problem bin ich zumindest nicht alleine, da ich im Forum bereits nachgefragt hatte:
https://github.com/evcc-io/evcc/discussions/12355

 
[image]
[image]

[evcc-nightly.csv](https://github.com/evcc-io/evcc/files/14387011/evcc-nightly.csv)

### Steps to reproduce

1. evcc Version 0.124.4 fe8ec4fb nightly installieren
2. Tesla und TWC3 konfigurieren 
3. Tesla mit mincurrent: 1 / maxcurrent: 16 konfigurieren
4. In der GUI im Loadpoint min 1A angegeben
5. Tesla lädt auf Min+PV (kein Überschuss vorhanden) mit 5A (erwartet 1A)
6. Umgeschaltet auf Schnell - Tesla lädt mit 16A (wie erwartet)
7. Wieder zurück geschaltet auf Min+PV - Tesla lädt wieder mit 5A (erwartet 1A)

### Configuration details

```yaml
chargers:
- type: template
  template: twc3 
  host: 192.168.2.170  
  name: wallbox5

vehicles:
- type: template
  template: tesla 
  title: Tesla  
  icon: car  
  accessToken: XX  
  refreshToken: EU_XX  
  vin: LRWXX  
  capacity: 79  
  phases: 3  
  mode: pv  
  minCurrent: 1  
  maxCurrent: 16  
  name: ev4

loadpoints:
- title: Einfahrt
  charger: wallbox5
  vehicle: ev4
  mode: pv
```

### Log details

```text
Siehe angehängte Datei oben. Wenn andere detaillierte Logs benötigt werden bitte kurz Bescheid geben.

Umschalten auf Min+PV z. B. an folgenden Stellen:
2024-02-23T14:36:00
2024-02-23T14:27:54
2024-02-23T13:56:54

Ausgabe evcc charger:
[main  ] INFO 2024/02/23 16:37:13 evcc 0.124.4 (fe8ec4fb)                                                                                                                                                                                                                                   
[... 23 lines cut ...]
Ausgabe evcc vehicle:
[main  ] INFO 2024/02/23 16:44:06 evcc 0.124.4 (fe8ec4fb)                                                                                                                                                                                                                                   
[main  ] INFO 2024/02/23 16:44:06 using config file: /etc/evcc.yaml                                                                                                                                                                                                                         
[db    ] INFO 2024/02/23 16:44:06 using sqlite database: /root/.evcc/evcc.db                                                                                                                                                                                                                
[mqtt  ] INFO 2024/02/23 16:44:06 connecting evcc-977816722 at tcp://192.168.2.189:1883                                                                                                                                                                                                     
[mqtt  ] DEBUG 2024/02/23 16:44:06 tcp://192.168.2.189:1883 connected                                                                                                                                                                                                                       
.Soc:          asleep                                                                                                                                                                                                                                                                       
Capacity:      79.0kWh                                                                                                                                                                                                                                                                      
Charge status: asleep                                                                                                                                                                                                                                                                       
Charged:       asleep                                                                                                                                                                                                                                                                       
Range:         asleep                                                                                                                                                                                                                                                                       
Odometer:      asleep                                                                                                                                                                                                                                                                       
Finish time:   asleep                                                                                                                                                                                                                                                                       
Target Soc:    asleep                                                                                                                                                                                                                                                                       
OnIdentified:  Mode:pv, MinCurrent:1, MaxCurrent:16                                                                                                                                                                                                                                         
Features:      [CoarseCurrent]
```

### What type of operating system are you running?

Docker container

### Version

0.124.4 (fe8ec4fb)

pezet1910, 2024-02-23:
den einzigen Unterschied sehe ich bei dem Fahrzeug. In meiner yaml sieht es so aus:vehicles:
- name: my_car
  type: template
  template: tesla
  title: tesla
  accessToken: token
  refreshToken: refresh token
  capacity: 82

mucki12, 2024-02-23:
@andig 
Das hatte ich schon befürchtet und daher im verlinkten Thread schon nach den benötigten Logs gefragt.
Sorry, ich bin maximal verwirrt und stehe da auf dem Schlauch. Wird jetzt ein detailliertes Log für den Tesla benötigt (für den TWC3 ja wahrscheinlich nicht, oder)?

Also in der evcc.yaml einfach:
levels:
  tesla: trace
eintragen?

Sorry, bin eigentlich in HA unterwegs. Habe für die Fehlersuche jetzt die nightly als Docker installiert und liefere gerne alles was nötig ist.

@pezet1910 
Habe die evcc.yaml wie bei dir reduziert und kämpfe mit dem gleichen Fehlerbild.

andig, 2024-02-23:
Mach mal bitte im morgigen nightly ein 

    evcc vehicle --log trace,db:error --current 1

Ich vermute da wird die 1 an Tesla geschickt, dort aber möglicherweise ignoriert...

themex85, 2024-02-24:
> den einzigen Unterschied sehe ich bei dem Fahrzeug. In meiner yaml sieht es so aus:vehicles:
> 
> * name: my_car
>   type: template
>   template: tesla
>   title: tesla
>   accessToken: token
>   refreshToken: refresh token
>   capacity: 82

fehlt da nicht ein 
control: true

Bei mir funktioniert die Steuerung grundsätzlich bis auf 1A runter. Ich habe nur das Problem, dass der Tesla nicht aufgeweckt werden kann.

mucki12, 2024-02-24:
@themex85 
Nein - in der aktuellen nightly fehlt da kein control: true :-)

Bin gerade am Testen und poste dann die Ergebnisse. Scheinbar unterscheidet sich das Verhalten tatsächlich ein wenig ob der Tesla wach ist oder schläft.

mucki12, 2024-02-24:
@andig 
Habe die nightly evcc 0.124.4 (3bf57f87) installiert und scheinbar ist das (zumindest bei mir?) ausschließlich ein Problem im Bezug auf einen schlafenden oder eines wachen Teslas.

Test mit frischer Installation und Tesla ist wach:
- Start mit Modus Aus
- Umstellung  auf Min+PV = es wird direkt mit 1A geladen
- Umstellung auf Schnell = es wird direkt mit 16A geladen
- Umstellung  auf Min+PV = es wird zunächst mit 5A geladen, aber nach 2-3 Minuten wird mit 1A geladen

Standard-Log Tesla ist schon wach:
[evcc-nightly-3.csv](https://github.com/evcc-io/evcc/files/14393717/evcc-nightly-3.csv)

Ausgabe von evcc vehicle --log trace,db:error --current 1:
```
/app # evcc vehicle --log trace,db:error --current 1                                                                                                                                                                                                                                        
[main  ] INFO 2024/02/24 16:26:27 evcc 0.124.4 (3bf57f87)                                                                                                                                                                                                                                   
[main  ] INFO 2024/02/24 16:26:27 using config file: /etc/evcc.yaml                                                                                                                                                                                                                         
[mqtt  ] INFO 2024/02/24 16:26:27 connecting evcc-500154744 at tcp://192.168.2.189:1883                                                                                                                                                                                                     
[mqtt  ] DEBUG 2024/02/24 16:26:27 tcp://192.168.2.189:1883 connected                                                                                                                                                                                                                       
[tesla ] TRACE 2024/02/24 16:26:27 GET https://fleet-api.prd.na.vn.cloud.tesla.com/api/1/users/region                                                                                                                                                                                       
[tesla ] TRACE 2024/02/24 16:26:28 {"response":{"region":"eu","fleet_api_base_url":"https://fleet-api.prd.eu.vn.cloud.tesla.com"}}                                                                                                                                                          
[tesla ] TRACE 2024/02/24 16:26:28 GET https://fleet-api.prd.eu.vn.cloud.tesla.com/api/1/vehicles                                                                                                                                                                                           
[tesla ] TRACE 2024/02/24 16:26:28 {"response":[{"id":929845894233529,"vehicle_id":1689266382481080,"vin":"LRWxxx","color":null,"access_type":"OWNER","display_name":"","option_codes":null,"granular_access":{"hide_private":false},"tokens":["xxx","xxx
23"],"state":"online","in_service":false,"id_s":"929845894233529","calendar_enabled":true,"api_version":73,"backseat_token":null,"backseat_token_updated_at":null,"ble_autopair_enrolled":false}],"count":1}                                                                                
[tesla ] TRACE 2024/02/24 16:26:28 POST https://tesla.evcc.io/api/1/vehicles/LRWxxx/command/set_charging_amps                                                                                                                                                                    
[tesla ] TRACE 2024/02/24 16:26:28 {"charging_amps": 1}                                                                                                                                                                                                                                     
--                                                                                                                                                                                                                                                                                          
{"response":{"result":true,"reason":""}}                                                                                                                                                                                                                                                    
/app
```

Test mit frischer Installation und Tesla schläft:

- Start mit Modus Aus
- Fehlermeldungen „asleep“ (siehe Screenshot)
- Umstellung  auf Min+PV = es wird mit 5 A geladen  
Log:  ERROR 2024/02/24 17:50:28 max charge current 1A: asleep
- Umstellung auf Schnell = es wird direkt mit 16A geladen
- Umstellung  auf Min+PV = es wird wieder mit 5A geladen (ändert sich auch nach knapp 10 Minuten nicht - Ausgabe von evcc vehicle --log trace,db:error --current 1 ist zu diesem Zeitpunkt die gleiche wie oben)
- Umstellung auf Schnell = es wird mit 16A geladen
- Umstellung auf Min+PV = es wird mit 5A geladen (ändert sich auch nach knapp 10 Minuten nicht)
- Umstellung erneut auf Schnell = es wird mit 16A geladen
- Umstellung erneut auf Min+PV = es wird mit 5A geladen, aber nach 2 bis 3 Minuten geht er runter auf 1A!

Standard-Log Tesla schläft:
[evcc-nightly-5.csv](https://github.com/evcc-io/evcc/files/14393728/evcc-nightly-5.csv)

Sorry, aber wahrscheinlich bin ich auch hier keine große Hilfe. Wenn ich die anderen Rückmeldungen so überblicke, scheint das ja kein allgemeines Problem zu sein. Was wird benötigt? Ein Trace Log für Tesla für das zuletzt beschriebene Verhalten mit einem schlafenden Tesla?

[image]

mucki12, 2024-02-24:
Nachtrag:
Gerade erst gesehen - an der Sleep Baustelle wird ja noch fleißig gearbeitet https://github.com/evcc-io/evcc/pull/12369

Das dürfte in der von mir getesteten nightly doch noch gar nicht enthalten sein, oder irre ich mich. Vielleicht wäre damit ja das Problem schon behoben?

andig, 2024-02-24:
Was hat das alles mit https://github.com/evcc-io/evcc/issues/12427#issuecomment-1961694130 zu tun?

mucki12, 2024-02-24:
Ja läuft super. Nach einem Start von evcc mit einem schlafenden Tesla halt in der Realität dann doch nur mit >=5A, aber gut, dann lebe ich halt damit.

mucki12, 2024-02-24:
Die angefragten? Angefragt war eigentlich /app # evcc vehicle --log trace,db:error --current 1   und die habe ich ja geliefert. Die Ausgabe zeigt ja keine Fehler und da der Fehler daher in der Theorie nicht vorkommen kann (und somit bei allen anderen aus dem verlinkten Thread auch nicht) ist ja alles ok.

andig, 2024-02-24:
Das Log beweist, dass die Nachricht zu Tesla geschickt wird. Vielleicht wendest Du Dich einfach mal dahin...

andig, 2024-03-20:
Replaced by https://github.com/evcc-io/evcc/issues/13007

closed 2024-03-20 by andig