evcc-io/evcc #12442: Questions regarding support for linux/arm/v7 (runtime and buildtime)

berndkrannich, 2024-02-24:
### Describe the bug

I'm running a bespoke Docker build to ultimately create an evcc container running on a Raspberry Pi 4 (`--platform=linux/arm/v7`). Running the build on Raspberry Pi 4 itself (`--build-arg BUILDPLATFORM=linux/arm/v7`), the build is breaking in the `make ui` stage since `0.124.2`:

```
npm run build

> build
> vite build

/build/evcc/node_modules/rollup/dist/native.js:38
  throw new Error(
        ^

Error: Your current platform "linux" and architecture "arm" combination is not yet supported by the Rollup build. Please use the WASM build "@rollup/wasm-node" instead.

The following platform-architecture combinations are supported:
android-arm
android-arm64
darwin-arm64
darwin-x64
linux-arm
linux-arm64
linux-arm64 (musl)
linux-riscv64
linux-x64
linux-x64 (musl)
win32-arm64
win32-ia32
win32-x64

If this is important to you, please consider supporting Rollup to make a native build for your platform and architecture available.
    at Object.<anonymous> (/build/evcc/node_modules/rollup/dist/native.js:38:8)
    at Module._compile (node:internal/modules/cjs/loader:1356:14)
    at Module._extensions..js (node:internal/modules/cjs/loader:1414:10)
    at Module.load (node:internal/modules/cjs/loader:1197:32)
    at Module._load (node:internal/modules/cjs/loader:1013:12)
    at ModuleWrap.<anonymous> (node:internal/modules/esm/translators:202:29)
    at ModuleJob.run (node:internal/modules/esm/module_job:195:25)
    at async ModuleLoader.import (node:internal/modules/esm/loader:336:24)
```

I was able to trace the issue down to the `vite` upgrade in #12207. The changes in vite and potential solutions are mentioned e.g. in:

* <https://rollupjs.org/migration/#general-changes>
* <https://github.com/vitejs/vite/issues/15167>
* <https://github.com/vitejs/vite/issues/15122>
* <https://github.com/vitejs/vite/issues/15120>
* <https://stackoverflow.com/a/77918301>

I'm not in all the npm details, but was able to get evcc to compile for both `linux/amd64` and `linux/arm/v7` by adding the following to `package.json` (potentially, some of the entries below aren't required but I was trying to replace all instances of rollup consistently):

```
  "overrides": {
    "vite": {
      "rollup": "npm:@rollup/wasm-node"
    },
    "@vitejs/plugin-legacy": {
      "rollup": "npm:@rollup/wasm-node"
    },
    "@vitejs/plugin-vue": {
      "rollup": "npm:@rollup/wasm-node"
    },
    "vite-plugin-toml": {
      "rollup": "npm:@rollup/wasm-node"
    },
    "vitest": {
      "rollup": "npm:@rollup/wasm-node"
    },
    "rollup-plugin-visualizer": {
      "rollup": "npm:@rollup/wasm-node"
    }
  },
```

A couple of questions, I guess:
1. Is there an interest to allow evcc to be compiled *for* `linux/arm/v7` or is this out of scope?
2. If 'yes' for number 1, would it make sense to include the platform into the standard list of binaries produced for a release (<https://github.com/evcc-io/evcc/blob/master/.github/workflows/release.yml#L47>)?
3. Is there an interest to allow evcc to be also compiled *on* `linux/arm/v7` or is this out of scope?
4. If 'yes' for number 3, I'm not sure if the above is a sustainable fix or rather a workaround. My understanding is that a wasm-based build should work on all platforms but I am not sure about this.

I'm not necessarily depending on the above for my bespoke build, but as it took me a bit to understand the error, how to fix/workaround it, and reproduce in the standard evcc Docker build, I wanted to make sure to document my findings and raise the above questions (also because others might have the same findings).

### Steps to reproduce

On a clean checkout of evcc `0.124.2` or `0.124.4`:

```

docker build --build-arg BUILDPLATFORM=linux/arm/v7 --platform linux/arm/v7 .
```

### Configuration details

```yaml
n/a
```

### Log details

```text
n/a
```

### What type of operating system are you running?

Docker container

### Version

0.124.2 onwards

andig, 2024-02-24:
> Is there an interest to allow evcc to be compiled for linux/arm/v7 or is this out of scope?

Imho not since there is little to no advantage of doing so.

> Is there an interest to allow evcc to be also compiled on linux/arm/v7 or is this out of scope?

Not sure. Can already be solved by cross-compiling. I'd rather go for performant build. @naltatis

naltatis, 2024-02-26:
Yes, being able to build evcc on all common platforms would be nice. But I don't really like the `override` fix. Especially the aspect that we won't catch mistakes in our main workflows (gh actions, local setup) if this override breaks due to regular dependency upgrades.
This is something the Vite project has to sort out.